MichaIng/DietPi #6593: OKDO Rock 4C+ Starter Kit: no more HDMI after 8.21 update, SSH is working

lfiorini, 2023-09-01:
#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Kernel version
Linux ****** 6.1.46-current-rockchip64 #3 SMP PREEMPT Wed Aug 16 16:27:31 UTC 2023 aarch64 GNU/Linux

- SBC model 
ROCK 4 (aarch64)

- Power supply used
5V 3A

- SD card used
Transcend 32GB A1

#### Steps to reproduce
1. The box was working with 8.19
2. dietpi-update
3. reboot

#### Expected behaviour
- HDMI should works

#### Actual behaviour
- HDMI says No signal

MichaIng, 2023-09-01:
Nasty. Could you test a different monitor and/or HDMI cable to rule out that HDMI signal strength has probably changed with the new kernel?

Also can you check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

A downgrade should be possible like this:
```sh
apt install --reinstall --allow-downgrades linux-{image,dtb}-current-rockchip64=23.02.2
```
In case check for the correct version string via:
```sh
apt show linux-image-current-rockchip64
```

lfiorini, 2023-09-04:
I tried to change both cable and monitor: same result. Today I applied 5 apt updates related to rockip64 firmware but same result, the kernel is: 6.1.50-current-rockchip64. Below there is output of dmesg -l 0,1,2,3.

After apt install --reinstall --allow-downgrades linux-{image,dtb}-current-rockchip64=23.02.2 HDMI worked again.
(uname -a says Linux ******* 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux)

I have several identical OKDO units so I can make other tests if nedded.

dmesg -l 0,1,2,3 before downgrading:

[    1.601789] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with 0-001b
[    1.747707] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with vcc3v3-sys
[    1.761588] phy phy-ff7c0000.phy.6: phy poweron failed --> -110
[    1.761737] dwc3 fe800000.usb: error -ETIMEDOUT: failed to initialize core
[    1.773994] rk808 0-001b: failed to read the chip id at 0x17
[    9.662243] rockchip-pinctrl pinctrl: pin gpio3-29 already requested by leds; cannot claim for ff880000.i2s
[    9.662289] rockchip-pinctrl pinctrl: pin-125 (ff880000.i2s) status -22
[    9.662311] rockchip-pinctrl pinctrl: could not request pin 125 (gpio3-29) from group i2s0-8ch-bus-bclk-off  on device rockchip-pinctrl
[    9.662340] rockchip-i2s ff880000.i2s: Error applying setting, reverse things back
[    9.662359] rockchip-i2s ff880000.i2s: bclk disable failed -22
[    9.710507] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    9.715913] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    9.753618] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x0000000c] -6
[    9.753854] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000003] -6
[    9.754708] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.754932] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.755169] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    9.755386] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    9.755621] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    9.755836] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    9.756051] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.756265] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.756479] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.756696] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.757105] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000009] -6
[    9.757320] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000a] -6
[    9.757832] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000b] -6
[    9.758050] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000001] -6
[    9.763013] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.763270] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.763499] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.763835] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.764128] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.764358] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.764589] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    9.764818] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.765046] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.765294] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    9.765681] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000a] -6
[    9.765940] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    9.766194] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.766422] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    9.766650] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[    9.766881] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[... cut ...]
[    9.771198] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000015] -6
[    9.771409] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000002f] -6
[    9.771619] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000001] -6
[    9.771830] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000d] -6
[    9.772040] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000019] -6

MichaIng, 2023-09-07:
I see the known USB 3.0/dwc3 host errors and some related to UART, I2C and GPIO. Then `es8316` errors, which is the onboard audio chip, AFAIK. Are these also present on the old kernel, resp. does onboard audio work with the new one?

I guess the only line which could be related to HDMI should be:
```
[    1.773994] rk808 0-001b: failed to read the chip id at 0x17
```
The RK808 chip is the internal power supply including converters etc.

If you find time to test, the Armbian 6.1.50 images have the same issue, right?
https://dl.armbian.com/rockpi-4cplus/archive/

lfiorini, 2023-09-18:
I've just booted ArmBian 23.8.1 6.1.50 31.08.2023 on the same board and HDMI works normally on a Acer 22" monitor but shows "incomplete" chars on a Samsung UE32 TV.

MichaIng, 2023-09-18:
Hmm, with the exact same kernel you had on DietPi, that is surprising. Not sure if the bootloader can have an effect. Did you try flashing the one on the DietPi image to SD card? Can be done via `dietpi-config` > `Advanced Options` > `Update MMC bootloader`. I'll als build a new image.

lfiorini, 2023-09-19:
cat /proc/device-tree/model returns Raxda ROCK PI 4B.

MichaIng, 2023-09-19:
Oh, that might be the issue. The device tree is not hardcoded in `/boot/dietpiEnv.txt`, is it? We removed that a while ago since the bootloader picked the correct one depending on the model automatically. And it really is a 4C+, right?

lfiorini, 2023-09-20:
The box says OKDO (Radxa) ) ROCK 4C+ Startert Kit.

cat /boot/dietpiEnv.txt

rootdev=UUID=e3efb75f-9006-499e-81df-daa2d8b8f213
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=tty1
verbosity=4
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=

lscpu:

Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          6
On-line CPU(s) list:             0-5
Thread(s) per core:              1
Core(s) per socket:              3
Socket(s):                       2
NUMA node(s):                    1
Vendor ID:                       ARM
Model:                           4
Model name:                      Cortex-A53
Stepping:                        r0p4
BogoMIPS:                        48.00
NUMA node0 CPU(s):               0-5
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Mmio stale data:   Not affected
Vulnerability Retbleed:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid

lfiorini, 2023-09-20:
I did Update MMC bootloader and then I run dietpi-update but I got the same result HDMI not working. Let me know if you need more tests.

lfiorini, 2023-09-28:
Just to inform that I tried another but the issue remains.

- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
bullseye

- Kernel version
Linux swbox 6.1.50-current-rockchip64 #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux

- SBC model
ROCK 4 (aarch64)

[    1.599889] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with 0-001b
[    1.747020] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with vcc3v3-sys
[    1.765241] phy phy-ff7c0000.phy.6: phy poweron failed --> -110
[    1.765480] dwc3 fe800000.usb: error -ETIMEDOUT: failed to initialize core
[    1.781983] rk808 0-001b: failed to read the chip id at 0x17
[    9.728672] rockchip-pinctrl pinctrl: pin gpio3-29 already requested by leds; cannot claim for ff880000.i2s
[    9.728729] rockchip-pinctrl pinctrl: pin-125 (ff880000.i2s) status -22
[    9.728762] rockchip-pinctrl pinctrl: could not request pin 125 (gpio3-29) from group i2s0-8ch-bus-bclk-off  on device rockchip-pinctrl
[    9.728806] rockchip-i2s ff880000.i2s: Error applying setting, reverse things back
[    9.728837] rockchip-i2s ff880000.i2s: bclk disable failed -22
[    9.771186] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    9.777012] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    9.813715] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x0000000c] -6
[    9.813959] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000003] -6
[    9.814782] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.815001] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.815238] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    9.815456] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    9.815693] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    9.815907] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    9.816124] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.816339] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.816554] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.816768] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    9.817152] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000009] -6
[    9.817402] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000a] -6
[    9.817620] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000b] -6
[    9.817832] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000001] -6
[    9.822704] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.822961] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.823189] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.823479] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.823742] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.823970] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.824198] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    9.824425] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.824653] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    9.824894] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    9.825145] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000a] -6
[    9.825480] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    9.825752] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    9.825999] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    9.826334] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[    9.826594] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[    9.826856] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[... cut ...]
[    9.831111] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000015] -6
[    9.831324] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000002f] -6
[    9.831538] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000001] -6
[    9.831749] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000d] -6
[    9.831961] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000019] -6

jvteleco, 2023-10-02:
I don't have a Rock4 C+, but I do have a Rock4 SE and I also had issues booting with the latest armbian or dietpi (basically the LED was only green, never booted, never blinked double as the heartbeat. With the official Radxa it booted OK.
I used SD card and sadly I don't have a serial cable for debugging.

I think is related to the latest 6.0 kernel (as you kind of proved when you downgraded to 5.15 (https://github.com/MichaIng/DietPi/issues/6593#issuecomment-1704759741)

I did manage to boot the rock4 SE without problem from an SD with the image **rock4 B armbian 23.05 bullseye 5.15** built by Radxa from the armbian sources (on their web under the "third party". https://wiki.radxa.com/Rockpi4/downloads and then finding the release that still has armbian:  "https://github.com/radxa-build/rock-pi-4b/releases/tag/20230314-1307  " -> Armbian_23.05.0-trunk_Rockpi-4b_bullseye_current_5.15.102_kde-plasma_desktop.img.xz

So I guess you could try the rock4 B above or the rock4 C (not sure which one is more appropiate for your 4C+):  https://github.com/radxa-build/rock-pi-4c/releases/tag/20230314-1308  
[Armbian_23.05.0-trunk_Rockpi-4c_bullseye_current_5.15.102_kde-plasma_desktop.img.xz](https://github.com/radxa-build/rock-pi-4c/releases/download/20230314-1308/Armbian_23.05.0-trunk_Rockpi-4c_bullseye_current_5.15.102_kde-plasma_desktop.img.xz)

There is a thread on armbian forum related to the emmc, but a person found some differences between kernel 5.1, 5.15 and 6.1 and it gets mentioned no HDMI output: https://forum.armbian.com/topic/24605-issue-with-emmc-nvme/?do=findComment&comment=156107

Hope it helps for both of you and dietpi is back soon running fine on Rock4 :)

Thanks @MichaIng for your incredible work.

MichaIng, 2023-10-02:
Actually here is a very current matching report about the missing HDMI output: https://forum.armbian.com/topic/30440-rock-pi-4c-no-hdmi-video-output-also-overlays/

Sadly conversation drifted quickly to funding and meta problems instead of focusing on the actual problem.

Here is the second major problem: missing USB 3.0 port support on many RK3399 boards since the update: https://forum.armbian.com/topic/30343-rock-pi-4c-top-usb-30-port-not-working-how-to-change-to-host-mode/

Same for Orange Pi 5, which is strange, since I do not remember we had issues there: https://forum.armbian.com/topic/29642-orange-pi-5-usb-30-ports-not-working/

Here a topic which seems to contain a fix "from LibreELEC": https://forum.armbian.com/topic/26818-opi-4-lts-no-hdmi-output/

The "subset" of the patches: https://github.com/paolosabatino/armbian-build/commit/a77a626
Uff, I am sure it would be possible to have this working with a few lines changed, as I don't think mainline Linux would have completely broken the HDMI subsystem for all RK3399 SBCs (which is one of the widely used ones). Would be actually great to just test it with an unmodified mainline kernel build.

jvteleco, 2023-10-04:
Just to add another test I just did with the Rock 4 SE (not C+).

Flashed again an SD card with the rock4 B armbian 23.05 bullseye 5.15.  HDMI Works.
Did an update + upgrade, eventually it installed the new kernel named as  "rockchip64 (23.8.1)"  and once I rebooted,  "dead". No HDMI and  I cannot ping to the device  (I actually forgot to try SSH).

Hope they fix the kernel issue as you mention with that patch or similar, and then dietpi and armbian should work again nicely on the Rock family :)

Only thing I actually noticed today: I use an HDMI to DVI cable for my monitor. Not sure if that could also be an issue, if the HDMI doesn't generate the correct signal for DVI with the new kernel?

MichaIng, 2024-04-21:
Could someone test the new image from here: https://dietpi.com/downloads/images/testing/
It uses a new ROCK 4C Plus U-Boot build, which selects the right device tree automatically.

MichaIng, 2024-05-02:
I moved the images to stable downloads and will add them to our download page now. Marking this as closed.

closed 2024-05-02 by MichaIng