MichaIng/DietPi #6594: please add support for the Orange Pi Zero 3

dirkhh, 2023-09-01:
### Creating an image request

#### Formal device information
- Device name | Orange Pi Zero 3
- Official product URL | [OrangePi.org](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-3.html)
- GitHub resource URL (if available) | not GitHub, but a seemingly useful [Wiki](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_3#Instructions_for_use_of_Debian.2FUbuntu_Server_and_Xfce_desktop_system)
- Image download URL | [Debian for 1G RAM image](https://drive.google.com/file/d/1S78xVIi8hh7vxqvSO3sttLljtInmMvcG/view?usp=drive_link)... it's odd that there are three different images, though, depending on the amount of RAM installed

#### Is the SBC officially supported by the Debian installer?
- no

#### If not, is a reliable 3rd party Debian image available for this SBC?
- from the Wiki it appears that there are images (see link above). How reliable...? no idea.

#### If not, are there install instructions for Debian available?
[Google Drive link](https://drive.google.com/file/d/1S78xVIi8hh7vxqvSO3sttLljtInmMvcG/view?usp=drive_link)
- ...

dirkhh, 2023-09-27:
so... how does one go about working towards adding a new image? This is such a cool board (great price point and more capable than anything else at that price point - wired and wireless network, 1GB (up to 4GB) - decent availability and under $25 / €30)

MichaIng, 2023-09-27:
Will be added with next release. We have samples already.

MichaIng, 2023-09-27:
There is a mainline kernel device tree in upcoming Linux 6.6: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner/sun50i-h618-orangepi-zero3.dts?h=v6.6-rc3

Ethernet seems to be not 100% stable yet: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=f1b3ddb3ecc2eec1f912383e01156c226daacfab
And likely other features are missing. The commit message indicates that basically only the onboard features defined in the device tree are what is functional, i.e. no WiFi, no HDMI. Will test it regardless, but I think for now we should use the vendor kernel sources, which are Linux 6.1 at least: https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.1-sun50iw9

Why does the device tree from Orange Pi kernel sources define it as Allwinner H616, while it is actually an H618, correctly defined in mainline Linux? This guarantees havoc when switching between kernels, respectively makes the vendor U-Boot default non-functional with mainline kernel as of the necessarily false device tree name. There are always these little annoying inconsistencies which cause distributors a lot of problems when dealing with vendor kernel and bootloader sources ... I'll never understand the reason for this:
- https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-6.1-sun50iw9/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero3.dts
- http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_3#Hardware_Features_of_Orange_Pi_Zero_3

dirkhh, 2023-09-27:
I have been playing with the vendor kernel (well, with their Debian distro) to assess how useful this board is - and there wifi seems to work, I haven't seen ethernet issue (but then, I haven't exactly put a lot of load on it).
The vendor kernel refers to sun50iw9 as the board identifier. But then it does indeed claim to be an h616... I didn't even notice that. UGH.

MichaIng, 2023-09-27:
Ah yes, the limitations I listed are with mainline kernel, as the device tree was just added very recently there. Hence for now we should stay with vendor kernel.

bigross8, 2023-10-20:
Any progress on making the image? This board is excellent for the price and I'm planning to replace my NanoPi NEO with it. A Diet-Pi image on OPi Zero 3 would be sweet. Thanks.

daniel-mller-87, 2023-10-21:
I ordered the same board today. Would be excellent to have a dietpi image ready for use.

dirkhh, 2023-10-21:
I used mine on and off with their factory build. The board is great - I'm really impressed at the price point (at least from AliExpress). But their image sucks :)

MichaIng, 2023-10-21:
First step done: https://github.com/MichaIng/DietPi/commit/5c20bbc
So far more dummy entries, but it allows us to generate images with stable branch during next development cycle, as only the build tools and in case linux branches need to be further extended.

daniel-mller-87, 2023-10-22:
> First step done: [5c20bbc](https://github.com/MichaIng/DietPi/commit/5c20bbc) So far more dummy entries, but it allows us to generate images with stable branch during next development cycle, as only the build tools and in case linux branches need to be further extended.

Thx.

daniel-mller-87, 2023-11-01:
any news? Got my hardware today but the standard image isn't the best. Will you support 4GB Ram as well?

bigross8, 2023-11-01:
> any news? Got my hardware today but the standard image isn't the best. Will you support 4GB Ram as well?

I tried the DietPi installation script (https://dietpi.com/docs/hardware/#make-your-own-distribution), but unfortunately the board can't seem to connect to my network after a reboot. I tried both Wi-Fi and Ethernet, setting AUTO_SETUP_AUTOMATED flag to 1 in /boot/dietpi.txt, beta and dev branch of the installer, and different versions of the official Orange Pi Debian image (both 1.0.0 and 1.0.2). I don't have a Micro HDMI cable or an UART reader so I don't know if there's even any output.

MichaIng, 2023-11-01:
I guess some firmware is missing. I'll start working on it this week.

daniel-mller-87, 2023-11-01:
I'm looking forward to. I also tried to make an image. It's possible to do so from the 4GB Bullseye Image. And then upgrade to Bookworm during the dietpi Script. It's not stable however. The only solution to get ethernet working ist do unplug power cable and then plug back in. If I try to reboot via terminal it just says: Failed to connect to bus: No such file or directory. Afterwards it becomes unreachable.

bigross8, 2023-11-03:
> I guess some firmware is missing. I'll start working on it this week.

I borrowed a Micro HDMI cable, and the board boots up fine. In dietpi-config however, the Ethernet and WiFi adapters are not found. So you're right about this, there's definitely firmware missing.

daniel-mller-87, 2023-11-08:
When will the next Dietpi released?

Joulinar, 2023-11-08:
https://github.com/MichaIng/DietPi/issues/6704

daniel-mller-87, 2023-11-10:
Thx a lot. Does it mean orange pi zero 3 will be supported with the upcoming release? Doesn't seem to be that easy.

MichaIng, 2023-11-10:
Images are ready for testing: https://dietpi.com/downloads/images/testing/

They currently use the unmodified kernel, dtb, bootloader and firmware packages from Orange Pi. I'll do own builds, but use the same build system, since it is an older fork of Armbian (the build system, not the package sources), which allows some consistency and potentially a simple migration if Armbian themselves add support in the future.

MichaIng, 2023-11-10:
The code to support it as an own hardware ID has been added with last release already. Now it has been added to our build script. So yes, initial support is there, now I want to enhance the kernel build and testing is needed.

daniel-mller-87, 2023-11-10:
Alright thx a lot. I just tried the image. Booted fine. So far I still see two major problems

1. reboot over ssh doesn't work:
I'm receiving: Failed to connect to bus: No such file or directory
only unplugging from power helps

2. if I let it run for some time. Top will always show utilisation of 1.0 So something unnecessary is running. When tried to create an image on my own I had this same issue but just deleted user orange pi to "solve" it. Maybe this is resolved when you do own an own build.

MichaIng, 2023-11-10:
> Failed to connect to bus: No such file or directory

Why the hack do others still see this while we added a workaround already which works perfectly fine on all my systems. Can you paste the output of the following command:
```sh
declare -f reboot
ls -l /etc/systemd/system/systemd-logind.service
```
And you do use the `reboot` command or some other, like `shutdown -r` or `systemctl reboot`?
And in case you use `reboot` and the above commands only tell that no such file exists: Can you repeat the exact setup steps you did after flashing the image? 

However, this is a visual error only and does not actually prevent reboot. I guess it does actually reboot but does not come online anymore since the Ethernet device is lost. At least this is what I am facing: On a reboot, the Ethernet adapter is detected for a short time, and the interface generated, but removed quickly again. I need to check back with Orange Pi. After cold boot (power cycle), this does not happen. Also no such issue with WiFi.

daniel-mller-87, 2023-11-10:
Shure here's the output. I'm using reboot now. I don't care if it's only a message. But as you say only cold boot reenables ethernet functionality. 

```shell
root@DietPi:~# declare -f reboot
ls -l /etc/systemd/system/systemd-logind.service
reboot () 
{ 
    local command='reboot';
    for i in "$@";
    do
        case $i in 
            '-p' | '--poweroff')
                command='poweroff'
            ;;
            '--reboot' | '--no-wall')
                :
            ;;
            '--halt')
                command='halt'
            ;;
            *)
                /sbin/reboot "$@";
                return $?
            ;;
        esac;
    done;
    systemctl start "$command.target"
}
lrwxrwxrwx 1 root root 9 Nov 10 00:35 /etc/systemd/system/systemd-logind.service -> /dev/null
```

daniel-mller-87, 2023-11-10:
I tried systemctl reboot as well. With the same issue

MichaIng, 2023-11-10:
Okay, the `reboot` function is defined. So when using `reboot`, the error message does not appear, right? Same with `poweroff` and `halt`. All other commands to shutdown or reboot the system are expected to throw this (visual-only) error, unless you installed (software which requires) `systemd-logind` and hence `dbus`.

Good to know that the Ethernet issue is not individual but seems to affect all Zero 3 boards 🤔. This is something which needs to be solved most likely in the bootloader by Orange Pi. Some device state is probably not correctly reset.

bigross8, 2023-11-10:
> 1. reboot over ssh doesn't work:
>    I'm receiving: Failed to connect to bus: No such file or directory
>    only unplugging from power helps

Just tried it on my board, I don't see the "Failed to connect to bus" error message. Same issue with the Ethernet, fortunately the Wi-Fi works after a reboot. Are there some ethernet logs I can check?

daniel-mller-87, 2023-11-10:
Exactly, power off and halt also don't generate an error. I think the ethernet bug is the biggest issue. I'd use it as pi-hole and home assistant wifi connection isn't an option here.

bigross8, 2023-11-10:
For me, even the reboot command doesn't generate the error. But what I've noticed is that logged in as "dietpi", this error will show up for all three commands.
`dietpi@DietPi:~$ reboot
Failed to connect to bus: No such file or directory
dietpi@DietPi:~$ poweroff
Failed to connect to bus: No such file or directory
dietpi@DietPi:~$ halt
Failed to connect to bus: No such file or directory`
When logged in as "root", the errors don't show up.

daniel-mller-87, 2023-11-10:
same here. First time I tried as normal user with sudo.

Joulinar, 2023-11-10:
you need to use them with `sudo` command

bigross8, 2023-11-10:
I know, my point was just to showcase that the error appears on all three commands.

MichaIng, 2023-11-10:
Right, unprivileged users of course cannot just reboot or poweroff the system 😉. The `dbus` connection is then used for another (reasonable) purpose, to check whether policykit grants users these permissions.

`poweroff`, `reboot` and `halt` all behave the same in this regards: They all usually should throw the error, but thanks to the functions we defined, the underlying reason for the `dbus` call is bypassed. When calling them as non-root user, it throws the same error for a different reason, same as `systemctl` itself does.

> Are there some ethernet logs I can check?

I have all logs about it here, but it is outside of my abilities to derive the underlying issue or fix it. This is something for Orange Pi kernel/bootloader developers.

bigross8, 2023-11-10:
That's unfortunate. I think there's an image with 5.4.125 kernel but I think it's too outdated to start tinkering with. Who knows, there may be a mainline kernel available by the time Orange Pi developers will fix the Ethernet issue. 😁

daniel-mller-87, 2023-11-10:
There's also an Image with Bulseye and 6.1. As long as you don't upgrade to bookworm it won't have this error. Also the Ubuntu 22.04 Image with 6.1 was working for me. However, I don't really trust their image, as there are always 3 Users running and weird high load sometimes.
Waiting for some Armbian or better Dietpi Release.

MichaIng, 2023-11-10:
> As long as you don't upgrade to bookworm it won't have this error.

I had the same error with it, without upgrading to Bookworm. Also the userland/distribution version has no effect on such kind of kernel and/or bootloader related issues.

There is a very similar issue with the NanoPi R4S, where on reboot one of the two Ethernet adapters can disappear. But it only happens when the board has reached a certain temperature in my case. Very strange but probably some bad quality thermal-dependant resistor or weak soldering (but then on many/all boards at the very same place). Probably it is similar here, so that it it does not always happen.

dirkhh, 2023-11-10:
Testing the image linked above here using wifi and the ADS-B Feeder Image software. Seems solid and performs surprisingly well for such a tiny board... I'm impressed. Can't wait for this to be part of the regular releases...

daniel-mller-87, 2023-11-10:
I just tried the Ubuntu 22.04 image with 6.1 Kernel again. The ethernet bug doesn't show up here. Is it possible to take the uboot or so from here? I gues the kernel is the same as in Debian.

Though this image is working I wouldn't use it because of its changed mirrors etc. I don't know what else is running there.

MichaIng, 2023-11-11:
I retested the original Debian Bullseye image form Orange Pi, and indeed at first the Ethernet adapter remains present after `reboot`. So I purged all the bloat in small chunks, including some of Orange Pi's own packages to see which one might be responsible, without finding any. Then I just enabled the Ethernet interface with a simple config:
```sh
cat << '_EOF_' > /etc/network/interfaces
iface eth0 inet static
address 192.168.1.15/24
gateway 192.168.1.1
_EOF_
ifup eth0
```
Internet is there, however, after reboot, the adapter is gone. Also without enabling the interface after `reboot`, it does not come back until a power cycle. It remains present, including after `reboot`s, until I just enable it once again.

@daniel-mller-87 since I removed NetworkManager already, can you replicate the same on the Debian and/or Ubuntu images, that the adapter gets lost after reboot once you actually enable/use it once? It sounds suspicious that in your case it was lost after a Bookworm upgrade, which of course requires Internet as well. So probably the upgrade was not the issue, so simply that the adapter was used.

I tried to narrow it down and set the interface UP with low level tools:
```sh
ip l set up dev eth0
```
This enables the carrier signal, but it does not trigger the issue. I will further try to add an IP address and a default route manually and see at which point the issue appears.

daniel-mller-87, 2023-11-11:
Maybe it needs the network manager? Tell me what I should check? It's still running the ubuntu 22.04. and it survives multiple reboots. Internet as well as normal network is working. Tried a HA Docker which was working fine. However, load average is always 1.0. Maybe I should just grab one of these Intel n100 mini pcs from Ali an just install standard debian with standard kernel. Should be easy to apply dietpi script to a standard kernel etc. This thing seems to be a mess for everyone.

MichaIng, 2023-11-11:
Are you using DHCP over Ethernet currently?

> Maybe it needs the network manager?

More likely `ifup` does something which NetworkManager does not, which breaks it. It is not something missing, but something done, causing the problem.

daniel-mller-87, 2023-11-11:
Yes, I do. However, my DHCP-Server will always assign the same ip.

MichaIng, 2023-11-11:
Do you have a monitor and keyboard to attach, just in case network does not come up again? If so, could you try this:
```sh
cat << '_EOF_' > /etc/network/interfaces
auto eth0
iface eth0 inet static
address 192.168.1.15/24
gateway 192.168.1.1
_EOF_
systemctl disable NetworkManager
reboot
```
After first reboot, network should come up again, but, if the issue is the same on the Ubuntu image, after a second reboot, it won't. Btw, the kernel is the same like on the Debian version, I mean exactly the same, including build date etc?
```console
root@orangepizero3:~# uname -a
Linux orangepizero3 6.1.31-sun50iw9 #1.0.2 SMP Mon Sep 25 13:54:37 CST 2023 aarch64 GNU/Linux
```

daniel-mller-87, 2023-11-11:
It's the same Kernel but with different date:
```shell
Linux orangepizero3 6.1.31-sun50iw9 #1.0.2 SMP Thu Sep 28 14:15:40 CST 2023 aarch64 aarch64 aarch64 GNU/Linux
``` 

I'm not shure if I understand. Should I try the first step - with network manager disabled? I have it only headless but don't mind losing the image if it breaks. I don't trust it at this stage.

MichaIng, 2023-11-11:
> It's the same Kernel but with different date:

Okay, I could not find any commit in Orange Pi's kernel, U-Boot or build repo, so it should all be practically the same.

> I'm not shure if I understand. Should I try the first step - with network manager disabled? I have it only headless but don't mind losing the image if it breaks. I don't trust it at this stage.

Yeah right, the commands configure `ifupdown` and disable NetworkManager, so it should be the absolute minimum change to verify that the same happens on all Orange Pi images, and that `ifupdown` is the issue, btw regardless whether DHCP or a static IP is used (in my case both trigger the issue). So it seems to be some invisible pre/post step `ifup` does.

That's the kind of issues we love, when we have no way to find out what's going on, Orange Pi itself does not have an issue, as NetworkManager does not trigger it, and `ifupdown`/Debian does not have an issue, as `ifup` does not cause issues like that on any other hardware (and likely any other kernel build) ...

daniel-mller-87, 2023-11-11:
That did it. Now Ubuntu is broken the same way. I need to cold cycle to access it again. Exactly the same behaviour.

MichaIng, 2023-11-11:
Okay, thanks for testing. Do this to revert to NetworkManager:
```sh
> /etc/network/interfaces
systemctl enable --now NetworkManager
```

daniel-mller-87, 2023-11-11:
thx. But I'll wait for the proper image now :D. Never planned on using the Chinese mirror ubuntu.

MichaIng, 2023-11-11:
> Never planned on using the Chinese mirror ubuntu.

You can change it in `/etc/apt/sources.list`. As the host key matches, you can be sure that it is an exact mirror of any other official Ubuntu APT server. However, the image composition itself with lots of bloat and stuff running + autologin on all (serial) consoles etc is not great.

I btw verified that using NetworkManager here also works fine. I'll open an issue at Orange Pi and one at the Debian bug tracker, in the hope to find out what additional thing `ifup` does on top of what its verbose output indicates. It looks like it is Debian themselves who manage the source code, at least I cannot find any external/upstream source for it.

jmescorcio, 2023-11-12:
I have the Orange PI Zero 3 4g, but when trying to use this test image, it always hangs on this screen, regardless of whether I select YES or NO
[image]

daniel-mller-87, 2023-11-12:
> I have the Orange PI Zero 3 4g, but when trying to use this test image, it always hangs on this screen, regardless of whether I select YES or NO [image]

It's working with my 4G version. However, you have to use the bulseye server or the the ubuntu 22.04 server image. Didn't test the other ones. Bookworm Server never really worked even in initial state. 

Best approach at the moment ist to use the Bullseye Server -> apply dietpi script (with bookworm) -> apply hack by Michaeling above -> kill all processes by user orangepi -> remove user orangepi. 
That leaves you with a system with average load of 0.0. But still with some bugs

bigross8, 2023-11-12:
Using the testing image running AdGuard Home, Network UPS Tools, Docker with a couple of containers and WireGuard, everything is running smoothly at the moment. The issue with the average load being at 1.00 only seems to be a visual one, because I'm not seeing elevated temperatures or anything running in htop. One thing I noticed is when I tested the ethernet using iperf3, the upload speed seems to top off at around ~800Mbit/s, probably the limitation of the controller or the driver. Download speed is full gigabit.

MichaIng, 2023-11-12:
@jmescorcio 
> when trying to use this test image, it always hangs on this screen, regardless of whether I select YES or NO

What do you mean by "hangs"? What do you see on the screen?

If you choose "Yes", you should see some output that the serial console is disabled, if you select "No", the next that should appear immediately is this message:
```sh
[ .... ] mount -o remount /var/log
```
with a moving dot animation. If really nothing happens, does it help to press CTRL+C to abort the script?

@daniel-mller-87 
> Bookworm Server never really worked even in initial state.

Since it does work here, aside of the Ethernet issue (which is the same on a migrated image), what exactly does not work for you?

> kill all processes by user orangepi -> remove user orangepi

I added these cleanup steps to our installer: https://github.com/MichaIng/DietPi/commit/7c37a6e

I btw found some adapter-specific Ethernet-related device tree adjustment in the Orange Pi boot script. Since their images are affected by the lost adapter as well, it obviously does not fix that issue, but probably it enhances performance and/or stability of Ethernet connections on some revisions:
```sh
if test "${ethernet_phy}" = "rtl8211f"; then
        fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <3100>
        fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <700>
fi

if test "${ethernet_phy}" = "yt8531c"; then
        fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <0>
        fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <600>
fi
```
According to the hardware schematic from their website it is a YT8531C, but on the other hand that document refers to the 4G variant and probably it is sometimes an RTL8211F, or even a 3rd possible chip (where the default device tree has values set correctly for). I'll add this to our boot script as well.

Also here a topic at Armbian forum regarding this board: https://forum.armbian.com/topic/29202-orange-pi-zero-3/
Some also face this Ethernet issue after "some" reboots. Quite possible that also NetworkManager (which is default on Armbian images, AFAIK) causes this in some circumstance. I mean it must be some regular adapter state toggled which causes this, done by `ifupdown` whenever setting up an interface, and by other network managers probably in more specific circumstances.

jmescorcio, 2023-11-12:
> @jmescorcio
> 
> > ao tentar usar essa imagem de teste, ela sempre trava nesta tela, independentemente de eu selecionar SIM ou NÃO
> 
> O que você quer dizer com "pendura"? O que você vê na tela?
> 
> Se você escolher "Sim", você deve ver alguma saída que o console serial está desativado, se você selecionar "Não", o próximo que deve aparecer imediatamente é esta mensagem:
> 
> ```shell
> [ .... ] mount -o remount /var/log
> ```
> 
> com uma animação de pontos em movimento. Se realmente nada acontecer, será que ajuda pressionar CTRL+C para abortar o script?
> 
> @daniel-mller-87
> 
> > Bookworm Server nunca realmente funcionou, mesmo no estado inicial.
> 
> Já que ele funciona aqui, além do problema da Ethernet (que é o mesmo em uma imagem migrada), o que exatamente não funciona para você?
> 
> > matar todos os processos pelo usuário orangepi -> remover usuário orangepi
> 
> Eu adicionei estas etapas de limpeza ao nosso instalador: [7c37a6e](https://github.com/MichaIng/DietPi/commit/7c37a6e)
> 
> Eu btw encontrei algum ajuste de árvore de dispositivo relacionado à Ethernet específico do adaptador no script de inicialização do Orange Pi. Uma vez que suas imagens são afetadas pelo adaptador perdido também, obviamente não corrige esse problema, mas provavelmente melhora o desempenho e/ou a estabilidade das conexões Ethernet em algumas revisões:
> 
> ```shell
> if test "${ethernet_phy}" = "rtl8211f"; then
>         fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <3100>
>         fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <700>
> fi
> 
> if test "${ethernet_phy}" = "yt8531c"; then
>         fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <0>
>         fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <600>
> fi
> ```
> 
> De acordo com o esquema de hardware de seu site é um YT8531C, mas por outro lado esse documento se refere à variante 4G e provavelmente às vezes é um RTL8211F, ou até mesmo um 3º chip possível (onde a árvore de dispositivos padrão tem valores definidos corretamente para). Vou adicionar isso ao nosso script de inicialização também.
> 
> Também aqui um tópico no fórum Armbian sobre esta placa: https://forum.armbian.com/topic/29202-orange-pi-zero-3/ Alguns também enfrentam esse problema de Ethernet após "algumas" reinicializações. É bem possível que também o NetworkManager (que é padrão em imagens Armbian, AFAIK) cause isso em alguma circunstância. Quero dizer que deve ser algum estado de adaptador regular alternado que causa isso, feito sempre que configurar uma interface, e por outros gerenciadores de rede provavelmente em circunstâncias mais específicas.`ifupdown`
I want to say that it simply hangs on this screen, so far the settings are fine, when I get to this screen, it doesn't matter if I choose yes or no, it freezes and doesn't leave this screen. I used the image you provided at this link: https://dietpi.com/downloads/images/testing/

jmescorcio, 2023-11-12:
> @jmescorcio
>
>
> O que você quer dizer com "pendura"? O que você vê na tela?
>
> Se você escolher "Sim", você deve ver alguma saída que o console serial está desativado, se você selecionar "Não", o [...]

I want to say that it simply hangs on this screen, so far the settings are fine, when I get to this screen, it doesn't matter if I choose yes or no, it freezes and doesn't leave this screen. I used the image you provided at this link: https://dietpi.com/downloads/images/testing/

jmescorcio, 2023-11-12:
> > Eu tenho o Orange PI Zero 3 4g, mas ao tentar usar essa imagem de teste, ele sempre trava nessa tela, independente de eu selecionar SIM ou NÃO [image]
> 
> Está funcionando com a minha versão 4G. No entanto, você tem que usar o servidor bulseye ou a imagem do servidor ubuntu 22.04. Não testou os outros. Bookworm Server nunca realmente funcionou, mesmo no estado inicial.
> 
> Melhor abordagem no momento ist para usar o Bullseye Server -> aplicar script dietpi (com bookworm) -> aplicar hack por Michaeling acima -> matar todos os processos pelo usuário orangepi -> remover usuário orangepi. Isso deixa você com um sistema com carga média de 0,0. Mas ainda com alguns bugs

I used the image from this link: https://dietpi.com/downloads/images/testing/

MichaIng, 2023-11-12:
Strange. Did you access via HDMI or SSH? Can you access via SSH (or a 2nd SSH session) while it hangs at this prompt to check what is going on?

jmescorcio, 2023-11-12:
> Estranho. Você acessou via HDMI ou SSH? Você pode acessar via SSH (ou uma 2ª sessão SSH) enquanto ele trava neste prompt para verificar o que está acontecendo?

I accessed it via HDMI, but I couldn't access it via SSH, although it showed the IP and had Ping. I tested it several times, made other cards with the images and the same thing happened.

daniel-mller-87, 2023-11-12:
> @jmescorcio
> 
> > when trying to use this test image, it always hangs on this screen, regardless of whether I select YES or NO
> 
> What do you mean by "hangs"? What do you see on the screen?
> 
> If you choose "Yes", you should see some output that the serial console is disabled, if you select "No", the next that should appear immediately is this message:
> 
> ```shell
> [ .... ] mount -o remount /var/log
> ```
> 
> with a moving dot animation. If really nothing happens, does it help to press CTRL+C to abort the script?
> 
> @daniel-mller-87
> 
> > Bookworm Server never really worked even in initial state.
> 
> Since it does work here, aside of the Ethernet issue (which is the same on a migrated image), what exactly does not work for you?
> 
> > kill all processes by user orangepi -> remove user orangepi
> 
> I added these cleanup steps to our installer: [7c37a6e](https://github.com/MichaIng/DietPi/commit/7c37a6e)
> 
> I btw found some adapter-specific Ethernet-related device tree adjustment in the Orange Pi boot script. Since their images are affected by the lost adapter as well, it obviously does not fix that issue, but probably it enhances performance and/or stability of Ethernet connections on some revisions:
> 
> ```shell
> if test "${ethernet_phy}" = "rtl8211f"; then
>         fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <3100>
>         fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <700>
> fi
> 
> if test "${ethernet_phy}" = "yt8531c"; then
>         fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <0>
>         fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <600>
> fi
> ```
> 
> According to the hardware schematic from their website it is a YT8531C, but on the other hand that document refers to the 4G variant and probably it is sometimes an RTL8211F, or even a 3rd possible chip (where the default device tree has values set correctly for). I'll add this to our boot script as well.
> 
> Also here a topic at Armbian forum regarding this board: https://forum.armbian.com/topic/29202-orange-pi-zero-3/ Some also face this Ethernet issue after "some" reboots. Quite possible that also NetworkManager (which is default on Armbian images, AFAIK) causes this in some circumstance. I mean it must be some regular adapter state toggled which causes this, done by `ifupdown` whenever setting up an interface, and by other network managers probably in more specific circumstances.

The 4GB image of bookworm wasn't accessible over ethernet-ssh. So I needed to drop it from the start.

MichaIng, 2023-11-12:
Okay, probably the missing Ethernet device tree changes are causing the SSH (network) issues.

MichaIng, 2023-11-12:
A new image is built: https://github.com/MichaIng/DietPi/actions/runs/6843288284

They contain this additional block in `/boot/boot.cmd` right before the `booti` command:
```sh
if test "${ethernet_phy}" = "rtl8211f"; then
        echo "Ethernet adapter: ${ethernet_phy}, adjusting device tree ..."
        fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <3100>
        fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <700>
elif test "${ethernet_phy}" = "yt8531c"; then
        echo "Ethernet adapter: ${ethernet_phy}, adjusting device tree ..."
        fdt set /soc/ethernet@5020000 allwinner,rx-delay-ps <0>
        fdt set /soc/ethernet@5020000 allwinner,tx-delay-ps <600>
else
        echo "Ethernet adapter: \"${ethernet_phy}\""
fi
```
And for informational reasons, this variable is also added to the cmdline, so it can be checked back after boot. In my case, the variable is empty. Not sure if this means that my adapter matches the default, where no adjustment is needed, and hence the reasons why I do not face any issues (aside of lost adapter after reboot), or if this code is a no-op.

@daniel-mller-87 
Did you use the `master` version of `dietpi-installer` or the `dev` version? If `master`, could you print the output of the following commands, to see whether one of the device tree changes has been applied in your case:
```sh
apt install xxd
xxd /proc/device-tree/soc/ethernet@5020000/allwinner,rx-delay-ps
xxd /proc/device-tree/soc/ethernet@5020000/allwinner,tx-delay-ps
```

[128 more replies]

closed 2024-02-23 by MichaIng