MichaIng/DietPi #6597: building from a different repo, or from a local tree

dirkhh, 2023-09-04:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- right now it is rather unintuitive how one can build from their own development branch. At least in my testing, I need to both set and export the `G_GITHUBOWNER` and `G_GITHUBBRANCH` environment variables, **and** start the `dietpi-build` script with `-o` and `-b`. That's... a bit confusing. But also... if I'm starting this in a local checked out tree... why are we fetching those scripts from GitHub in the first place? Just use the one that's **right there**

#### Describe the solution you'd like:

- I would love to be able to build from a local tree. I'm happy to work on a PR for that if that's interesting to you
- I would at least like to understand why it makes sense to have to use both the command line arguments and the exported variables to state which repo to use

#### Additional context

- it's possible that I'm simply not using this the way it's supposed to be used. If there's some document that would help me understand all this, please feel free to simply point me to that

MichaIng, 2023-09-07:
You are right and it makes sense to make this easier, but some notes to our defence:
- This script was created for our own image builds in the first place, where I wanted to get this automated quickly on GitHub Actions. I did not find the time yet to make this more user friendly, e.g. also adding long/self-explaining CLI option variants, some documentation etc. Making it easier to build with an own/local branch is certainly part of this.
- Generally it makes sense to have the branch of the build scripts themselves, and the branch of the DietPi code installed on the resulting image separate: We often want to build new images with recent changes done to our build scripts (i.e. `dev` or or feature branch), while the DietPi scripts/code on the resulting images almost always must be `master`/stable.

I think easiest solution is to check whether `dietpi-installer` and `dietpi-imager` scripts exist beside the `$0` script path itself, which would then indicate a local clone/branch, and then just executing those instead of downloading from GitHub (using `$G_GITOWNER/$G_GITBRANCH`).

dirkhh, 2023-09-07:
You should never feel the need to defend yourself. This is an **amazing** project.
I'm simply trying to help and to figure out how best to do that.

I hadn't thought about the logic of WANTING to build the `master` branch with newer tools in a different branch. That's just not a use case I even thought about... but it makes total sense.
Instead of doing magic detection of file locations, my recommendation would be to go with a straight forward command line switch. `--build-in-place` or something like that. And then it's explicit and doesn't cause confusion. And by definition it doesn't break the other existing use cases.

MichaIng, 2023-09-07:
Consequently we could also add `--toolchain-gitowner`/`--toolchain-gitbranch` flags to define those for the tools. Generally I like CLI options more than environment variables. Probably we find better names.

dirkhh, 2023-09-07:
very, very much agree. The whole environment variable thing is way too fragile for my taste... I only did this in my `add-fat` branch because it looked like that was the preferred way used here. I'm happy to turn that into a command line option if you want. 😎