MichaIng/DietPi #6600: /etc/hosts/ on DietPI

privacyguy123, 2023-09-05:
Mine currently is:
```
127.0.0.1 localhost
127.0.1.1 DietPi
```

Could ```127.0.1.1``` ever interfere with routing from OpenVPN or WireGuard connected client PC wanting to use AdguardHome installed on DietPi as a DNS resolver? Currently trying to wrap my head around OpenVPN/WireGuard refusing to connect to internet with DNS **SPECIFIED** in the confs as ```192.168.0.100``` ... I see no traffic hitting my AdguardHome admin panel yet I can connect to it with the VPN on. When I leave the DNS settings **BLANK** it goes back to working again. 😕

Joulinar, 2023-09-05:
127.0.0.1 is the localhost and has nothing to do with VPN. Try to use the VPN server IP 10.x.x.x instead.

privacyguy123, 2023-09-05:
> 127.0.0.1 is the localhost and has nothing to do with VPN. Try to use the VPN server IP 10.x.x.x instead.

If I use the VPNs server then my traffic won't be routed through AdguardHome which is what I'm trying to achieve ...

Joulinar, 2023-09-05:
Does AGH and VPN server are running on same device?

MichaIng, 2023-09-07:
AGH and the VPN are running on different systems, i.e. IP forwarding is required, how did you install WireGuard resp. OpenVPN? Our WireGuard implementation enables the needed IP forwarding automatically, but our OpenVPN server implementation indeed does not, at least not the iptables NAT rules. Something like this would be required:
```sh
iptables -A FORWARD -i tun0 -j ACCEPT
iptables -t nat -A POSTROUTING -s 10.8.0.0/24 -o "$(G_GET_NET iface)" -j MASQUERADE
ip6tables -A FORWARD -i tun0 -j ACCEPT
ip6tables -t nat -A POSTROUTING -o "$(G_GET_NET iface)" -j MASQUERADE
```
with `10.8.0.0/24` as OpenVPN subnet.