MichaIng/DietPi #6601: Unable to uninstall softwares after system crash

Blistex77, 2023-09-05:
#### Details:
- Date           | Tue Sep  5 13:11:37 CEST 2023
- DietPi version | v8.21.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | `Linux DietPi 5.10.0-25-amd64 #1 SMP Debian 5.10.191-1 (2023-08-16) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `pip3 uninstall -y motioneye`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Try to uninstall software
2. Have system crash during above operation
3. Try to uninstall same software again as it still appears in the list
#### Expected behaviour:

- Software should uninstall and not appear in the installed applications anymore.
#### Actual behaviour:

- Software still appears in the installed applications.
#### Extra details:

- This issue occurs for several applications on my system. It seems I can't uninstall any software now.
#### Additional logs:
```
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 9, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main_parser.py", line 8, in <module>
    from pip._internal.cli import cmdoptions
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/cmdoptions.py", line 24, in <module>
    from pip._internal.cli.parser import ConfigOptionParser
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/parser.py", line 12, in <module>
    from pip._internal.configuration import Configuration, ConfigurationError
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/configuration.py", line 26, in <module>
    from pip._internal.utils.logging import getLogger
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 29, in <module>
    from pip._internal.utils.misc import ensure_dir
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/misc.py", line 42, in <module>
    from pip._internal.locations import get_major_minor_version
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/locations/__init__.py", line 67, in <module>
    from . import _distutils
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/locations/_distutils.py", line 20, in <module>
    from distutils.cmd import Command as DistutilsCommand
ModuleNotFoundError: No module named 'distutils.cmd'

Joulinar, 2023-09-05:
If you are hit by the issue, you should see the error handling menu. There should be an option to enter an alternative command. Pls remove the whole command and replace it with `true`. This should skip the step and continue uninstall.

Blistex77, 2023-09-05:
Thanks a lot!

Blistex77, 2023-09-05:
I just tried and it did the trick, thanks a lot!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sans
virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2023-09-05 by Blistex77

MichaIng, 2023-09-07:
Note that the underlying issue as not solved, just bypassed. As of the error message, I guess `pip` is still installed while the `distutils` module is not. Try this:
```sh
sudo apt install python3-distutils
sudo pip3 uninstall -y motioneye
```
If this was your only Python application, you could then also uninstall Python 3 and pip:
```sh
sudo apt-mark auto python3-distutils
sudo dietpi-software uninstall 130
```