MichaIng/DietPi #6603: CasaOS / Umbrel integration?

DartPower, 2023-09-06:
Can you add a CasaOS and/or Umbrel in app install menu? 
This is a very user-friendly web control systems with docker-based apps.

MichaIng, 2023-09-07:
Note that such "OS" integrations often double and hence conflict with other DietPi features, and either turn the system into a single purpose environment, or add/change so much that there is not much left of the DietPi being a minimal base distro.

It might be different in these particular cases, but often it is better to just download and use their images than installing them via scripts on DietPi or any other Debian base. This also guarantees full support from their end.

Also UmbrelOS has a one-line installer:
```sh
curl -sSfL https://umbrel.sh | sudo bash
```
For completeness the one for CasaOS:
```sh
curl -sSfL https://get.casaos.io | sudo bash
```
I'm open to be proven wrong, so feel free to run those and test/compare all DietPi and Umbrel/CasaOS features to see which one double, which one in case conflict and what works well concurrently OOTB. CasaOS is making vast use of Docker, and UmbrelOS comes especially with a web interface as far as I can see. So both at least have some potential to avoid conflicts and/or extend DietPi with something we do not have onboard OOTB.

MichaIng, 2026-10-27:
Umbriel is now an OS image only, and cannot be installed onto an existing distro:
- https://github.com/getumbrel/umbrel#download-and-install
- https://raw.githubusercontent.com/getumbrel/umbrel/master/scripts/install

And for CasaOS, I do not see much reason to add it to `dietpi-software`. It is a one-line installer script, with not much benefit but only maintenance efforts, if we reimplemented it as a somewhat tailored version ourselves. Also it really only installs Docker containers, where I'd probably prefer upstream instructions and templates, or `dietpi-software` for more flexible and lightweight setups. The GUI is nice though.

The reason that their banner looks like `dietpi-banner`, is that its installer and welcome banner codes are a copy&paste from `dietpi-banner` (before we switched to named colour indexes). Theoretically violating the GPL for our code, if I'd care 😄.

Hence closing this.

closed 2026-10-27 by MichaIng