MichaIng/DietPi #6604: Warning during reboot

fra87, 2023-09-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `bookworm`
- Kernel version | `Linux Server 6.1.50-current-rockchip64 #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`
- SBC model | `ROCK64 (aarch64)`

#### Additional Information (if applicable)
- Software title | reboot
- Was the software title installed freshly or updated/migrated? Not actively
- Can this issue be replicated on a fresh installation of DietPi? Not sure, but this happens on two different servers

#### Steps to reproduce
1. Connect through SSH
2. Perform a reboot through `sudo reboot`

#### Expected behaviour
- The SBC shall reboot without warnings

#### Actual behaviour

The SBC outputs this warning before actually rebooting

```
Failed to set wall message, ignoring: Unit dbus-org.freedesktop.login1.service failed to load properly, please adjust/correct and reload service manager: File exists
Call to Reboot failed: Unit dbus-org.freedesktop.login1.service failed to load properly, please adjust/correct and reload service manager: File exists
```

#### Extra details
The system actually reboots, so it's just a warning, but this may be a symptom of something worse.

This happens on both servers I have (both based on the same rockchip, but different SBCs).

This started happening recently

Joulinar, 2023-09-06:
Nothing to worry about, this is expected actually and not doing any harm to the system.

MichaIng, 2023-09-06:
Btw, systemd guys, at least Lennart Poettering, see `dbus` and `systemd-logind` as mandatory and will not add a setting to prevent reboot/poweroff from doing the dbus>logind call for sending a wall message (even that none is/would be actually sent, the connection is still done) or bypassing shutdown inhibitors: https://github.com/systemd/systemd/issues/28902

I hate the fact that more and more systemd features are tied deeply into each other, for this to grow more and more into an all or nothing complex. None of all these features is used by anything on a clean Debian, wall messages are never sent on immediate reboots/poweroffs and nonsense on single admin systems, as shutdown inhibitors are (this is that e.g. an open SSH session can prevent others from utilising a shutdown), extended shutdown log entries and timed shotdowns. Basically all these features are useful mostly on systems where multiple users are working concurrently on. Disillusioning that there is zero understanding for small IoT/home/office server use cases, at least keeping features optional. But well, this thinking is the reason for ongoing criticism from large parts of the traditional UNIX/Linux community.

I also checked the `dbus-broker` mentioned in the issue. Sadly it is an even more complex system than the `dbus-daemon`, with dedicated master and child processes, so even less an option for our case. That it is seen as successor (well, it might be, if one has any use of all its features) is hence another negative development for our use case.

The only thing we could do to avoid pre-installing dbus and logind is creating aliases for `poweroff` and `reboot` to `systemctl start poweroff.target` resp. `systemctl start reboot.target`, and remove them when logind is unmasked. Of course we could also give have have it enabled OOTB, but I still do not like shipping our images with two additional processes and APIs never used on 90% of systems and where most users won't know what they exactly do 🤔.

fra87, 2023-09-06:
To be honest I think the main advantage of using dietpi is that it is a very lightweight system, so I totally agree with your concern of adding useless services. After all, normal PCs have plenty of horsepower, while usually SBCs have very few resources.

I did not know the reason behind the warning, but I already suspected (and Joulinar confirmed) that this was just a pointless warning. Honestly I think this is something that can be fixed if there is a simple fix (I don't know if the aliasing can have downsides), but since it is only "cosmetic" I'd consider it a very low priority issue and, most important, it shall not have any negative impact

Thank you both for the very quick feedback and the (as usual) great work in keeping this distro well maintained

MichaIng, 2023-09-07:
Just to clarify: It is not only cosmetic, but indeed mentioned features provided by logind do not work. They are just usually not required for a headless single-admin server system. As fast as a desktop is installed, dbus and logind become mandatory to grant desktop users required permissions and manage desktop sessions etc, which is why we automatically install dbus and unmask logind when a desktop is selected in `dietpi-software`.

SteveJM, 2024-09-25:
It is not only cosmetic as it confuses the [Anisible Reboot Module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/reboot_module.html). It thinks it failed. Ok, you can get around that by adding `ignore_errors: yes`, but that is only a partial fix. Because Ansible thinks the command failed, it does not bother to wait for the reboot to complete :-(

MichaIng, 2024-09-26:
The warnings have been "solved" already in the meantime. We basically created aliases for `reboot`/`poweroff`/`halt`, which bypass the logind call by starting `reboot.target` and alike directly. This still does a regular shutdown via systemd, but skips the logind request.

However, we probably need a menu entry to toggle logind:
```sh
sudo apt install dbus
sudo systemctl unmask systemd-logind
sudo systemctl start systemd-logind
```

MichaIng, 2024-09-26:
I'll however close this old issue, which was about the shutdown warnings only. In case create a new one about logind toggle. We do have the `dietpi.txt` option for this already, and `dietpi-software` toggles it whenever needed for a software title, and it is enabled for QEMU based VM and container systems, which require it for ACPI functionality. But of course there are other cases where one might want/need logind, and I am thinking about ways to make it somehow more transparent that it is disabled by default on DietPi, without bloating 99% of users with too much information. A menu option however is the minimum we should do.

closed 2024-09-26 by MichaIng