MichaIng/DietPi #6605: Prioritize Network Initialization on Kiosk

nowell-morris, 2023-09-07:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
When setting up a kiosk, using chromium as my boot-to target, if my chromium setup fails to load, then the whole OS fails to load.  My monitor will go into standby mode, and the network is not reachable.  I have to power off (non-graceful, power-cut) to mount the sdcard on a different computer.
- ...

#### Describe the solution you'd like:
I would like the system to be network accessible so that I can SSH in and see why my setup crashed. It would also allow me to gracefully shutdown the OS (being nicer to the file system).
- ...

#### Describe alternatives you've considered:
It would also be helpful if the /var/log contents remained somehow, so that I could check them on the sdcard.  But I would prefer SSH access to look at it live.
- ...

#### Additional context
A number of years ago, I troubleshot a similar issue with RedHat 6.  We prioritized network by changing the init script priority so that the network came up before our crashed daemon that came up afterwards.  I do not know how to accomplish this with systemd, nor with the order of execution in Dietpi.
- ...

MichaIng, 2023-09-07:
Many thanks for your report. With "boot-to target" you mean when you selected Chromium in `dietpi-autostart`?

If so, this does not affect network at all. It triggers the `dietpi-login` script (which is called after an auto-login on TTY1) to run Chromium in kiosk mode. If it fails, you'll just land back in the bash console, and SSH is not touched by this anyway.

For `/var/log` to be permanent, you can just uninstall `dietpi-ramlog`:
```sh
sudo dietpi-software uninstall 103
```
Most system logs are however in systemd journal which is by default stored to `/run` (a tmpfs). To make it persistent:
```sh
mkdir /var/log/journal
```
Alternatively use `rsyslog` to write system logs to legacy plain text failed in `/var/log` by selecting "full logging" in `dietpi-software`.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.