MichaIng/DietPi #6607: Unable to open serial communications

Kreeblah, 2023-09-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.21.1
- Distro version | Bullseye and Bookworm
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4B
- Power supply used | Official RPi PSU
- SD card used | SanDisk Extreme Pro

#### Additional Information (if applicable)
- Software title | `gpsd`
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
I had a bug report for my GPS-based stratum 0 time server at https://github.com/Kreeblah/DietPiTimeServer that indicated that the user couldn't get a GPS lock.  After looking into it, it seems that the current RPi images have changed something which prevents the steps I have from allowing `gpsd` from accessing the serial interface on the GPIO pins, because it never receives data from the GPS hat.

Currently, the only working image is the Bullseye image in the "old" directory on the DietPi site ([this one from 31 July](https://dietpi.com/downloads/images/old/DietPi_RPi-ARMv8-Bullseye.7z)).  The current Bullseye image and both Bookworm images exhibit the issue.

#### Expected behaviour
The same steps should be able to be used to obtain data from the GPS hat from the serial GPIO pins.

#### Actual behaviour
`gpsd` never seems to be able to obtain that data.

#### Extra details
I'm attaching a sample config.txt here from one of the non-working images (after configuration) in case there's something I'm missing, but as far as I know, this should have the mini UART device correctly configured.

[config.txt](https://github.com/MichaIng/DietPi/files/12564537/config.txt)

MichaIng, 2023-09-09:
Hmm, we did not recently change anything in this regards, or the content of `config.txt`, which should be identical on all images you tested. Also the kernel is identical on all of them and generally independent of the distro version. I.e., compare `config.txt` and `uname -a` on working and non-working systems and they should both match 100%.

Checking your `config.txt`, `dtparam=disable-bt` is wrong, it should be `dtoverlay=disable-bt`. However, Bluetooth is disabled OOTB, i.e. this line present, I wonder where it was going, as long as you did not manually enable Bluetooth?

The mini UART is enabled as of `enable_uart=1`, but with Bluetooth enabled, the PL011 UART is blocked. I guess this is causing the issue?

Kreeblah, 2023-09-09:
Ah, that was it.  Switching it from `dtparam` to `dtoverlay` works.  And, actually, thinking about it, that makes sense that it really should be `dtoverlay`, not `dtparam`.

`dtparam` worked fine before, with the older images (and I can replicate that), but now I'm confused as to why.  Regardless, I'll update my directions.  And, yep, I did notice when I was setting this up before that unless Bluetooth was explicitly disabled, I can't get serial output from the GPIO pins on this board.

So, I guess this can be chalked up to user error.

Though, I guess now that I'm thinking about it, I'd be curious as to whether you have any way of testing the mini UART on another device.  If you get the same results with other devices (which I'd expect), would it be possible to have `dietpi-config` automatically make the mini UART set `dtoverlay=disable-bt` and disable Bluetooth if it's enabled (with appropriate warnings, of course)?  And maybe add a warning if the mini UART is enabled and somebody tries to enable Bluetooth?  If so, I can open another issue as a feature request for that.