MichaIng/DietPi #6608: NanoPi R4S| Fix Ethernet LED triggers

d-me3, 2023-09-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bookworm`

- Kernel version | `uname -a`
`Linux r4s 6.1.50-current-rockchip64 #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`NanoPi R4S (aarch64)`

- Power supply used | (EG: 5V 1A RAVpower)
`PD 35W 5V 3A`

- SD card used | (EG: SanDisk ultra)
`SandDisk Ultra 32GB Class 10`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`dietpi-led_control`

- Was the software title installed freshly or updated/migrated?
 `installed freshly`

- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

- Bug report ID | `echo $G_HW_UUID`
`60ca6906-3c09-48a0-8133-a6c5b50951ab`

#### Steps to reproduce
1. Boot fresh install, LAN and WAN leds off
2. Cable connected to LAN port, ethernet connection working OK, LAN led remains off:
```
root@r4s:~# cat /sys/class/net/eth0/operstate
up
root@r4s:~# ip r
default via 192.168.1.1 dev eth0
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.174
root@r4s:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        master-slave cfg: preferred slave
        master-slave status: slave
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: external
        MDI-X: Unknown
        Supports Wake-on: pumbg
        Wake-on: d
        Link detected: yes

```

3. Run `dietpi-led_control` - all LEDs show "none":
```
Please select an LED to configure its trigger:                                                                       │
      │                                                                                                                      │
      │                                             Reset Reset all LED triggers                                             │
      │                                             1     : green:lan [none]                                                 │
      │                                             2     : green:wan [none]                                                 │
      │                                             3     : red:power [none]
```

4. Change `green:lan` and `green:wan` to `netdev`
```
Please select an LED to configure its trigger:                                                                       │
      │                                                                                                                      │
      │                                             Reset Reset all LED triggers                                             │
      │                                             1     : green:lan [netdev]                                               │
      │                                             2     : green:wan [netdev]                                               │
      │                                             3     : red:power [none]
```

5. Reboot
6. LAN and WAN Leds still off (despite of having an ethernet cable connected to LAN port, working and link up)
7. Change `green:lan` and `green:wan` to `(r8169-0-100:00:link)`
```
      │                                   r8169-0-100:00:link                             ▒                                  │
      │                                   r8169-0-100:00:1Gbps                            ▒                                  │
      │                                   r8169-0-100:00:100Mbps                          ▒                                  │            │                                   r8169-0-100:00:10Mbps
```

8. Reboot
9. LAN and WAN Leds still off (despite of having an ethernet cable connected to LAN port, working and link up)

#### Expected behaviour
- By changing LEDs to "netdev" or "(r8169-0-100:00:link)" they should light up/blink upon network activity or ethernet link status

#### Actual behaviour
- LEDs LAN and WAN are not responding to network events.
#### Extra details

- ...

MichaIng, 2023-09-11:
Fixed for next release already: https://github.com/MichaIng/DietPi/commit/bdde5bd

d-me3, 2023-09-12:
Thank you for the prompt reply. I just tested with dev build 8.22 (below) and it is working now. Closing this issue.

`root@r4s:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='dev'
G_GITOWNER='MichaIng'`

closed 2023-09-12 by d-me3