MichaIng/DietPi #6609: Rock Pi 4 more unstable after recent updates

andreagdipaolo, 2023-09-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.1.50-current-rockchip64 #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`
- SBC model | `ROCK 4 (aarch64)`
- Power supply used | 65W PD - Seems to be running at 20V.
- SD card used | None: booting from NVME via armbian SPI bootloader

#### Steps to reproduce

1. Power on the device
2. Wait until it crashes

#### Expected behaviour

- The device should be running fine until a reboot or a poweroff

#### Actual behaviour

- The device freezes/crashes. The blue LED set to heartbeat mode is completely off or always on, it varies from time to time.
- After failing, the cpu seems to be warmer than idle, like if it's under load.
- TTL interface (usually off but enabled for investigating these crashes) and SSH stop working.
- Ethernet LEDs keep blinking.

#### Extra details

- I'm running the system with the 2 GHz overclock overlay. It happen with the stock frequency too, so I assume it's not its fault.
- Schedutil governor, cpu is cooled via heatsink and fan.
- The system has benne behaving relatively well in the two years that I owned it, and I've been always running DietPi on it.
Previously it was run via an SD, later via a USB key and since one month via NVME.
- The system has crashed other times but before the last updates it was happening something like once every 1-2 months. Sometimes it did crash for ages. It has not crashed three times in the last two days, after upgrading the `armbian-firmware` package.
- I tried logging the crashes but I have been unsuccessful. I don't know how to investigate it better.
- I've tried different SSDs and different power supplies.

Thanks!

MichaIng, 2023-09-11:
Generally, if the system does crash already, even just rarely, I'd check for the reason of this. If it is power or heat related, quite possible that the kernel upgrade raises the chance for this to happen, while the kernel itself does not have an issue. Check via `dmesg -l 0,1,2,3` whether you see any kernel errors related to power or voltage. Via `dietpi-config` > Tools you can do a stress test to observe CPU temperature development and test system stability at high loads.

As you say you use the 2 GHz overlay: With Linux 5.15, this overlay was broken and had no effect, hence you were running with 1.8 GHz until now. With Linux 6.1 it has been fixed, so you are now running at 2 GHz. You can check it via `cpu` command. This strengthens the theory that CPU temperature or voltage are the issue.

The USB-C power socket supports 5V/3A or PD with 9V and 12V. Your PSU seems to be more than sufficient, but there are still quality differences, especially when it is about holding a voltage reliably in narrow borders on variable power draw. Quite many common but cheap PD chargers do not do this well. They provide high power for charging a battery quickly, but do not hold the voltage well, also because it is simply not sufficient for arching a battery and current changes only very slowly during charging, other than when an SBC changes switches from idle to full load.

Also for investigating the issue, if SSH crashes, can you still access via keyboard and monitor or serial console?

andreagdipaolo, 2023-09-11:
Thanks for the detailed reply. If you believe we should move to the forum I'll create a thread there.

> Check via dmesg -l 0,1,2,3 whether you see any kernel errors related to power or voltage.

Since the last boot, nothing comes up.

> Via dietpi-config > Tools you can do a stress test to observe CPU temperature development and test system stability at high loads.

I will check that. I had a fan for long time, I removed it and added it back some weeks ago, but even without it the CPU has never exceeded 58 degrees.
I will buy a more powerful fan, right now I added a 12V fan to the 5V pins so it runs quite slow.
Do you happen to know at what temperature the system shuts down? Shouldn't it be able to throttle?

> As you say you use the 2 GHz overlay: With Linux 5.15, this overlay was broken and had no effect, hence you were running with 1.8 GHz until now.

That's so strange, I am quite sure seeing 1.5 and 2.0 GHz every time I checked the temperature via the `cpu` command. But at the same time I believe you. For how long has DietPi been running on Linux 5.15? What came before it and when?

> The USB-C power socket supports 5V/3A or PD with 9V and 12V

Yep, I've read the Radxa schematics and they say it should be 12V max. But apparently the onboard chip and my power supply negotiate 20V. I checked via a multimeter on the board and it gets 20V on the Vcc line. [Radxa schematics](https://dl.radxa.com/rockpi4/docs/hw/rockpi4/rockpi4_v13_sch_20181112.pdf) describe a voltage tolerance from 5 to 24V.
[image]

> Quite many common but cheap PD chargers do not do this well. They provide high power for charging a battery quickly, but do not hold the voltage well, also because it is simply not sufficient for [...]

I have a USB-C voltage and power tester that I could connect to my PC for getting graphs. Maybe I could check that even if my Pi sits a bit far from my desk. However I believe my power adapter to be fairly good, but I can confirm that my system is quite power hungry because I run an ADS-B receiver that gets quite warm. I will publish the results during a stress test too.

> Also for investigating the issue, if SSH crashes, can you still access via keyboard and monitor or serial console?

As I said the serial interface seems to be dead after the system crashes. I don't remember what happened after connecting an HDMI cable. If it happens again I will try to check.

Thanks!

MichaIng, 2023-09-12:
This was the Armbian kernel? The bug was reported on May 2022 first where I could replicate it repeatedly after every kernel upgrade shipped with the Armbian APT repo, latest with 5.15.93, AFAIK: #4596
And it was reported at the Armbian forum for ROCKPro64 and Orange Pi 4 at least.

However, probably it did work on some SBCs or some intermediate kernel versions. Regardless, when it is about a crashing system, ruling out any overclocking make sense in any case, regardless whether it worked before or not.

Might the 20V PD even be the problem? Even if it has the tolerance and somehow negotiates it, probably transforming from the higher voltage leads to a higher variability on the PCB-internal voltages, or the PSU is not able to keep 20V as stable as lower voltages? I do not know much about the reliability of PD and these voltage negotiations (Radxa's official PSU is also 30W PD up to 12V), but when it is about SBC stability, at least the PSU is among the first things to take into view.

I know an accelerated heartbeat as max/high CPU usage, but is differs among individual SBCs with same kernel+SoC as well. E.g. the NanoPi R4S has no heartbeat at all by default, just an always-on PWR LED and an activity SYS led. But probably I can replicate it when manually switching one trigger to heartbeat and running a stress test with this.

MichaIng, 2023-09-12:
Jep, indeed the heartbeat trigger increases the frequency on higher CPU loads. I guess the 1 minute average is what affects it. It somehow makes sense, as human heartbeats also increase on "load" 😉. So then it indeed seems that you crashes come along with high CPU load. But of course it could be also RAM or disk usage, happen concurrently, or a faulty process or something.

One idea to get more info could be to have a `journalctl -f` running on some serial console or SSH session. This prints all system/service logs along with kernel errors when they happen. So as long as the crash is not too abrupt, preventing any further logs about it, this might give a hint.

andreagdipaolo, 2023-09-12:
> This was the Armbian kernel? The bug was reported on May 2022 first where I could replicate it repeatedly after every kernel upgrade shipped with the Armbian APT repo, latest with 5.15.93, AFAIK: https://github.com/MichaIng/DietPi/issues/4596

It was DietPi without any kernel modification, I don't remember which one was shipped with it.

> Might the 20V PD even be the problem?

I will try recording the power line while the system is stressed. But I used more than one power supplies.

> Jep, indeed the heartbeat trigger increases the frequency on higher CPU loads.

That makes sense. Even if I didn't remember it varying previously. It was something like a fixed one second between a double beat.

One idea to get more info could be to have a journalctl -f running on some serial console or SSH session.

I tried it in the past but it did not show anything. I will try it again. It's easy via SSH but it's a bit difficult via serial.

I will update you as soon as I have new data.

andreagdipaolo, 2023-09-15:
I'm doing the stress tests while monitoring the power draw.
I chose 5 minutes ad the duration because I believe it's a reasonable amount of time. I never stress my board at 100% for that long. It never went above 55°C

However I noticed I was powering it via a A-C cable, not able to do power delivery. It was running in QuickCharge mode.
In QC mode it can only deliver 20W max... I've been running in QC all the time, damn. I checked 20V when i was at my desk via a C-C cable and I assumed it was always like that.

I will use these chargers for the tests:

**Charger A**
PD3.0 / PPS output: 5V / 3A, 9V / 3A, 12V / 3A, 15V / 3A, 20V / 3.25A (65W)
QC3.0 output: 3.6-6.5V / 3A, 6.5-9V / 2A, 9V-12V / 1.5A (20W)

**Charger B**
Output port: QC3.0:DC5V/3A,9V/2A,12V/1.5A
PD:DC 5V/3A,9V/2.22A,12V/1.67A(20W)

Tests below:

Charger A via USB-A
[image]

Charger A via USB-C
[image]

Charger B via USB-A
[image]

Charger B via USB-C
Dead? Outputs 1V and obviously the SBC doesn't like it.

I will try to run it always in PD because with QuickCharge I am almost at its maximum power draw.
I will install a better fan too.

At this point I am convinced it's a power issue but I will monitor it over time and I will keep recording logs.
I don't know if you want to close this issue or keep it open while waiting for a happy ending.
Thanks!

MichaIng, 2023-09-16:
At least you can see that 5V 3A world be actually sufficient, if the cable does not drop too much voltage then. That is in line with the RK3588 where we never observed above 13W, at least noy in headless mode.

Did you have the stability issues with charger A or charger B?

andreagdipaolo, 2023-09-16:
> At least you can see that 5V 3A world be actually sufficient

I don't like the idea of charging with low volts and high amps, I prefer the opposite.

> Did you have the stability issues with charger A or charger B?

I believe with both, but in A to C mode, so in QC.
I will try to run it in PD mode and see how it behaves.

MichaIng, 2023-09-19:
> I don't like the idea of powering it with low volts and high amps, I prefer the opposite.

Generally yes, for higher efficiency at least. It is just the question whether from PSU and/or SBC side all voltages lead to the same stability.

How do you know that it runs in QC mode instead of PD mode (as long as voltage is not above 12V, of course)? And does it really vary the voltage then? For charging this seems clever, but for powering a device of course it sounds problematic 😄. But I have not much knowledge about all these USB powering/charging modes.

andreagdipaolo, 2023-09-19:
> Generally yes, for higher efficiency at least. It is just the question whether from PSU and/or SBC side all voltages lead to the same stability.

A the moment, since I switched to C-C PD it never crashed. Fingers crossed...

> How do you know that it runs in QC mode instead of PD mode (as long as voltage is not above 12V, of course)?

Because as you can see in my graphs when I use C to C it goes to 20V.
And also I have a USB-C tester (Ruideng TC66) which can detect protocols.
Moreover, PD is not available using USB-A cables, by standard.

> And does it really vary the voltage then?

The voltage starts at 5V and after a few milliseconds it switches to the best combo accepted by the two devices. 
It should not change overtime until you disconnect and connect again the cable.

> For charging this seems clever

As far as I know phones won't ask for different voltages across the charging process over PD. My Pixel 6 goes to 9V and stays there.

andreagdipaolo, 2023-09-26:
I've been finally able to track a kernel log of a crash.
Heartbeat stopped beating, blue LED always on.

```
[21179.609247] systemd-journald[366]: Data hash table of /run/log/journal/ce64bb42aa984804a5042a690ee59bf3/system.journal has a fill level at 75.0 (6600 of 8796 items, 5066752 file size, 767 bytes per hash table item), suggesting rotation.
[21179.609296] systemd-journald[366]: /run/log/journal/ce64bb42aa984804a5042a690ee59bf3/system.journal: Journal header limits reached or header out-of-date, rotating.
[21377.883559] Unable to handle kernel paging request at virtual address ffff8000edfac02f
[21377.883630] Mem abort info:
[21377.883651]   ESR = 0x0000000096000005
[21377.883681]   EC = 0x25: DABT (current EL), IL = 32 bits
[21377.883717]   SET = 0, FnV = 0
[21377.883743]   EA = 0, S1PTW = 0
[21377.883769]   FSC = 0x05: level 1 translation fault
[21377.883799] Data abort info:
[21377.883821]   ISV = 0, ISS = 0x00000005
[21377.883847]   CM = 0, WnR = 0
[21377.883873] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000003842000
```

MichaIng, 2025-01-09:
Is this still an issue with recent kernel?

andreagdipaolo, 2025-01-24:
> Is this still an issue with recent kernel?

Hi, thanks for asking. For a while it has been crashing something like once a month. 
In the last few weeks, it's more like once a week. I powered off the cooling fan, but the temperature is absolutely in range, stable at around 40° celsius. With the fan it was around 30. I have the official heavy heatsink. I don't believe this is related to the crashes increase.

Some days ago, it also had a strange crash: the cli was reachable by ssh, but commands were not working and my docker containers froze, but the system was reporting being online.
The last crash happened some hours ago.
I am running it now from a 45W power supply. It only has a Zigbee usb key and a ADSB usb key, nothing power hungry.
The system has an average load of 15%. 

I was planning to migrate to che CM3588 but I would use this ROCK4 in a remote location as a backup server, so It would be nice not crashing at all. I accept any suggestion and I am willing to try and troubleshoot if someone could help me along the way.
Someone suggested to me the use of a watchdog to reboot the board once it crashes, but It doesn't solve the problem and i don't even know if this could work at all.

Thanks in advance

MichaIng, 2025-01-30:
Hmm, this all sounds like some kind of overload, maybe some OOM process kills or so. The previous logs do not reveal much.

First of all, about the Docker logs. In `/etc/docker/daemon.json` and `/etc/containerd/config.toml` the log level can be defined, and the `log-driver` setting also allows to move them away from the journal completely: https://docs.docker.com/engine/logging/configure/

But it can be overridden for each container via Docker Compose YAML or `docker run` arguments.

Probably enabling persistent system logs reveals more:
```sh
dietpi-software uninstall 103 # uninstalls RAM log
mkdir -p /var/log/journal # enables persistent journalctl logs
reboot # needed to disable the tmpfs at /var/log
```

andreagdipaolo, 2025-02-02:
> First of all, about the Docker logs. In `/etc/docker/daemon.json` 

```
{
    "data-root": "/mnt/dietpi_userdata/docker-data",
    "log-driver": "journald",
    "log-level": "warn",
    "debug": false
}
```

> and `/etc/containerd/config.toml` the log level can be defined

```
disabled_plugins = ["cri"]

#root = "/var/lib/containerd"
#state = "/run/containerd"
#subreaper = true
#oom_score = 0

#[grpc]
#  address = "/run/containerd/containerd.sock"
#  uid = 0
#  gid = 0

[debug]
#  address = "/run/containerd/debug.sock"
#  uid = 0
#  gid = 0
  level = "warn"
```

> and the `log-driver` setting also allows to move them away from the journal completely: https://docs.docker.com/engine/logging/configure/
> 
> But it can be overridden for each container via Docker Compose YAML or `docker run` arguments.

I mainly run Home Assistant and many containers from ADSB.im
I don'know if it's wise to touch anything there.

> Probably enabling persistent system logs reveals more:
> 
> dietpi-software uninstall 103 # uninstalls RAM log
> mkdir -p /var/log/journal # enables persistent journalctl logs
> reboot # needed to disable the tmpfs at /var/log

Done. When it crashes again what should I do/check?

MichaIng, 2025-04-19:
Sorry for the late reply. Check `journalctl -e` after a crash. The output contains delimiter lines for each boot session, so it ahould be possible to identify the last logs or the previous boot session that belong to the crash, by scrolling up to the last boot session delimiter line.

MichaIng, 2025-05-31:
Closing this for now. Feel free to reopen if the issue persists.

closed 2025-05-31 by MichaIng