MichaIng/DietPi #6611: Rock Pi S - USB C OTG as Host

starsoccer, 2023-09-11:
I am trying to figure out how to use my rock pi s with the USB C OTG port as host instead of OTG. This forum post more or less summarizes what I am trying to do but does not seem to be work/be applicable for dietpi, https://forum.radxa.com/t/rock-pi-s-otg-in-host-mode/3572

Version:
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

MichaIng, 2023-09-11:
OTG should btw switch between host and gadget mode automatically. However, sometimes this does not work. Can you test this overlay:
```sh
G_CONFIG_INJECT 'overlays=' 'overlays=dwc3-0-host' /boot/dietpiEnv.txt
reboot
```

starsoccer, 2023-09-11:
Just tried that command and confirmed it was set in `dietpiEnv.txt` but still no luck

[image]

MichaIng, 2023-09-11:
Ah, this overlay has a different prefix:
```sh
G_CONFIG_INJECT 'overlay_prefix=' 'overlay_prefix=rockchip' /boot/dietpiEnv.txt
reboot
```

starsoccer, 2023-09-11:
Sadly no dice either. Any other suggestions?

For context, I am running `lsusb` to determine if there are any new devices. I am expecting to see 2 Serial

[image]

starsoccer, 2023-09-11:
Some more info in case its useful
[image]

starsoccer, 2023-09-13:
@MichaIng Any guidance you can provide on things to try? Or any logs or information I can provide to further debug?

MichaIng, 2025-07-18:
@starsoccer has been a while, but maybe you are still working with your ROCK Pi S?

I created a new kernel build with overlays for USB-C OTG and host modes added: https://github.com/MichaIng/build/commit/f77b55d
_The WiFi driver switch is something I was testing in parallel, reverting the switch to mainline driver, which does not perform so well._

Generally switching the USB mode works from device tree POV, but I have no good way to test it right now, lacking POE or what. If you want to give it a try, I would be happy:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-rockchip64.deb
dpkg -i linux-{image,dtb}-current-rockchip64.deb
sed -i '/^overlays=/s/$/ usb-c-host/' /boot/dietpiEnv.txt
reboot
```
By default, it is in `peripheral` mode, which explains pretty well why it cannot work as host at all. So I added the `usb-c-otg` overlay as well, for actual OTG mode.