MichaIng/DietPi #6614: php 7.4 fails to install after upgrade to Bookworm

InnovoDeveloper, 2023-09-12:
### Creating a bug report/issue

#### Required Information
 - DietPi version: v8.21.1 (MichaIng/master) | HW_MODEL: 60 | HW_ARCH: 2 | DISTRO: 6                                 │
            │  - Image creator: DietPi Core Team                                                                                   │
            │  - Pre-image: Armbian             

#### Additional Information (if applicable)
- Software title | PHP
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `c57ae1b3-4ea0-4cb6-98be-cae15a392ca8`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- `... APT install php7.4-fpm php7.4-apcu php7.4-curl php7.4-gd php7.4-mbstring php7.4-xml php7.4-zip php7.4-sqlite3        │
            │  - Command: apt-get -y --allow-change-held-packages install php7.4-fpm php7.4-apcu php7.4-curl php7.4-gd             │
            │ php7.4-mbstring php7.4-xml php7.4-zip php7.4-sqlite3                                                                 │
            │  - Exit code: 100                                                                                                    │
            │                                                                                    │
            │  - Error log:                                                                                                        │
            │ Reading package lists...                                                                                             │
            │ Building dependency tree...                                                                                          │
            │ Reading state information...                                                                                         │
            │ Some packages could not be installed. This may mean that you have                                                    │
            │ requested an impossible situation or if you are using the unstable                                                   │
            │ distribution that some required packages have not yet been created                                                   │
            │ or been moved out of Incoming.                                                                                       │
            │ The following information may help to resolve the situation:                                                         │
            │                                                                                                                      │
            │ The following packages have unmet dependencies:                                                                      │
            │  libgd3 : Depends: libavif9 (>= 0.8.4) but it is not installable                                                     │
            │  php7.4-common : Depends: libffi7 (>= 3.3~20180313) but it is not installable                                        │
            │                  Depends: libssl1.1 (>= 1.1.0) but it is not installable                                             │
            │  php7.4-fpm : Depends: php7.4-cli but it is not going to be installed                                                │
            │               Depends: libssl1.1 (>= 1.1.0) but it is not installable                                                │
            │ E: Unable to correct problems, you have held broken packages.       `

MichaIng, 2023-09-12:
On Bookworm PHP 8.2 should be installed.

Are you probably affected by this? https://github.com/MichaIng/DietPi/issues/6227#issuecomment-1713688577

The Armbian `base-files` package has a bad version string, which prevents it from being upgraded when upgrading the distro to Bookworm. Our scripts then see `/etc/os-release` and `/etc/debian_version` still reporting Bullseye, trying to install PHP 7.4, which is not possible on Bookworm.

InnovoDeveloper, 2023-09-12:
Yes, this seems to be what's happening.  I did notice that the /boot/.hw_version file also still says bullseye
cat .hw_model
G_HW_MODEL=60
G_HW_MODEL_NAME='NanoPi NEO (armv7l)'
G_HW_ARCH=2
G_HW_ARCH_NAME='armv7l'
G_HW_CPUID=1
G_HW_CPU_CORES=4
G_DISTRO=6
G_DISTRO_NAME='bullseye'

I think update the G_DISTRO=7 and G_DISTRO_NAME='bookworm' might solve the issue?

Joulinar, 2023-09-12:
Following should do it

```
apt install base-files=23.8.1-bookworm
```

InnovoDeveloper, 2023-09-12:
I tried, but didn't change the /boot/dietpi/.hw_model.  Maybe I did something wrong.   

If I am updating a new system from bullseye to bookworm that already has php7.4 and home assistant, what's the best way to upgrade?  I usually use your script, but that seems to cause some conflicts.

Joulinar, 2023-09-12:
PHP should be upgraded during Destro upgrade to Bookworm. A reboot should recreate the .hw_model file

InnovoDeveloper, 2023-09-12:
I have a system on bullseye, I'll run the script again and report the issue.  This is how my issues reported startd.

MichaIng, 2023-09-12:
`/boot/dietpi/.hw_model` is (re-)generated at boot and also just reads `/etc/debian_release` 😄. After the package upgrade you can force it being regenerated via:
```sh
/boot/dietpi/func/dietpi-obtain_hw_model
```
Then `dietpi-software` sees the correct distro version.

Ruud68, 2023-09-14:
Was facing same issue (Home Assistant not working / server not responding, php files displaying as plain text instead of being extecuted)
So did the apt install base-files=23.8.1-bookworm, and rebooted.

Next I had to (In my LAMP setup) do: dietpi-software reinstall 76

That reinstalled AND enabled the php 8.2 enviroment.

php, phymysql, homeassistant etc. is now working again.

Thought I post here as only doing the suggested step to reinstall the base files was for my setup not doing the complete trick.

Joulinar, 2023-09-14:
Yes a software reinstall is necessary for PHP8.2 to enable all modules. Usually this is done by the Bookworm script but actually fault for Armbian images due the package naming. We will handle this in our script soon.

MichaIng, 2023-09-17:
The issue with the `base-files` package will be solved with next DietPi update: https://github.com/MichaIng/DietPi/commit/b38163c

Generally we are now blocking the installation of any package from their repo, aside of kernel, dtb, headers, bootloader and firmware.

Ruud68, 2023-09-18:
Good the here and thanksf or following up.
Just to be 'complete' in my upgrade: is there any other thing needed other then reinstalling LAMP (in my case)? Is PHP 8.2 the only affected module or are there more things that need attention?

MichaIng, 2023-09-19:
As long as everything works, I don't think so. The Python version is implicitly correctly installed and in all other cases the installation would either fail, e.g. because of missing dependencies, or the service start fail because of an incompatible config file entry.

closed 2023-09-22 by MichaIng