MichaIng/DietPi #6616: Error running dietpi-update on Odroid C1

jncl, 2023-09-14:
#### Details:
- Date           | Thu Sep 14 17:07:10 BST 2023
- Bug report     | 7380ec5a-7ef2-4cc4-83a2-47def438d769
- DietPi version | v8.18.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | Odroid C1 (armv7l) (ID=10)
- Kernel version | `Linux boinc-node5 5.4.75-meson #20.08.22 SMP Sun Nov 8 13:48:56 CET 2020 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce

1. run 'sudo dietpi-update 1'

#### Expected behaviour

- Successful Update

#### Actual behaviour

- Error encountered

#### Extra details

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up initramfs-tools (0.140) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.140) ...
update-initramfs: Generating /boot/initrd.img-5.4.75-meson
cp: '/boot/uInitrd-5.4.75-meson' and '/boot/uInitrd' are the same file
Image Name:   uInitrd
Created:      Thu Sep 14 17:07:08 2023
Image Type:   AArch64 Linux RAMDisk Image (uncompressed)
Data Size:    7612418 Bytes = 7434.00 KiB = 7.26 MiB
Load Address: 00000000
Entry Point:  00000000
cp: '/boot/uInitrd-5.4.75-meson' and '/boot/uInitrd' are the same file
E: /etc/initramfs-tools/hooks/uInitrd failed with return 1.
update-initramfs: failed for /boot/initrd.img-5.4.75-meson with 1.
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2023-09-14:
Looks like a bug or conflict in the initramfs hook script of Armbian. We actually create an own script for this task for our new images. Can you check the following:
```sh
dpkg -S /etc/initramfs-tools/hooks/uInitrd
dpkg -l | grep linux
```
Also there should be a much newer Linux version available in the meantime. However, actually we had a case in the forum where it broke the Odroid C1 from booting. Since Armbian does not support this board anymore, it is thinkable, so probably you should stay with this older kernel version. Hence the second command to check which package it actually is.

jncl, 2023-09-15:
Here's the output from those commands:

--> ls -l /boot
total 26472
-rwxr-xr-x 1 root root       3734 Mar  7  2023 boot.ini_c1
-rwxr-xr-x 1 root root  209869 Nov  8  2020 config-5.4.75-meson
drwxr-xr-x 4 root root     4096 Jun 19 15:44 dietpi
-rwxr-xr-x 1 root root      10130 May 10  2020 dietpi-README.md
-rwxr-xr-x 1 root root     13089 Apr 28 16:24 dietpi.txt
-rwxr-xr-x 1 root root       2955 May 18  2020 dietpi-wifi.txt
-rw-r--r-- 1 root root  7612418 May 13 17:28 initrd.img-5.4.75-meson
-rwxr-xr-x 1 root root      17758 Mar  7  2023 meson8b_odroidc1.dtb
-rwxr-xr-x 1 root root 3583128 Nov  8  2020 System.map-5.4.75-meson
-rwxr-xr-x 1 root root 8016448 Feb  8  2021 uImage
lrwxrwxrwx 1 root root          20 May 13 17:28 uInitrd -> uInitrd-5.4.75-meson
-rw-r--r-- 1 root root  7612482 Sep 14 17:52 uInitrd-5.4.75-meson

--> dpkg -S /etc/initramfs-tools/hooks/uInitrd
dpkg-query: no path found matching pattern /etc/initramfs-tools/hooks/uInitrd

--> dpkg -l | grep linux
ii  console-setup-linux                         1.205                           all          Linux specific part of console-setup
ii  libselinux1:armhf                               3.1-3                           armhf        SELinux runtime shared libraries
ii  linux-base                                          4.6                              all          Linux image base package
ii  linux-buster-root-legacy-odroidc1 21.02.3                        armhf        Armbian tweaks for buster on odroidc1 (legacy branch)
ii  linux-dtb-legacy-meson                  20.08.22                     armhf        Linux DTB, version 5.4.75-meson
ii  linux-image-legacy-meson             20.08.22                     armhf        Linux kernel, version 5.4.75-meson
ii  linux-u-boot-odroidc1-legacy        21.02.3                        armhf        Uboot loader 2011.03
ii  util-linux                                           2.36.1-8+deb11u1       armhf        miscellaneous system utilities

If you need anything else, please let me know.

Vaelatern, 2023-09-22:
I wrote a preferences file

```
# cat /etc/apt/preferences.d/linux
Package: linux-dtb-current-meson
Pin: version 23.02.2
Pin-Priority: 1000

Package: linux-image-current-meson
Pin: version 23.02.2
Pin-Priority: 1000

Package: linux-u-boot-odroidc1-current
Pin: version 21.02.3
Pin-Priority: 1000
```

MichaIng, 2023-09-22:
@jncl 
Sorry for getting back to you so late. You are using a very old image with an old kernel. As the current latest kernel seems to not support the board anymore, this is however not bad, until we found a working recent kernel version. But the initramfs hook script on your system seems to be not compatible with other post config scripts and the partition layout, not sure how this happened. The purpose of the script is to generate a U-Boot version of the initramfs generated by initramfs-tools, and to then, on a single partition image, link it to a generic file name. It however tries to copy it, which fails if there is a symlink already. Replace it with a modern variant of this script we ship with out images nowadays:
```sh
cat << '_EOF_' > /etc/initramfs/post-update.d/99-dietpi-uboot
#!/bin/dash
echo 'update-initramfs: Converting to U-Boot format'
mkimage -A arm -O linux -T ramdisk -C gzip -n uInitrd -d "$2" "/boot/uInitrd-$1"
ln -sfv "uInitrd-$1" /boot/uInitrd || mv -v "/boot/uInitrd-$1" /boot/uInitrd
exit 0
_EOF_
rm /etc/initramfs-tools/hooks/uInitrd
dpkg --configure -a
```
Copy&paste this at best at once into an SSH terminal, as the first block is a "here document" with a single command over several lines. This new script does the same as the old one, but instead of trying to copy `/boot/uInitrd-5.4.75-meson` to `/boot/uInitrd`, it first tries to update the latter as symlink (which is the correct thing in your case), and only if it fails (if `/boot` is located on a FAT partition), moves the file.

MichaIng, 2023-09-22:
@Vaelatern 
Note that OP uses the "legacy" kernel, hence is not affected by the issue you likely ran into with the "current" kernel. If you find time, could you test the "edge" kernel with Linux 6.4.13?
```sh
apt install linux-{image,dtb}-edge-meson
```
The bootloader btw does not need to be pinned. The last package is from 2021 and nowadays it does not build anymore + Armbian dropped support a while ago, so there won't be any new update anyway. And even if, that would be actually a good thing, as it is tailored for exactly this board and hence, other then the kernel, could make things better, e.g. solve the cold reboot issue.

APT preferences btw support multiple packages per block and wildcards 🙂:
```
Package: linux-image-current-meson linux-dtb-current-meson linux-headers-current-meson
Pin: version 23.02.2
Pin-Priority: 1000
```

jncl, 2023-09-22:
That worked. Thank you very much for your help resolving this.

closed 2023-09-22 by jncl

Vaelatern, 2023-09-24:
Sorry, this was a means to an end, I don't expect to test the odroid update for at least a few weeks, if ever (I'd want to image the SD card first so I could restore working functionality quickly).

MichaIng, 2023-09-24:
@Vaelatern 
Jep, no problem. The DietPi release from today pins the C1 "current" and "edge" kernel package versions to 23.02.2 as suggested by you and in case enforces a downgrade (if one upgraded them already but did not reboot yet). Then it sets the packages on hold, to prevent an upgrade even after 23.02.2 package versions are gone from the repo (the pins then loose their effect).

The "edge" kernel btw is Linux 6.1.12 and has been tested successfully by someone on our forum, just in case a newer kernel is needed at some point. So a change since February in mainline Linux backported down to v5.15 or one in Armbian's build system since v23.02.2 is causing the issue.