MichaIng/DietPi #6618: Error when running dietpi-update on OdroidC2

StormCh, 2023-09-14:
#### Details:
- Date           | Thu Sep 14 20:32:02 BST 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | `Linux OdroidC2 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

run dietpi-update

#### Expected behaviour:

update

#### Actual behaviour:

error occures, also if using --fix-missing

#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  base-files
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 53.0 kB of archives.
After this operation, 71.7 kB disk space will be freed.
Err:1 https://fi.mirror.armbian.de/apt bullseye/main arm64 base-files arm64 23.11.0-trunk.86-bullseye
  404  Not Found [IP: 65.21.120.247 443]
E: Failed to fetch https://fi.mirror.armbian.de/apt/pool/main/b/base-files/base-files_23.11.0-trunk.86-bullseye_arm64__1-bullseye-1armbian1-Bbecb-R55fa.deb  404  Not Found [IP: 65.21.120.247 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2023-09-14:
Looks like you got redirected to a faulty Armbian package server mirror

> Failed to fetch https://fi.mirror.armbian.de/

You could wait and try again later.

MichaIng, 2023-09-14:
Jep, currently there are only 2 mirrors for the Armbian repo, one in Helsinki/Finland, which is out of sync, and one in St. Petersburg/Russia, which is extremely slow. I hope tomorrow the first is in sync, i.e. has the missing version 23.11.0 packages files available.

MichaIng, 2025-05-31:
Closing this issue. In the meantime the Armbian repo should have been removed, using our own instead.

closed 2025-05-31 by MichaIng