MichaIng/DietPi #6620: libcamera-apps can't be installed on RPI B

Nama, 2023-09-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bookworm 1
```
- Kernel version | `uname -a`
```
Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi B (armv6l)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
Noname 1A
```
- SD card used | (EG: SanDisk ultra)
```
Some cheap and old 1GB for testing, atm
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`libcamera-apps`
- Was the software title installed freshly or updated/migrated?
Tried to freshly install
- Can this issue be replicated on a fresh installation of DietPi?
It happened to me on a fresh install. My very first command was `apt install libcamera-apps`

#### Steps to reproduce

1. Flash DietPi on SD card with Rufus
2. Do first-time-setup
3. `apt install libcamera-apps`

#### Expected behaviour

- Should install the package

#### Actual behaviour

> root@DietPi:~# apt install libcamera-apps
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  libcamera-apps : Depends: libavcodec58 (>= 7:4.2)
>                   Depends: libavdevice58 (>= 7:4.0) but it is not going to be installed
>                   Depends: libavformat58 (>= 7:4.2) but it is not going to be installed
>                   Depends: libtiff5 (>= 4.0.3) but it is not installable
> E: Unable to correct problems, you have held broken packages.

#### Extra details

I tracked it down to the package libsrt1.4-gnutls, which doesn't seem to be in the mirror?

> root@DietPi:~# apt install libsrt1.4-gnutls
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Package libsrt1.4-gnutls is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'libsrt1.4-gnutls' has no installation candidate

Nama, 2023-09-16:
Same with `libcamera-apps-lite`.
`libtiff5` is missing for this one.

Joulinar, 2023-09-17:
Looks like this one is not available on Raspbian Bookworm repository. Any ideas @MichaIng ??

MichaIng, 2023-09-17:
Those are not shipped by Raspbian but by the RPi repo, which still has no (filled) Bookworm suite: https://archive.raspberrypi.org/debian/dists/bookworm/main/binary-armhf/

So yes, until this is present, respectively the `Packages` file in this link is not empty, and `/etc/apt/sources.list.d/raspi.list` then switched from Bullseye to Bookworm, those packages cannot be installed.

Instead, you could compile them from source: https://github.com/raspberrypi/libcamera-apps

Nama, 2023-09-18:
I flashed the DietPi Bullseye image now, so this is not really an issue to me anymore.
Since this is RPiOS related, this can be closed?

closed 2023-09-18 by Nama

MichaIng, 2023-09-19:
Jep, it will solve itself once RPi starts to build Bookworm packages. I'm regularly having a look at that, and once it is the case, RPi DietPi system will have it applied automatically on DietPi update afterwards.