MichaIng/DietPi #6621: ADS-B Feeder: Link to local PiAware page seems to be wrong

StephanStS, 2023-09-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='dev'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

`bookworm`

- Kernel version | `uname -a`

`6.1.47-current-sunxi #1 SMP Wed Aug 23 15:52:42 UTC 2023 armv7l GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

`NanoPi M1 Plus (armv7l)`

#### Additional Information (if applicable)
- Software title | ADS-B Feeder
- Was the software title installed freshly or updated/migrated? | Fresh installation
- Can this issue be replicated on a fresh installation of DietPi? | yes

#### Steps to reproduce
1. Installation of ADS-B Feeder (`#141`)
2. Go to aggregators setup page and browse to `Submit your data to FlightAware`
3. Activate Flightaware by clicking the check box
4. Klick on the link `Piaware page` given in the text

[image]

#### Expected behaviour
6. Link opens new tab

#### Actual behaviour
4. Link opens new tab but leads to a load error, i.e. it does not open the PiAware page (link directs to port localhost:8081)

#### Extra details
Maybe the port is wrong and needs to be set to port 1092 or 1093 (did not investigate which of the PiAware pages should open)

@dirkhh: Maybe you could look into this.

dirkhh, 2023-09-16:
Will do

dirkhh, 2023-09-16:
Yup, exactly as I assumed... simply a hardcoded port that I missed.
https://github.com/dirkhh/adsb-feeder-image/commit/9abd49b9db8a09d25049dac871f3dbc0909d8309

Will be fixed in the next version update... which I should be able to do in time for the beta on Monday. And as this doesn't require any changes on the DietPi side, this should cause no issue for @MichaIng

dirkhh, 2023-09-16:
BTW, if you are testing... currently the signup for FlightRadar24 fails. That's a bug in the upstream container that I'm using. I working with upstream to get that fixed... again, completely inside of the ADS-B Feeder app, so users of the app on DietPi will automatically get the fixed version if/when they upgrade.

dirkhh, 2023-09-18:
these should all be addressed upstream.

MichaIng, 2023-09-18:
Great, many thanks 🙂!

StephanStS, 2023-09-22:
Bug is fixed, could be tested with the actual ADS-B docker update.
Thx to @dirkhh

closed 2023-09-22 by StephanStS