MichaIng/DietPi #6629: Ethernet interfaces in orange pi 5 plus

zappydood, 2023-09-20:
DietPi version: 8.21
Distro version: bookworm
Kernel version: Linux DietPi 5.10.160-legacy-rk35xx #1 SMP Mon Aug 28 01:21:24 UTC 2023 aarch64 GNU/Linux
SBC model: Orange Pi 5 Plus
Power supply used: included power adaptor with board
SD card used: Image is on EMMC

I am facing persistent network issues on my Orange Pi 5 Plus running DietPi version 8.21.

DHCP doesn't work: 
Despite being the only device on the network where DHCP doesn't work, the system can't get a DHCP address.
Static IP Resets:
Static IPs work but are unable to get connection upon reboot.
Manual Flush and Route Add:
I've had to manually flush eth0 and eth1 and add the routes manually to get it to work, which isn't sustainable.
Issues with dietpi-config:
Using dietpi-config doesn't resolve the issue and it even adds a non-existent WiFi interface.

Steps to reproduce: 
Reboot the system.
Observe that DHCP doesn't work and static IP say connected, but internet doesnt work.
Have to manually flush and add routes.

I've tried adding drop-in configs but they don't work properly.

I don't know how to fix this. I've fixed issues like this before in debian by flushing the routes before doing a restart of networking, however dietpi continuously has something going on that doesnt allow it to work properly. Using drop in configs or not using them makes it not work because its a toss up if internet is going to work on the next reboot. Dietpi-config also populates my static IP address with 0.0.0.0 in dietpi config even after saving it. The only thing so far that seems to allow it to work is to avoid dietpi-config altogether and only use the main /etc/network/interface but I do tend to have issues with that as well, just not as bad and persistently unusable as it has been trying it with drop ins.

MichaIng, 2023-09-21:
Can you revert all your manual `/etc/network/interfaces` and `/etc/network/interfaces.d` changes, reboot, and print the output of:
```sh
journalctl -u ifup@eth0.service -u ifup@eth1.service
```
This shows when the interface is being brought up, including the DHCP client trying to get a lease.

zappydood, 2023-09-21:
I'm having a really hard time doing this because I run it headless and don't have this anywhere near a monitor. But I did do those commands, I did revert everything to basic default settings and its still as usual unable to get a DHCP reservation despite their very clearly being one for that IP and its not working with static configuration also. When I run those commands, the only one that actually gives me anything for it is for eth0, the other one returns "no entries"

kc@DietPi:/etc/network$ sudo journalctl -u ifup@eth0.service
Sep 21 10:49:38 DietPi system [1]: Starting ifup@eth0.service
ifup for eth0...
Sep 21 10:49:38 DietPi system [1] : Finished ifup@eth0.service
- ifup for eth0.
kc@dietPi:/etc/network$

This has made it very challenging to do, I had to take a picture and copy and paste it as there is no other way to get it out of the dietpi. I can't seem to get anything up and running anymore. I'll try to edit drop ins again, but no matter what drop ins dont seem to work either.. I'm not sure why this is happening but I do know with my 10.0.0.0/8 network with netmask 255.0.0.0 it only has trouble getting DHCP on my main network. I don't really want to have to change my whole network configuration for the whole network as that is a nightmare and so far this is the only device unable to get an IP. For some reason static is failing too. I don't understand it seeing as my tmobile 5g gateway (secondary ISP, I use it as backup) it has no issue getting DHCP but its also on a much smaller 255.255.255.0 netmask. I can get it up and running here and there but not very easily and it takes a lot of flushing ip addr and link setting, otherwise I wont be able to get it back up with dietpi-config it seems nor manually editing..

MichaIng, 2023-09-22:
@zappydood 
I see the difficulty to test 🤔. According to your logs, DHCP is not enabled for this interface, i.e. `/etc/network/interfaces` contains `iface eth0 inet static` instead of `iface eth0 inet dhcp`?

zappydood, 2023-09-22:
Yeah, today I plan on relocating it next to my desk and so it'll be easier for me to run more testing and dedicate time to doing it with just switching my monitor input so I will test this more extensively today and provide as much time as I can to this. But I do want to mention for some reason DHCP on my 10.0.0.0/8 network on both my orange pi 5 plus as well as on a orange pi 4/orange pi 800 (the keyboard computer - I know it's not a supported install but I used the diet pi script on it a while back) seems to have this same issue with finding my DHCP lease. My range is a little unconventional but other than that and unicast dhcp I'm not really sure why it struggles with it when other devices don't seem to struggle. I don't really want to switch back to broadcast dhcp because of various reasons but I'll trouble shoot this a lot more thoroughly today and I'll make sure I get the entire session saved so I can provide further and better information. Thank you for the patience! I didn't have much time yesterday for moving everything.

Kasey Cannon

zappydood, 2023-10-03:
Hey friend! @MichaIng 

Just wanted to let you know I apologize for the length of time it took me to update I had some other things come up but I ended up getting it all working and thank you for the support. I don't know what initially caused my issue but after reverting back to defaults and then manually bringing down with ip addr flush eth0 and for eth1 and then setting them down for whatever reason after reboot it seemed to like my newly written config. I'm not the best with networking by hand with having dyscalculia (dyslexic with numbers) and so I guarantee I jumbled my ip address up in the config file since historically it's caused me misplacing an octet or two in the most odd place 😅

I do want to ask though, will dietpi-config in the future be able to manage dual Ethernet interfaces? In the case of the orange pi 5 plus it only sees one Ethernet port and then automatically adds in a WiFi interface it doesn't actually have inserted and I don't use WiFi on any of my servers as I even go as far as to remove the card if a computer ships with one. 

Thanks again, I really do appreciate your help and am proud to know that dietpi is my #1 choice for any servers. It seems to implement so well even on Debian remixes.

closed 2023-10-03 by zappydood