MichaIng/DietPi #6633: Enable wildcards for dietpi-letsencrypt

DarrenSchwartz, 2023-09-23:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- dietpi-letsencrypt throws an error when requesting a wildcard certificate like *.example.com.

#### Describe the solution you'd like:

- Should be able to get a cert for *.example.com.

#### Describe alternatives you've considered:

- ...

#### Additional context

[image]

MichaIng, 2023-09-23:
Wildcard certificates are generally supported via DNS authentication only, not via webroot/webserver based authentication, as Let's Encrypt cannot test whether ALL possible subdomains really are controlled by you. This again requires additional packages/plugins to be installed, and every DNS provider has its own plugin which works a little different.

Here is the documentation about this: https://eff-certbot.readthedocs.io/en/stable/using.html#dns-plugins

As there is no generic method for this, it is hard or practically impossible to reasonably implement this into DietPi-LetsEncrypt. What you can do for now is adding all subdomains space-separated.

DarrenSchwartz, 2023-09-25:
Yeah, you're right. Thanks for the docs and the explanation.

closed 2023-09-25 by DarrenSchwartz