MichaIng/DietPi #6634: dpkg-deb: error: archive 'package.deb' uses unknown compression for member 'control.tar.zst', giving up

stl3, 2023-09-23:
Date           | Sun Sep 24 07:37:50 +08 2023
- Bug report     | 13ad6dea-3b6b-43da-b4e8-230b31d58719
- DietPi version | v8.22.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux omitted 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `dpkg -i --force-confdef,confold package.deb`
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:

1. run dietpi-update and select update:
```
 Update available:                                                                                                 │
│  - Current version : v8.21.1                                                                                      │
│  - Latest version  : v8.22.3 
```
#### Expected behaviour:

- fetch update and install smoothly :)
#### Actual behaviour:

- getting an unknown compression error on package.deb
#### Extra details:

- ...
#### Additional logs:
```
dpkg-deb: error: archive 'package.deb' uses unknown compression for member 'control.tar.zst', giving up
```

MichaIng, 2023-09-23:
Many thanks for reporting. This was the `raspberrypi-sys-mods` package, right? I tested it on a Bookworm RPi, so probably zstd compressed packages are not supported on Bullseye yet, or they do require the `zstd` package. Can you try this one?
```sh
apt install zstd
```
E.g. do this from subshell opened from the error handler, then retry the step. If it does not work, I'll rebuild the package with xz compression.

stl3, 2023-09-24:
> Many thanks for reporting. This was the `raspberrypi-sys-mods` package, right? I tested it on a Bookworm RPi, so probably zstd compressed packages are not supported on Bullseye yet, or they do require the `zstd` package. Can you try this one?
> 
> ```shell
> apt install zstd
> ```
> 
> E.g. do this from subshell opened from the error handler, then retry the step. If it does not work, I'll rebuild the package with xz compression.

Just tried it and still giving me this:
```
  - Command: dpkg -i --force-confdef,confold package.deb                                                           │
│  - Exit code: 1                                                                                                   │
│  - DietPi version: v8.22.0 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                               │
│  - Image creator: DietPi Core Team                                                                                │
│  - Pre-image: Raspberry Pi OS Lite 64-bit                                                                         │
│  - Error log:                                                                                                     │
│ dpkg-deb: error: archive 'package.deb' uses unknown compression for member 'control.tar.zst', giving up           │
```

MichaIng, 2023-09-24:
Makes sense. New package is on the way, done it a minute or what:
- https://github.com/MichaIng/DietPi/commit/e44ed23
- https://github.com/MichaIng/DietPi/actions/runs/6286720628

... ah, not done in a minute. It is queued as all images are built as well. I'll repack it on the server directly.

MichaIng, 2023-09-24:
Okay done, please retry.

stl3, 2023-09-24:
:thumbsup: 😌 
Working now!
```
[  OK  ] DietPi-Patch | curl -fo package.deb https://dietpi.com/downloads/binaries/raspberrypi-sys-mods.deb
[ INFO ] DietPi-Patch | dpkg -i --force-confdef,confold package.deb, please wait...
(Reading database ... 100748 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking raspberrypi-sys-mods (2:20230510-dietpi1) over (2:20220915-dietpi2) ...
Setting up raspberrypi-sys-mods (2:20230510-dietpi1) ...
[  OK  ] DietPi-Patch | dpkg -i --force-confdef,confold package.deb
[  OK  ] DietPi-Patch | rm package.deb
[  OK  ] DietPi-Patch | Patched to DietPi v8.22
[ INFO ] DietPi-Update | APT autopurge, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | systemctl daemon-reload
[  OK  ] DietPi-Update | Incremental patching to v8.22.3 completed
[ INFO ] DietPi-Update | Checking for new available live patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v8.22.3
[ INFO ] DietPi-Update | Latest version  : v8.22.3
[  OK  ] DietPi-Survey | Sending survey data
[  OK  ] DietPi-Update | sync
[ SUB1 ] DietPi-Services > restart

```

Falko2000, 2023-09-24:
Thanks, working for me, too !

MichaIng, 2023-09-24:
Great, thanks for your report and testing guys!

closed 2023-09-24 by MichaIng