MichaIng/DietPi #6635: Update error using dietpi-update

Falko2000, 2023-09-24:
#### Details:
- Date           | Sun Sep 24 00:12:03 CEST 2023
- DietPi version | v8.22.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux HAL1000 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:      16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `dpkg -i --force-confdef,confold package.deb`
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:

1. the issue can be reproduced by just starting the update procedure again. 

#### Expected behaviour:
 The update process dietpi-update should update to the newest version of DietPi

#### Actual behaviour:
 An error message occurs and the update is halted
- ...
#### Extra details:

- ...
#### Additional logs:
```
dpkg-deb: error: archive 'package.deb' uses unknown compression for member 'control.tar.zst', giving up
```

MichaIng, 2023-09-24:
Should be fixed just now, see here and please retry: #6634

Falko2000, 2023-09-24:
Hi, will try, many thanks

Falko2000, 2023-09-24:
Working fine now, thanks for the fast and easy fix !

closed 2023-09-24 by MichaIng