MichaIng/DietPi #6636: wifi hotspot fails on

lukaszsobala, 2023-09-24:
#### Details:
- Date           | Sun Sep 24 22:08:00 BST 2023
- DietPi version | v8.21.1 (MichaIng/master)
- Image creator  | Vasco
- Pre-image      | Rock5A
- Hardware       | Generic Device (aarch64) (ID=22) (actually Rock5a using this image: https://dietpi.com/forum/t/rock-5a-support/17739/8)
- Kernel version | `Linux DietPi 5.10.160-legacy-rk35xx #1 SMP Mon Aug 28 01:21:24 UTC 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `systemctl start hostapd`
- Exit code      | 1
- Software title | DietPi-Config
- Report ID: 0c93b578-ff42-4ef2-ba4c-8e6f18c242d0
#### Steps to reproduce:
1. enable wifi6 branch 
2. go to the wifi hotspot settings
3. enable wifi6 support
4. apply
5. #### Expected behaviour:

- wifi hotspot works
#### Actual behaviour:

- hostapd fails to start with any wifi mode selected.
#### Extra details:

- Seems to be related to the channel frequency not being supported? My wifi chip is ax210, as a client it connects without a problem.
- #### Additional logs:
```
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xeu hostapd.service" for details.
```
Some of the journalctl log is appended here:
```
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sep 24 22:05:21 DietPi hostapd[1776]: Frequency 5180 (primary) not allowed for AP mode, flags: 0x30053 NO-IR
Sep 24 22:05:21 DietPi hostapd[1776]: Primary frequency not allowed
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: IEEE 802.11 Hardware does not support configured channel
Sep 24 22:05:21 DietPi hostapd[1776]: Could not select hw_mode and channel. (-3)
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: interface state COUNTRY_UPDATE->DISABLED
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: AP-DISABLED
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: Unable to setup interface.
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: interface state DISABLED->DISABLED
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: AP-DISABLED
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: CTRL-EVENT-TERMINATING
Sep 24 22:05:21 DietPi hostapd[1776]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sep 24 22:05:21 DietPi hostapd[1776]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Sep 24 22:05:21 DietPi hostapd[1776]: wlan0: IEEE 802.11 Hardware does not support configured channel
Sep 24 22:05:21 DietPi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Sep 24 22:05:21 DietPi systemd[1]: Failed to start hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sep 24 22:05:23 DietPi hostapd[1809]: Frequency 5180 (primary) not allowed for AP mode, flags: 0x30053 NO-IR
Sep 24 22:05:23 DietPi hostapd[1809]: Primary frequency not allowed
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: IEEE 802.11 Hardware does not support configured channel
Sep 24 22:05:23 DietPi hostapd[1809]: Could not select hw_mode and channel. (-3)
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: interface state COUNTRY_UPDATE->DISABLED
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: AP-DISABLED
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: Unable to setup interface.
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: interface state DISABLED->DISABLED
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: AP-DISABLED
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: CTRL-EVENT-TERMINATING
Sep 24 22:05:23 DietPi hostapd[1809]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sep 24 22:05:23 DietPi hostapd[1809]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Sep 24 22:05:23 DietPi hostapd[1809]: wlan0: IEEE 802.11 Hardware does not support configured channel
Sep 24 22:05:23 DietPi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
```
This goes on and on. Why does it say it's on mode 802.11a?

MichaIng, 2023-09-24:
Thanks for testing WiFi 6.

> enable wifi6 branch https://github.com/MichaIng/DietPi/pull/6575

What do you mean with `wifi6` branch? The branch of the linked PR was a feature branch which does not exist anymore, but has been merged into regular DietPi `master` branch. If not done, please try to update to latest DietPi v8.22.

However, the related code has not changed. I'm quoting again the description of the modes from the example config:
```sh
# Operation mode (a = IEEE 802.11a (5 GHz), b = IEEE 802.11b (2.4 GHz),
# g = IEEE 802.11g (2.4 GHz), ad = IEEE 802.11ad (60 GHz); a/g options are used
# with IEEE 802.11n (HT), too, to specify band). For IEEE 802.11ac (VHT), this
# needs to be set to hw_mode=a. For IEEE 802.11ax (HE) on 6 GHz this needs
# to be set to hw_mode=a. When using ACS (see channel parameter), a
# special value "any" can be used to indicate that any support band can be used.
# This special case is currently supported only with drivers with which
# offloaded ACS is used.
# Default: IEEE 802.11b
hw_mode=g
```
So `hw_mode=a` implies IEEE 802.11a, but is also correct for IEEE 802.11ac (5 GHz) and "IEEE 802.11ax on 6 GHz". It does not explicitly mention "IEEE 802.11ax on 5 GHz", but there is no other mode which could possible be correct/required for this.

But just to rule it out, since the AX210 supports 6 GHz as well, could you just try to enter one of the 6 GHz channels? Interestingly some of them have the same numbers as 2.4 GHz channels, so I hope also hostapd supports them in combination with `hw_mode=a`: https://en.wikipedia.org/wiki/List_of_WLAN_channels#6_GHz_(802.11ax_and_802.11be)

And of course also try some of the other 5 GHz channels. 36 seems to be the most compatible one, but I'm not sure whether it works in all countries/with all country codes.

Last but not least, I guess the kernel does not come with a driver for this WiFi adapter OOTB. Where did you get it from? Probably it has limitations with hostapd. Does it work well with WiFi 4 and 5?

MichaIng, 2023-09-24:
> `Frequency 5180 (primary) not allowed for AP mode, flags: 0x30053 NO-IR`

https://github.com/lwfinger/rtw88/issues/29

> Anyway, `no IR` means it's restricted in your country, which channel 36 happens to be. Just change the iw region to a different one.

So indeed it's the country code in combination with this channel. Which country code do you use, respectively where is the AP located? Try a different WiFi 5 channel, or for testing, temporarily change the country code, to verify that generally 5 GHz channel 36 is supported by the adapter. _Note that it is illegal to operate an AP like this._

lukaszsobala, 2023-09-25:
I actually did not set any locale and it might have been the problem, but I don't think so.

Today I used a fresh install and set all the locales to mine (PL). What combinations work?
- 2.4 GHz with wifi 4/5/6 toggles off (channel 3)
- 2.4 GHz with wifi 4/5 off and 6 on (channel 3)
- 2.4 GHz with any other combination

I can't seem to change the channel for WiFi 5 GHz, it defaults to channel 36 but when I try to get to the list, no channels are available. It's as if the channel list for 5 GHz did not exist (I tried a few country codes).

AX210 works because these files are present in the image (it came with them in `/lib/firmware/` and `/lib/firmware/intel/` - and many others - predownloaded):

```
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.sfi
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.ddc
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-ty-a0-gf-a0-59.ucode
```

At the time when I installed that version of DietPi, the branch was probably still unmerged. With this fresh install I just updated to 8.22.

The good thing is that WiFi 6 works as expected, I get a little "6" when connected with it enabled. Also, I think that there must be a way to make the hotspot use both 5 GHz and 2.4 GHz bands like every WiFi 6 router does (without specifying it - I think that with WiFi 6 enabled there should be an option saying "2.4 + 5", for example). Maybe even use the 6 GHz band as WiFi 6E should be able to. But the hardware capabilities might be tricky to detect within dietpi.

MichaIng, 2023-09-25:
> I actually did not set any locale and it might have been the problem, but I don't think so.

The locale is irrelevant, but the WiFi country code is mandatory. Channel 36 btw worked well here with German (DE) country code.

> I can't seem to change the channel for WiFi 5 GHz, it defaults to channel 36 but when I try to get to the list, no channels are available. It's as if the channel list for 5 GHz did not exist (I tried a few country codes).

Oh, and this is a bug. It should be an inputbox (no list, as of the fragmented and per-country supported codes). Fixed with: https://github.com/MichaIng/DietPi/commit/f943e3c

Please try this:
```sh
sed -i "s/G_WHIP_MENU 'Please enter a 5 GHz WiFi channel/G_WHIP_INPUTBOX 'Please enter a 5 GHz WiFi channel/" /boot/dietpi/dietpi-config
dietpi-config
```
Now you should be given an inputbox, with 36 entered as default, and the URL given to the Wikipedia article I linked above to find a suitable channel.

lukaszsobala, 2023-09-26:
Indeed, now I can choose the channel but still none of the services are able to start up on 5 GHz, even with the exact country code/channel combination that you used.

MichaIng, 2023-10-12:
@lukaszsobala 
Does your Wi-Fi 6 router support WPA3, or better, does it even enforce WPA3 when Wi-Fi 6 is enabled? According to [this info](https://github.com/raspberrypi/linux/issues/4718#issuecomment-1155509563), WPA3 is mandatory for Wi-Fi 6, but we know that it is often confused with 6 GHz, so would be good to know which one is actually meant. Actually you tested Wi-Fi 6 with 2.4 GHz and WPA2 successfully already, it seems, but probably it would then work with 5 GHz (and even 6 GHz) as well when using WPA3. The link contains an example config, and you can adjust the related settings in `/etc/wpa_supplicant/wpa_supplicant.conf` and then retry connecting to the network.

For reference: #6636

lukaszsobala, 2023-10-12:
I don't see how my router is relevant here as it is the dietpi system that serves as an access point creating its own independent network.

Yes, my router (Huawei AX3) supports it but that's another network.

I will try messing with wpa_supplicant using info in the link. And yes, dietpi-hotspot does work with 2.4 GHz.

MichaIng, 2023-10-12:
Ah right, of course, I was not thinking correctly, just remembered the 5/6 GHz issue with Wi-Fi 6 here after reading the requirements.

However, the underlying issue might be the same. So you could try to add the `ieee80211w=1` to `/etc/hostapd/hostapd.conf` and for WPA3 `wpa_key_mgmt=WPA-PSK SAE`, i.e. just adding "SAE" space-separated to that config. So that way WPA2 and WPA3 should be supported. I cannot find any up-to-date `hostapd.conf` manpage or documentation, but some forum threads which indicate this possibility:
- https://www.linuxquestions.org/questions/showthread.php?p=6269450#post6269450
- https://community.infineon.com/t5/Wi-Fi-Bluetooth-for-Linux/Please-provide-a-tested-example-of-a-hostapd-conf-file-configured-for-WPA3/td-p/86690

The second link would then be WPA3-only.

lukaszsobala, 2023-10-13:
Indeed, I managed to change it to WPA3 (even enforcing SAE) by editing `hostapd.conf`, deleting WPA-PSK and adding SAE, but I don't know what to edit to get 5 GHz working. I tried changing the channel in that file to "36" but this did nothing except it made hostapd not start after a restart.

Confusingly, the option `wpa=3` doesn't mean WPA3 but WPA 1 or 2 (for WPA3 it needs to be `wpa=2`)

actraiser30, 2024-01-10:
Hi I also noticed that when I select the 5ghz band the max speed set is 54mbps instead of 300mbps or moreI don't know if this is a bug also

MichaIng, 2025-05-31:
Does the issue still exist with recent kernel (Linux 6.1)?

lukaszsobala, 2025-05-31:
I'd have to reimage and check, will do

MichaIng, 2025-09-21:
Closing this for now. Feel free to reopen if the issue persists.

closed 2025-09-21 by MichaIng

Anutrix, 2026-05-03:
Facing this issue. 5 Ghz does not working at all. Tried US, GB, JP with channel 36 and 44.
2.4 Ghz works perfectly fine.

Joulinar, 2026-05-03:
as this topic is a couple of years old, better to start on own one, providing all required information.