MichaIng/DietPi #6642: Software installation of O!MPD [129] pulls to many dependencies

maartenlangeveld, 2023-09-27:
I recently did a fresh install of DietPi bookworm on my NanoPi Neo2.

When installing O!MPD using DietPi software installer (choice [129]), it appeared to me that it installed many programs that I don't need since I am running mpd on another SBC in my network, having O!MPD remotely control mpd.
O!MPD only needs a webserver, PHP, mysql server and mpd (not in my specific case, but I understand the logic to have ik installed).

I do not understand why the script also installs:
- ffmpeg
- alsa-utils
- avahi-daemon

Furthermore uninstalling O!MPD using DietPi uninstaller, other programs, that I had installed myself, i.e. upmpdcli were uninstalledby DietPi...

MichaIng, 2023-09-27:
While O!MPD can be used as web player to play music via browser on a client system, MPD is mainly used as jukebox-like player to play music from the server's audio system and only control it via MPD clients/interfaces like O!MPD. Hence ALSA (the fundamental/low-level Linux audio system), is mandatory for MPD to work. AFAIK, the service would even fail to start, if ALSA was not installed.

O!MPD comes with additional features, including playing audio stream directly from YouTube, which requires `youtube-dl`, which again requires FFmpeg. FFmpeg is generally a widely used A/V library for encoding and decoding audio streams into/from various modern formats, so it comes together with many media server software.

The Avahi daemon could be indeed removed from (MPD) dependencies. It is commonly used to have the MPD server quickly detected in LAN, but not mandatory.

Btw, the MPD package itself pulls the ALSA, FFmpeg (libav*) and Avahi libraries (not the services/CLI tools) as strict dependencies, so even if we did remove those three dependencies in `dietpi-software`, 70% of it would be installed anyway: https://packages.debian.org/bookworm/mpd

> Furthermore uninstalling O!MPD using DietPi uninstaller, other programs, that I had installed myself, i.e. upmpdcli were uninstalledby DietPi...

No, here is the exact code which runs on O!MPD uninstall:
```sh
		if To_Uninstall 129 # O!MPD
		then
			[[ -d '/var/www/ompd' ]] && G_EXEC rm -R /var/www/ompd
			[[ -f '/etc/lighttpd/conf-enabled/99-dietpi-ompd.conf' ]] && G_EXEC lighty-disable-mod dietpi-ompd
			[[ -f '/etc/lighttpd/conf-available/99-dietpi-ompd.conf' ]] && G_EXEC rm /etc/lighttpd/conf-available/99-dietpi-ompd.conf

			Remove_Database ompd
		fi
```
So it removes the web interface, removes the Lighttpd configuration (if used as webserver) and removes the MariaDB database, nothing more. But it might be uninstalled together with MPD, if it depends on the `mpd` package. Ah, and also if it depends on `libmpdclient2` which is currently purged with MPD together, while it is actually obsolete and in code for historical reasons when we used own MPD packages with no dependencies package-wise defined. I'll remove that step.

maartenlangeveld, 2023-09-27:
Hello @MichaIng,

Thanks for your elaborate and clear answer. Makes sense. I think upmpdcli might be uninstalled by mpd dependy but not sure.
DietPi is a gem, thanks for that!

Kind regards,
Maarten

closed 2023-09-27 by maartenlangeveld