MichaIng/DietPi #6646: Fresh new installation 8.22.3  `apt-get -y dist-upgrade` fail

milapointe, 2023-09-29:
#### Details:
- Date           | Fri Sep 29 03:24:20 BST 2023
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y dist-upgrade`
- Exit code      | 132
- Software title | DietPi-Software

#### Steps to reproduce:

Fresh new installation on a RPi4 and cant get past this.

#### Expected behaviour:
Installed

#### Actual behaviour:
Crash on dist upgrade

#### Extra details:

First time user
Kingston 64gb sdxc

#### Additional logs:
logs sent : b895d3fa-f285-443c-9cc4-94d2c4bbd58a

Joulinar, 2023-09-29:
pls can you share the full output of following lines

```
apt update
apt upgrade
apt dist-upgrade
```

milapointe, 2023-09-30:
root@DietPi:/tmp/DietPi-Software# apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:3 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [308 kB]
Fetched 331 kB in 4s (83.1 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/arm64/release-notes/ch-information.html#non-free-split

root@DietPi:/tmp/DietPi-Software# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Illegal instruction

root@DietPi:/tmp/DietPi-Software# apt dist-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Segmentation fault

I'm reading its possible that corrupt occured....I will try a new run flashing the img....

ravenclaw900, 2023-09-30:
Yeah, errors like `Illegal instruction` and `Segmentation fault` generally indicate some sort of corruption in the filesystem / specific binary. If it still doesn't work after reflashing, you might want to consider trying a new SD card.

milapointe, 2023-09-30:
Happened with my other SD Card. I'm buying brand new SD Card + SD Card Reader....

Joulinar, 2023-09-30:
On a RPI4, you should be able to use a USB pen stick fur testing instead of SD card.

MichaIng, 2023-09-30:
Also check for kernel errors:
```sh
dmesg -l 0,1,2,3
```
Undervoltage, i.e. an insufficient PSU could be also the reason for such issues.

Btw, as I just had some discussion about whether SD card are generally suitable as OS boot root media or not, would you be open to share some meta data of your SD card(s) to check whether they are original or fakes?
```sh
cat /sys/block/mmcblk0/device/cid
cat /sys/block/mmcblk0/device/csd
```
The two codes can be decoded here: https://gurumeditation.org/1342/sd-memory-card-register-decoder/
It gives a bunch of information, and especially the serial number, device name, manufacturer ID and "oemid" can be easily compared with the actual manufacturer and what is expected for the card. If they do not match and/or the serial number is very low, it is mostly assured or very likely that it is a fake card, which is then expected to die earlier than the original would do.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen it the problem persists.

closed 2025-05-31 by MichaIng