MichaIng/DietPi #6649: Tailscale requires iptables but they aren't installed

dirkhh, 2023-09-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.22.3`
- Distro version | `bookworm`
- Kernel version | `Linux adsb-feeder 6.1.50-current-rockchip64 #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO3 (aarch64)` (but I doubt it's specific to the SBC)
- Power supply used | 5V 2A noname
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | tailscale
- Was the software title installed freshly or updated/migrated? | fresh install from scratch
- Can this issue be replicated on a fresh installation of DietPi? | yes

- Bug report ID | N/A

#### Steps to reproduce

1. install DietPi
2. select software 58 (tailscale)

#### Expected behaviour

- tailscale should work :)

#### Actual behaviour

```
Sep 29 17:05:00 adsb-feeder systemd[1]: tailscaled.service: Failed with result 'exit-code'.
[...]
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: Program starting: v1.50.0-ta920f0231-geb5b0beea, Go 1.21.1: []string{"/usr/sbin/tailscaled", "--cleanup"}
[...]
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: dns: [rc=unknown ret=direct]
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: dns: using "direct" mode
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: dns: using *dns.directManager
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: linuxfw: clear iptables: exec: "iptables": executable file not found in $PATH
Sep 29 17:05:00 adsb-feeder tailscaled[5473]: linuxfw: clear ip6tables: exec: "ip6tables": executable file not found in $PATH
[...]
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: Program starting: v1.50.0-ta920f0231-geb5b0beea, Go 1.21.1: []string{"/usr/sbin/tailscaled", "--state=/var/lib/tailscale/tailscaled.state", "--socket=/run/tailscale/tailscaled.sock", "--port=41641"}
[...]
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: wgengine.NewUserspaceEngine(tun "tailscale0") ...
Sep 29 17:05:01 adsb-feeder systemd[1]: Started tailscaled.service - Tailscale node agent.
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: router: default choosing iptables
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: router: using iptables
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: wgengine.NewUserspaceEngine(tun "tailscale0") error: creating router: exec: "iptables": executable file not found in $PATH
Sep 29 17:05:01 adsb-feeder tailscaled[5480]: getLocalBackend error: createEngine: creating router: exec: "iptables": executable file not found in $PATH
Sep 29 17:05:01 adsb-feeder systemd[1]: tailscaled.service: Main process exited, code=exited, status=1/FAILURE
```

#### Extra details

- ...

MichaIng, 2023-09-29:
They demoted it from a dependency to a recommendation 🤔: https://github.com/tailscale/tailscale/releases/tag/v1.50.0
> ### Linux
> - debian package lists iptables+iproute2 packages as recommended, not required.

A little strange if it is indeed strictly required for the service to start with default config (we do not touch any of this). I thought it may have been due to the fact that `nftables` and `ufw` may be now supported, but those do not appear anywhere in package meta data. For this it would have been possible to define dependencies as `iptables | nftables | ufw`, which is an a OR b OR c.

I just tested it with `nftables` installed instead, and it still fails. `ufw` pulls `iptables` as dependency anyway, being a frontend only. So looks like a faulty change to me.

MichaIng, 2023-09-29:
Okay here is the reason: https://github.com/tailscale/tailscale/issues/9236
It seems to be possible to make use of `nftables` via `/etc/default/tailscaled` setting: https://tailscale.com/kb/1294/firewall-mode/#how-to-set-the-firewall-mode

The alternative dependency `iptables | nftables` makes sense in this case. I just suggested it, let's see. Otherwise we can add `iptables` in `dietpi-software`, of course.

Joulinar, 2023-09-29:
We have a related post within our forum as well

MichaIng, 2023-10-03:
Thanks for chiming in. So Tailscale generally runs without any of both `iptables` and `nftables`? Then it makes sense as it is.

> I was under impression that on most Debian-derived distros package manager is by default configured to install `recommends` dependencies. Is this not true for DietPi?

It is the `dpkg` default to install `recommends` indeed, but we disable it in DietPi to minimise the amount of used disk space. However, no problem, we'll install `iptables` out end then.

Btw, also in recommends, you could use "`iptables | nftables`", so that also with `dpkg` defaults, `iptables` won't be installed if `nftables` is installed already.

Joulinar, 2023-10-03:
@MichaIng we might need to add a step within next patch to install `iptables` if not present?

knyar, 2023-10-04:
FYI, we had a discussion internally and will return a strict dependency on `iptables`. In the future when Tailscale client can auto-detect and use `nftables`, either of the two will be sufficient as a dependency.

MichaIng, 2023-10-04:
Thanks for the info. I'll mark the issue as closed then.

closed 2023-10-04 by MichaIng