MichaIng/DietPi #6651: The script /etc/cron.daily/dietpi can't reach G_AGUP function and it is exiting prematurely

lz1aam, 2023-09-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.22.3;
- Distro version | Bullseye;
- Kernel version | 6.1.21-v8+;
- SBC model | RPi Zero 2 W;

#### Steps to reproduce
1. Set `CONFIG_CHECK_DIETPI_UPDATES=` in /boot/dietpi.txt to 0;
2. Run /etc/cron.daily/dietpi;

#### Expected behaviour
- The script /etc/cron.daily/dietpi shouldn't exit prematurely;

#### Actual behaviour
- The script /etc/cron.daily/dietpi can't reach G_AGUP function from line 31 and it is exiting prematurely with:
   > /etc/cron.daily/dietpi: line 31: G_AGUP: command not found

#### Extra details
- Adding the extra line `source /boot/dietpi/dietpi-update` just before line 31:  `G_AGUP -f` solves the issue.

MichaIng, 2023-09-30:
Many thanks for your report.

Indeed dietpi-globals are missing there, crazy that this was never detected until now. Seems like this config combination is rare.

The script however should not exit prematurely, does it? It just continues on error, but the intended APT update check is indeed missing then.

You should not source `/boot/dietpi/dietpi-update`, but the globals only:
```sh
. /boot/dietpi/func/dietpi-globals
```
Fixed for next DietPi version with: https://github.com/MichaIng/DietPi/commit/b04b43b

lz1aam, 2023-09-30:
Looks like the script trow the error message and continue without ending prematurely, but since the `G_AGUP -f` isn't running it takes very litle time for the script to finish and that trick me. Sorry, I also paste the wrong source.
 
 It will be better if the source `. /boot/dietpi/func/dietpi-globals` is declared in the begining of the script since `G_AGUP` is used also in row `[[ $APT_UPDATE_MODE == 2 && -f '/run/dietpi/.update_available' ]] && G_AGUP -f` earlier in the script.

I'm pleased to have been able to assist. Thank you for your dedication to the DietPi project.

MichaIng, 2023-09-30:
Keeping the load minimal: https://github.com/MichaIng/DietPi/commit/fa0646e
Now we have all globals function calls 😄. Would be cleaner to do all of this properly within `dietpi-update` I guess, having all kind of system updates in one script, also as part of the interactive menu.

Also, as we are working on an own APT server, we'll likely move parts and finally DietPi completely into the APT repository as packages, So that `dietpi-update` will loose its current major purpose. Only for APT key/source/config updates, which are currently done in "pre-patches", an additional script is still required. So until we know a bit clearer where the journey is going to, I will keep changes to `dietpi-update` minimal.

closed 2023-10-21 by MichaIng