MichaIng/DietPi #6652: dietpi-backup | stuck at Searching for previous backups...

kingspeedy95, 2023-09-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm

- Kernel version | `uname -a`
Linux DietNUC 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Native PC (x86_64)

- Power supply used | (EG: 5V 1A RAVpower)
ASRock Deskmini X300 120W PSU

- SD card used | (EG: SanDisk ultra)
NVME SSD -> KINGSTON SNV2S2000G

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. open dietpi-backup
2. set manual any location (I tested a local directory and an NTFS mount.)
3. hit ok

#### Expected behaviour

- If available a backup should be found or there should be a message that no backup could be found. Or at least a timeout.

#### Actual behaviour

- stuck at "[ INFO ] DietPi-Backup | Searching for previous backups, please wait..." and nothing happens. No further output. No timeout takes effect either. It must be aborted with CTRL + C.

#### Extra details

- ...

MichaIng, 2023-09-30:
This is what the script does to find backups "Search"ing for previous backups:
```sh
find / -type f -name '.dietpi-backup_stats'
```
So it searches really the whole root directory tree, including all attached drives, which naturally lakes long, especially when you have network drives attached. You can test how long it takes:
```sh
time find / -type f -name '.dietpi-backup_stats'
```
What works much faster is selecting the location form the "List" of attached drives. And no searching is done at all when you type in the location "Manual"ly. So your steps to reproduce do not match the behaviour, or did the script really spit out this "Searching for previous backups" message when you select "Manual" and type in the path of backup dir manually? That would be a bug I cannot reproduce here.

kingspeedy95, 2023-09-30:
Sorry, I was just being silly. I had "Search" highlighted when I selected "OK". Then of course it starts searching. I just had to close the dialog with "Cancel" to save the new location.

closed 2023-09-30 by kingspeedy95

MichaIng, 2023-09-30:
Ah makes sense. I'll rename this button to "Back", better indicating what it is for.

Done with: https://github.com/MichaIng/DietPi/commit/77c9b12

So it is `<Select>` on the left side to select one of the menu options, and `<Back>` to navigate back to the main menu. We generally had some internal talk about improving the whiptail menu button text throughout our scripts to better indicate what they do. The default is "Ok" and "Cancel", which is no always matching well.

kingspeedy95, 2023-10-01:
That sounds like a good improvement! Thanks a lot!