MichaIng/DietPi #6654: dietpi-update / dietpi-config | Select which services must not be stopped during an update

kingspeedy95, 2023-10-01:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- On a dietpi-update, docker is stopped. But in docker the pihole is running. If it is stopped, the update fails.

#### Describe the solution you'd like:

- One possibility would be to add a function in dietpi-config that allows you to select which services should not be stopped during an update.

#### Describe alternatives you've considered:

- An alternative would be to include an automatic detection in the update script which services must not be stopped. But this is probably not so trivial to solve.

#### Additional context

- What I have not tried yet is to install the PiHole via dietpi-software. Maybe the automatic detection is already there and the service is not stopped during an update.

Joulinar, 2023-10-01:
you already can do that using `dietpi-services` > select `docker` > set `Include/Exclude` to `Exclude`

I hope PiHole is not the only application running in Docker. If so, it would be a waste of system resources. Anyway, you should not set the local DietPi DNS to PiHole. The best practice would be to use a STATIC IP configuration inside DietPi and choose a public upstream DNS. This way you avoid DNS queries failing inside DietPi when Docker is down. There can be a whole range of reasons for a Docker failure. Not just a DietPi update. That doesn't affect the clients on your local network. 😉

kingspeedy95, 2023-10-01:
Cool, thanks a lot for the information. I didn't know dietpi-services yet.

No, pihole is not the only stack running in docker. I have taken the PiHole out of the docker again. So I only need to put the PiHole on the exclude list. I will think about your hint regarding the best practice as soon as I have more time.

Joulinar, 2023-10-01:
Using DietPi installation should exclude Pihole by default. We don't stop native Pihole install by default.