MichaIng/DietPi #6655: Error when dietpi-update through Kitty ssh: syntax error: operand expected (error token is "/  ")

Voklen, 2023-10-01:
#### Required Information
- DietPi version:
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bookworm 1`
- Kernel version | `Linux rpi-uk 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi B (armv6l)` 
- Power supply used | 5.1V 2A Apple
- SD card used | SanDisk SD from September 2011

#### Steps to reproduce

1. Install [kitty terminal](https://sw.kovidgoyal.net/kitty/) on your desktop computer
2. SSH into DietPi through `kitty +kitten ssh <user>@<ip>`
3. Run `sudo dietpi-update`

#### Expected behaviour

- Update system or say it is up to date

#### Actual behaviour

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

tput: unknown terminal "xterm-kitty"
/boot/dietpi/func/dietpi-globals: line 267: ( 59 + 5 ) /  : syntax error: operand expected (error token is "/  ")
```

#### Extra details

I noticed this when I was updating DietPi, so I tried to set `export TERM='dumb'` but then it didn't come up with the updating UI so I had to get the keyboard out, plug it into a monitor and update it from there.
This issue potentially has some relation to #5347, running
```
/boot/dietpi/func/run_ntpd 1
```
still gives
```
/boot/dietpi/func/dietpi-globals: line 267: ( 58 + 5 ) /  : syntax error: operand expected (error token is "/  ")
```

MichaIng, 2023-10-01:
First of all, I think on login, there should be a warning that no matching terminal definition has been found, isn't it? It should even automatically set the `TERM` variable to `xterm`, working around the issue. Or do you use another shell than `bash`?

However, best is anyway to install native `kitty-xterm` support. Two options:
- Install via APT `apt install kitty-xterm`. I'd be actually interested whether this works well, as we might want to add/offer this automatically on SSH login via kitty terminal. It is not necessary the most recent definition, so if there are significant changes, it might not support all new features of the terminal.
- Install via kitty directly, which is pretty simple and assured to match 100% the version: https://sw.kovidgoyal.net/kitty/faq/#i-get-errors-about-the-terminal-being-unknown-or-opening-the-terminal-failing-or-functional-keys-like-arrow-keys-don-t-work

Voklen, 2023-10-01:
Ahhh that's interesting and explains why the normal way with `+kitty` didn't work but installing `kitty-terminfo` did indeed fix the issue. Thank you very much for the quick response.

MichaIng, 2023-10-01:
As you are not the first running into this, let me reopen and think about how we can solve this a clever way.

Voklen, 2023-10-01:
There is the option of modifying `visudo` to preserve `TERMINFO` with
```
Defaults env_keep += "TERM TERMINFO"
```

MichaIng, 2023-10-01:
It is preserved, which is the problem, as `root` does not have this definition in its home dir. It would be explicitly reset/set to `xterm`, but then the nice kitty terminal features are lost.

If `kitty +kitty ssh` does not only install but also update the users' terminal definition, I though about a symlink, to have the global one kept in sync:
```sh
sudo ls -s "$HOME/.terminfo/x/xterm-kitty" /etc/terminfo/x/xterm-kitty
```
But I guess ever other non-root user would then run into errors, as `$HOME/.terminfo/x/xterm-kitty` within a directory they have no read access to. Copying/moving it over means that it won't be updated. Via `apt` it will get bug fixes and security patches, if needed, from Debian, but feature-wise can be up to 2 years behind. 🤔