MichaIng/DietPi #6656: After update Tailscale not Working

lutfor-diu, 2023-10-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8`
- Distro version | `
bookworm 1`
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi B (armv6l)'
- Power supply used | (5V 1.5A Mobile)
- SD card used | (Original RP MMC)

#### Additional Information (if applicable)
- Software title | (Tailscale)
- Was the software title installed freshly : Not updated
- Can this issue be replicated on a fresh installation of DietPi? : yes, same

- Bug report ID | `8b8d80e6-2ca3-4b3f-bb16-eff5f7058706`

#### Steps to reproduce

1. install tailscale
2. use android tailscale app to connect dietpi server
3. cant connect

#### Expected behaviour

connect to server

#### Actual behaviour

- cant connect to server

#### Extra details

- ...

Joulinar, 2023-10-01:
Probably similar to this one https://github.com/MichaIng/DietPi/issues/6649

MichaIng, 2023-10-07:
Please try:
```sh
apt install iptables
systemctl restart tailscaled
```

lutfor-diu, 2023-10-07:
> Please try:
> 
> ```shell
> apt install iptables
> systemctl restart tailscaled
> ```

It works!
Thanks a lot brother 😁

closed 2023-10-08 by MichaIng