MichaIng/DietPi #6657: Add 'dietpi' command

infinitytec, 2023-10-01:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Feature request: add 'dietpi' as a command. 

#### Describe the solution you'd like:

- This command would then list all the DietPi-specific commands (such as dietpi-update and dietpi-config) as well as a short description for what is what. This would make referencing how to access these tools easier.

#### Describe alternatives you've considered:

- Looking at docs or remembering the commands.

#### Additional context

- It could be useful. I sometimes forget the specific commands.

Joulinar, 2023-10-01:
did you tried `dietpi-launcher` tool?

infinitytec, 2023-10-01:
Ah, I forgot about that.

closed 2025-07-18 by MichaIng