MichaIng/DietPi #6658: External SSD failed to format

wsbeaton, 2023-10-02:
### Creating a bug report/issue

#### Details:
- Date           | Mon Oct  2 03:29:51 CDT 2023
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `mkfs.fat -I /dev/sda1`
- Exit code      | 1
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. Mount 256GB external hard drive
2. Open dietpi-drive_manager
3. Format as FAT32
#### Expected behaviour:

- Disk is formatted in FAT32
#### Actual behaviour:

- Format failed. Last 4 lines:
- [ INFO ] DietPi-Drive_Manager | Creating new gpt partition table
  [  OK  ] DietPi-Drive_Manager | parted -s /dev/sda mklabel gpt
  [  OK  ] DietPi-Drive_Manager | parted -s /dev/sda mkpart primary 0% 100%
  [FAILED] DietPi-Drive_Manager | mkfs.fat -I /dev/sda1
#### Extra details:
- Disk was already formatted in FAT32. I was only trying different things in this new OS.
- ...
#### Additional logs:
```
mkfs.fat: unable to open /dev/sda1: No such file or directory
```

MichaIng, 2023-10-02:
Hmm, is the drive itself still present when you exit the script from there?
```sh
lsblk
```
Possibly it got disconnected due to insufficient power or so. Please also check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen it the problem persists.

closed 2025-05-31 by MichaIng