MichaIng/DietPi #6659: Orange pi 3b network problem

GremlineQ, 2023-10-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOHNER= 'MichaIng'`
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 5.10.160-rockchip-rk356x #1.0.0 SMP Tue Aug 22 11:38:20 CST 2023 aa`
- SBC model | `Generic Device (aarch64)`
- Power supply used | 5V 3A
- SD card used | SanDisk extreme 128GB

#### Additional Information (if applicable)
- Converting debian to dietpi
- I tried to convert fresh installation of debian to dietpi

- If I could, I would send it, but opi 3b doesn't have access to the Internet, that's why there is this ticket 

#### Steps to reproduce
1. I tried to convert Debian to dietpi, everything worked until I rebooted opi 3b  

#### Expected behaviour

- subsequent installation stages should be completed without problems (at least from what I read in the conversion guide) 

#### Actual behaviour

- the installation cannot be completed due to lack of internet access (wired and wireless internet does not work even though the option for wifi to work is selected).

#### Extra details

- as I wrote above, I use Orange Pi 3b (has rockchip rk3566)  with Debian bookworm prepared by Orange Pi.

MichaIng, 2023-10-02:
Did you download the Debian image from Orange Pi website directly? If so, please try with the Armbian Bookworm image, which also uses a much newer Linux version: https://www.armbian.com/orangepi3b/

This looks like a low hanging fruit for us to implement natively into DietPi. Just made an image request out of it.

I was confused that Orange Pi 3, 3 LTS, and 3B are such different SBCs. Only the latter uses the RK3566, the other two are Allwinner H6 😄.

GremlineQ, 2023-10-03:
Yes, I downloaded it from the Orange Pi website. If I understand correctly, you want me to try to download Armbian and run the dietepi script on it and see if it works? 

I can't wait for native support, Thanks. 

I understand this might be confusing. Anyways, thanks for the quick response. Now, the only thing left to do is wait.

Joulinar, 2023-10-04:
@MichaIng a successful try at our forum https://dietpi.com/forum/t/orange-pi-3b-problem/18008/6?u=joulinar

GremlineQ, 2023-10-13:
Just out of curiosity, when will we be able to expect native support?

Kenzu, 2023-12-17:
Hi
What is needed to get native support?
How to help?

Jesper

MichaIng, 2023-12-17:
Support added with: https://github.com/MichaIng/DietPi/commit/6336fc6
Test image builds are running and can be found here once done, in an hour or so: https://dietpi.com/downloads/images/testing/

MichaIng, 2023-12-20:
Would be great if someone could give the images a try, the Bookworm one if unsure. I do not expect surprises, but better be safe than sorry, since we do not own this device to test ourselves.

Kenzu, 2023-12-21:
Hi

The image boot on my OPI3B, but i have no output on my 7" Bigtreetech LCD (1024x600@60). The screen works in armbian with legacy kernel 5.10.160. It does give output on my 1920x1080p monitor, so some problem with the 1024x600 mode or some DVI/HDMI feedback from the screen.
I did try armbian with edge kernel, and it's the same problem.

Thanks

MichaIng, 2023-12-21:
I guess with the Armbian edge images it is the same? https://dl.armbian.com/orangepi3b/archive/

The LCD is attached via HDMI, i.e. just the resolution wrong? For DSI LCDs it will be the same issue reported here: #6740
I also linked two other issues there regarding how to change console mode resolution (and rotation) for console mode, either via `fbset`, or via `video=` kernel command-line parameter, but needs testing.

GremlineQ, 2023-12-31:
sorry, I forgot to check the topic, and I didn't receive the notification either. If it is needed to test it, I can test it tomorrow

Kenzu, 2024-01-01:
Hi

I cannot get wifi to work with onboard wifi. With my rpi usb wifi stick it works.
And my 7" HDMI LCD is also not working. 
Tried with:
extraargs=net.ifnames=0 video=HDMI-A-1:1024x600-24@60
in /boot/dietpiEnv.txt but it gives my this in dmesg:
[    3.069590] rockchip-drm display-subsystem: [drm] User-defined mode not supported: "1024x600": 60 51200 1024 1192 1224 1344 600 615 621 635 0x68 0x5
[    3.070903] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
It's the same in Armbian mainline kernel. LCD works with legacy rockchip kernel.

Kenzu, 2024-01-01:
Some more info:
root@DietPi:~# cat /sys/class/drm/card0-HDMI-A-1/status
connected
root@DietPi:~# cat /sys/class/drm/card0-HDMI-A-1/modes
root@DietPi:~# cat /sys/class/drm/card0-HDMI-A-1/enabled
disabled
root@DietPi:~#

disablewong, 2024-01-07:
Hello all! I tried the image and the on board wifi is not working either, nothing for wifi was detected through lspci and lshw. Maybe the driver for that is not there is the kernel?

GremlineQ, 2024-01-07:
> Hello all! I tried the image and the on board wifi is not working either, nothing for wifi was detected through lspci and lshw. Maybe the driver for that is not there is the kernel?

from what I've been looking, the driver for this wireless WiFi card is at the kernel level

MichaIng, 2024-01-07:
@Kenzu 
Hmm, so EDID info shows 1024x600 is supported but kernel driver reports 1024x600 is not supported 🤔. Did you try to skip depth and frequency, hence only `video=HDMI-A-1:1024x600`? AFAIK the others are optional.

> from what I've been looking, the driver for this wireless WiFi card is at the kernel level

Drivers on Linux are always kernel drivers 😉. Let's see whether the driver really is missing, or just the firmware, or it fails to load. Are there any kernel errors?
```sh
dmesg -l 0,1,2,3
```
I am not sure how it is internally attacked, hence it might be expected that `lspci` does not show it.

Also check `dmesg` whether there is any other info regarding detected (WiFi) devices.

Kenzu, 2024-01-07:
Yes i did try without -24@60 and a lot of other combinations, but they all give:

```
[    3.069590] rockchip-drm display-subsystem: [drm] User-defined mode not supported: "1024x600": 60 51200 1024 1192 1224 1344 600 615 621 635 0x68 0x5
[    3.070903] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
```

I don’t have time today to look at WiFi, but it works in armbian so we might be able to find the needed info there.

Jesper

Kenzu, 2024-01-07:
https://forum.armbian.com/topic/26315-armbian-build-pr-consolidate-uwe5622-driver-add-v61-kernel-support/

Some info about WiFi chipset and driver.

disablewong, 2024-01-07:
Um.... Actually i had tried to install dietpi to the board through script from armbian and it was succeeded.
https://dietpi.com/forum/t/orange-pi-3b-problem/18008/6?u=joulinar

The wifi was working with that image and most of the function i use seems to be normal.  However, when i try to move it to my optane drive at e-key pcie slot then it failed to boot no matter how i flash the spi... It is strange....

When trying to boot the optane drive with official image from orangepi and the test image from dietpi, they were able to boot successfully.  But then it seems the wifi was dropped....

Dmesg, lspci, lshw, iw and even lsmod shows no wireless interface, i think it suggest the driver was not compiled into kernel, not the firmware issues

MichaIng, 2024-01-07:
Armbian provides "edge" images only, so probably it works with edge kernel, but not with "current" kernel. Can someone test this on a DietPi image (not converted):
```sh
apt install linux-{image,dtb}-edge-sunxi64
reboot
```

MagicShrimp22, 2024-01-09:
I tried that command on the bookworm and trixie images of dietpi for the 3b. After the reboot command my device gets stuck on a black screen and won't boot up again.

disablewong, 2024-01-09:
> I tried that command on the bookworm and trixie images of dietpi for the 3b. After the reboot command my device gets stuck on a black screen and won't boot up again.

maybe you can try to rescue the system through mounting the sd card on another linux machine, then editing /boot/extlinux/extlinux.conf .  Change the default to original kernel and try to boot it again.

MagicShrimp22, 2024-01-10:
> maybe you can try to rescue the system through mounting the sd card on another linux machine, then editing /boot/extlinux/extlinux.conf . Change the default to original kernel and try to boot it [...]

I ended up using the image you shared in another thread and everything works, so I'm sticking with that one.

disablewong, 2024-01-11:
I am glad to hear that, however, that image cannot boot my optane disk.... That's annoyed me... I would try to get an updated image for that...

jake5253, 2024-01-11:
I used the Bookworm version from 12/31/23 that's available on https://dietpi.com/downloads/images/testing/ and from what I can tell it's working fine.  Built-in ethernet works straight out of the gate, hdmi thus far has worked fine.  Did you guys erase your old u-boot from SPI before trying DietPi?  My guess is you're booting with the wrong u-boot and you need to wipe that out. 
HDMI problems 9 times outta 10 are caused by P.O.S. cables. Try a better cable?
WiFi doesn't work because the wifi module doesn't get loaded.
type `modprobe sprdwl_ng`
now you have WiFi. You can use a systemd service to do that at boot-time.
from Armbian: [See Here](https://github.com/armbian/build/blob/main/packages/bsp/sunxi/aw859a-wifi.service)
```
[Unit]
Description=WIFI AW859A

[Service]
Type=forking
ExecStart=/sbin/modprobe sprdwl_ng
TimeoutSec=0
RemainAfterExit=yes
SysVStartPriority=99

[Install]
WantedBy=multi-user.target
```

Bluetooth might not work. I didn't test it, I have no need for it. 
But, I saw info about it while fixing WiFi so, i'll include what I know.
```
# Bluetooth on orangepi3b board is handled by a Spreadtrum (sprd) chip and requires
# a custom hciattach_opi binary, plus a systemd service to run it at boot time
install -m 755 hciattach_opi /usr/bin
cp sprd-bluetooth.service /lib/systemd/system/
.......
systemctl enable sprd-bluetooth.service

```
Those files are here:
# https://github.com/armbian/build/blob/main/packages/bsp/rk3399/hciattach_opi
# https://github.com/armbian/build/blob/main/packages/bsp/rk3399/sprd-bluetooth.service

disablewong, 2024-01-11:
> Built

Nice to learn that wifi is back! But it sound strange that I need to load the wifi manually as they are usually detected and loaded during boot-up. Anyways it at least got a workaround! Thanks for sharing!

jake5253, 2024-01-11:
I agree that a systemd service to load the wifi module seems... excessive. But that's the way Armbian does it.

F.W.I.W: the official orangepi-build uses a "more normal(?)" method of doing it; they add the following two entries in `/etc/modules`:
```
sprdbt_tty
sprdwl_ng
```

In case you want to see more about their method, have a look at [scripts/distributions.sh Line 74](https://github.com/orangepi-xunlong/orangepi-build/blob/next/scripts/distributions.sh#L74) (for the /etc/modules file) and [line 797](https://github.com/orangepi-xunlong/orangepi-build/blob/next/scripts/distributions.sh#L797) (Where they (possibly?) add it to initrd. Unverified if it's added to initrd)

The official orangepi-build appears to use a systemd service for the bluetooth helper app, much like  Armbian. Interestingly, the services are setup quite differently between the two.
[See here](https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/packages/bsp/sunxi/aw859a-bluetooth.service)

MichaIng, 2024-01-11:
Ah nice, so it is the same as on Orange Pi Zero 3, where we add these modules already when WiFi/BT is enabled. The `_tty` one is for Bluetooth only, btw. Yeah, the systemd unit from Armbian is kinda overkill given one can use a `/etc/modules-load.d` config to have it done more naturally, but in the end both does exactly the same.

I'll have the code we already use for Orange Pi Zero 3 running for Orange Pi 3B then as well, probably with a condition in both cases that the module must exist, so that we can move to mainline kernel without having the scripts broken.

MagicShrimp22, 2024-01-11:
Can confirm that using "modprobe sprdwl_ng" on a fresh install of the bookworm test image enables wifi. Still had to use Ethernet to set it up but all is working now.

MichaIng, 2024-01-20:
Onboard WiFi/BT will work with next version: https://github.com/MichaIng/DietPi/commit/d53811d

About the potential SPI bootloader issue: Those where the image does not boot (at all, not just blank HDMI), do you have a USB-UART adapter to check serial console output? Probably there is some syntax in our `boot.scr` which is not understood by the vendor bootloader and which we can adjust. We did so for Odroid N2/C4 as well with their strange pre-installed petitboot SPI bootloader.

nyx-rattapoom, 2024-01-22:
Last time I tried, armbian base seems like it have a problem with packet loss when running ookla speedtest, but this problem does not occur with the official build. @disablewong Can you confirm this problem still exists ?

MichaIng, 2024-03-05:
I'll mark this as closed. Orange Pi 3B images have been added to download page. For any existing or new problem, please open a new issue.

closed 2024-03-05 by MichaIng

Kenzu, 2024-03-07:
How would I go about making a dietpi image with linux-rk35xx-legacy kernel 5.10.160 from armbian? 
That works with my 7” lcd in 1024x600.

Thanks

MichaIng, 2024-03-07:
You can just install the legacy kernel afterwards:
```sh
apt install linux-{image,dtb}-rk35xx-legacy
reboot
```
In case, remove the "current" one:
```sh
apt autopurge linux-{image,dtb}-rk35xx-current
```

MichaIng, 2024-04-06:
Does NVMe work for you guys?

https://dietpi.com/forum/t/19827

GremlineQ, 2024-04-06:
> Does NVMe work for you guys?
> 
> https://dietpi.com/forum/t/19827

Not for me.

Katarn, 2024-04-06:
It worked for me after installing nvme-utils (I don’t remember exactly the name of the package and can’t check it now) and rebooting.

MichaIng, 2024-04-06:
You mean `nvme-cli`? Hmm, this is for managing and monitoring NVMe drives, but not a driver for general access of NVMe drives. No other NVMe-capable SBC requires userland tools or libraries to access and mount NVMe drives 🤔.

GremlineQ, 2024-04-06:
maybe it's just my problem, I don't know, I have one WD Black sn720 drive, it is not detected in general (lspci or any other command does not display it) but it also heats up, and the second drive is adata sx8200pro which is detected by lspci but not by others commands, e.g. those that display partitions and disks, also get warm

Katarn, 2024-04-06:
Yes, `nvme-cli`. I was surprised myself, but in my case it worked. SSDs from WD and Gudga began to be detected, before that even `lspci` had not seen them.

Katarn, 2024-04-08:
No, everything was fine with the physical connectivity. I checked on two different SSD. I also tested on Armbian - and it was after Armbian detected it that I realized the device was working, so I went back to my research with DietPi.

disablewong, 2024-04-11:
Sorry just read this email.
I am using Optane m10 on my orangepi 3b and it works well for me.
I am using it as the boot disk.