MichaIng/DietPi #6662: '99-dietpi-pihole-block_public_admin.conf' blocks access to Pi-hole dashboard in IPv6 enabled environment

maartenlangeveld, 2023-10-03:
### Creating a bug report/issue

#### Required Information
Concerns DietPi's  '99-dietpi-pihole-block_public_admin.conf' lighttpd script that is installed when installing Pi-hole usingDietPi's dietpi-software install function.

#### Additional Information (if applicable)
Above mentioned script breaks access to pi-hole's admin web dashboard when browsing to 'pi.hole/' in a environment that supports ipv6.

#### Steps to reproduce
1. Enable ipv6 on pi-hole server and host browsing Pi-hole's admin web dashboard;
2. Browse to the dashboard using 'pi.hole/' as web address in the internet browser.

#### Expected behaviour
- Access to Pi-hole admin web dashboard

#### Actual behaviour
- Lighttpd response with '403 Forbidden', hence no access.

#### Extra details
For some (unknown) reason Pi-hole's pihole-FTL (dnsmasq) resolver responds to AAAA query for pi.hole with it's global IPv6 address in stead of it's link-local IPv6 address (for all other local hosts pihole-FTL repsonds with their respective link-local IPv6 addresses, which seems logical). Then DietPi's 99-dietpi-pihole-block_public_admin.conf lighttpd scripts blocks access.

```
nslookup pi.hole

Server: pi.hole
Address: fe80::w:x:y:z

Name: pi.hole
Addresses: r:s:t:u:w:x:y:z
           192.168.0.2
```

MichaIng, 2023-10-08:
Yes, this is an expected limitation when you connect via IPv6 and without ULAs. The filter is done via IP address ranges, and excludes all non-local/LAN IPv4 and IPv6 addresses. For IPv4 this is reliable, as every LAN node has a LAN IPv4 address, preferred for accesses within the LAN. But in case of IPv6, the only non-global IP assured to be present is the LLA (Local-Link Addresses ranging `fe80:*` - `febf:*`), which can only be used to connect to clients directly linked to, i.e. any rounter/switch between two links may break it, and in this case the GUA is used, which is a public/global IPv6 address, and hence treated as public access.

The workaround for this is to either enable ULAs (Unique Local Addresses ranging `fc00:*` - `fdff:*`) in your router, or connect via IPv4 within your LAN. Or to enforce a local IPv6 address in your local DNS resolver, which is what you did already to solve the issue (reading the Pi-hole forum topic) 👍.

Is this issue solved then, or do you still face issues?

We could add a clear info about this, or a hint in our docs (as it is too much to explain properly in a console dialogue prompt.

maartenlangeveld, 2023-10-08:
Hi @MichaIng,

Thanks, issue is solved. I have configured Pi-hole by setting the LLA address as fixed LOCAL_IPV6.
This issue maybe closed.

MichaIng, 2023-10-09:
Okay, marking this as closed.

closed 2023-10-09 by MichaIng