MichaIng/DietPi #6664: visionfive 2 wont boot lastest firmware

ardenking, 2023-10-04:
### Creating a bug report/issue

Updated bootloader on visionfive 2 now not starting dietpi image from SD card was working before update

Debian boots fine on same SD card so its the image with this bootloader not working 

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 5A RAVpower)
- SD card used | ( SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... update bootloader 
2. ...try booting from SD

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2023-10-04:
Indeed. I'm working on an own U-Boot environment to fit this. If you have serial console access, here is an example of how to enable nice generic boot targets: https://github.com/MichaIng/DietPi/discussions/6641

MichaIng, 2023-10-19:
The current [beta](https://github.com/MichaIng/DietPi/issues/6687) applies a major kernel upgrade, including support for accessing and editing the U-Boot environment and a reasonable default environment.

If you have a UART adapter to access the serial console, you can boot with the new bootloader via:
```sh
setenv devnum 1
run mmc_boot
```
Then apply the DietPi update, in case for the beta:
```sh
G_DEV_BRANCH beta
reboot
```
Repeat above steps on serial console, then from within the DietPi system, you can apply the U-Boot environment:
```sh
fw_setenv
```
From then on, all boot media are supported and booted from in this priority:
- SD card > USB > NVMe > eMMC > DHCP/TFTP

MichaIng, 2023-10-21:
On update to DietPi v8.23 as well as when trying to update the bootloader via `dietpi-config`, one will be informed about the necessity and offered to flash the new U-Boot environment. I'll close this issue. Further discussion, feedback etc here: #6212

closed 2023-10-21 by MichaIng