MichaIng/DietPi #6666: Raspberry Pi Zero 2 W - bluetoothctl problem

Rhiz3K, 2023-10-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- G_DIETPI_VERSION_CORE=8
- G_DIETPI_VERSION_SUB=22
- G_DIETPI_VERSION_RC=3
- G_GITBRANCH='master'
- G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bookworm 0

- Kernel version | `uname -a`
- Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr   3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- RPi Zero 2 W (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
- AXAGON ACU-DPQ65

- SD card used | (EG: SanDisk ultra)
- SanDisk MicroSDHC 32GB Extreme

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- bluetoothctl
- Was the software title installed freshly or updated/migrated?
- Fresh installation
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. dietpi-config / Advanced Options / Bluetooth set from Off to On then restart
2. bluetoothctl

#### Expected behaviour

- Agent registered
- [bluetooth]#

#### Actual behaviour

- after bluetoothctl is showing only Waiting to connect to bluetoothd....

#### Extra details

- journalctl -u bluetooth
- It showed: configurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
- Used:
- sudo chmod 555 /etc/bluetooth
- sudo systemctl restart bluetooth
- Which fixed the issue

MichaIng, 2023-10-06:
Can you check the hciuart service logs?
```sh
journalctl -u hciuart
```

Rhiz3K, 2023-10-06:
> Can you check the hciuart service logs?
> 
> ```shell
> journalctl -u hciuart
> ```

-- No entries --

MichaIng, 2023-10-06:
Strange, that service is forcefully started and enabled when enabling Bluetooth on RPi via `dietpi-config`. Does this work?
```sh
systemctl enable --now hciuart
sleep 2
journalctl -u hciuart
```

Rhiz3K, 2023-10-07:
> Strange, that service is forcefully started and enabled when enabling Bluetooth on RPi via `dietpi-config`. Does this work?
> 
> ```sh
> 
> systemctl enable --now hciuart
> 
> sleep 2
> 
> journalctl -u hciuart
> 
> ```

New fresh install done today:

root@DietPi:~# bluetoothctl
Waiting to connect to bluetoothd...
root@DietPi:~# systemctl enable --now hciuart
sleep 2
journalctl -u hciuart
Unit /lib/systemd/system/hciuart.service is added as a dependency to a non-existent unit dev-serial1.device.
Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xeu hciuart.service" for details.
Oct 07 10:50:52 DietPi systemd[1]: Starting hciuart.service - Configure Bluetooth Modems connected by UART...
Oct 07 10:50:52 DietPi btuart[985]: Can't open serial port: No such file or directory
Oct 07 10:50:52 DietPi btuart[985]: Can't initialize device: No such file or directory
Oct 07 10:50:52 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Oct 07 10:50:52 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Oct 07 10:50:52 DietPi systemd[1]: Failed to start hciuart.service - Configure Bluetooth Modems connected by UART.
root@DietPi:~#

Rhiz3K, 2023-10-07:
Also dbus.service is not active, cannot reload. appeared in red when tried to turn off/on bluetooth again:

Setting up dbus-bin (1.14.8-2~deb12u1) ...
Setting up dbus-daemon (1.14.8-2~deb12u1) ...
Setting up dbus (1.14.8-2~deb12u1) ...
dbus.service is a disabled or a static unit, not starting it.
Setting up bluez (5.66-1) ...
dbus.service is not active, cannot reload.
invoke-rc.d: initscript dbus, action "force-reload" failed.
Created symlink /etc/systemd/system/dbus-org.bluez.service → /lib/systemd/system/bluetooth.service.
Created symlink /etc/systemd/system/bluetooth.target.wants/bluetooth.service → /lib/systemd/system/bluetooth.service.
Setting up pi-bluetooth (0.1.19) ...
Created symlink /etc/systemd/system/dev-serial1.device.wants/hciuart.service → /lib/systemd/system/hciuart.service.
Unit /lib/systemd/system/hciuart.service is added as a dependency to a non-existent unit dev-serial1.device.
Processing triggers for libc-bin (2.36-9+deb12u3) ...

MichaIng, 2023-10-07:
The required secondary serial device is missing. Can you check the following:
```sh
ls -l /dev/{ttyAMA0,ttyS0,serial0,serial1}
```

Rhiz3K, 2023-10-07:
ls -l /dev/{ttyAMA0,ttyS0,serial0,serial1}
ls: cannot access '/dev/serial0': No such file or directory
ls: cannot access '/dev/serial1': No such file or directory
crw-rw---- 1 root dialout 204, 64 Oct  7 11:31 /dev/ttyAMA0
crw-rw---- 1 root dialout   4, 64 Oct  7 11:31 /dev/ttyS0

MichaIng, 2023-10-08:
I can replicate the issue of the missing `serial0`/`serial1` symlinks. Currently investigating the reason for this.

MichaIng, 2023-10-08:
Fixed with: https://github.com/MichaIng/DietPi/commit/0812ad7

Please do this:
```sh
curl -fo package.deb 'https://dietpi.com/downloads/binaries/raspberrypi-sys-mods.deb'
dpkg -i --force-confdef,confold package.deb
rm package.deb
```

Rhiz3K, 2023-10-08:
Working! Thank you very much for superb support!

closed 2023-10-21 by MichaIng