MichaIng/DietPi #6667: Failed to start systemd-timesyncd.service - Network Time Synchronization

mattsmithuk, 2023-10-06:
### Creating a bug report/issue
Occurs when trying to to run `dietpi-update` on ASUS Tinkerboard and on NanoPi Neo
Aiming to update from 8.20.1 to 8.22.3

#### Required Information
- DietPi version | 8.20.1
- Distro version | bookworm
- Kernel version |  `6.1.11-rockchip #23.02.2 SMP PREEMPT Fri Feb 17 23:55:05 UTC 2023 armv7l GNU/Linux` and `5.15.93-sunxi #23.02.2 SMP Fri Feb 17 23:49:46 UTC 2023 armv7l GNU/Linux`
- SBC model | ASUS Tinker Board (armv7l) and NanoPi NEO (armv7l)

#### Additional Information (if applicable)
- Software title | systemd-timesync
- Was the software title installed freshly or updated/migrated? No
- Can this issue be replicated on a fresh installation of DietPi? Not attempted
- Bug report ID | 37cf6590-3db8-44b2-9c5f-ba614a273899

#### Steps to reproduce

1. Run dietpi-update on either device
2. Alternatively, run `sudo systemctl start systemd-timesyncd.service`

#### Expected behaviour

- systemd-timesyncd should run
#### Actual behaviour

```
systemd-timesyncd.service: Start request repeated too quickly.
systemd-timesyncd.service: Failed with result 'exit-code'.
Failed to start systemd-timesyncd.service - Network Time Synchronization
```

#### Extra details
- Attempting reinstalling systemd-timesyncd via apt on both devices - does not resolve issue
- Tried running directly from root, rather than sudo - did not resolve issue

Joulinar, 2023-10-06:
pls can you reboot your system, wait for a couple of minutes and run following

`journalctl -u systemd-timesyncd.service`

mattsmithuk, 2023-10-06:
That seems to have worked. Thanks!

Joulinar, 2023-10-06:
OK I'm going to close the issue. Feel free to reopen if needed.

closed 2023-10-06 by Joulinar