MichaIng/DietPi #6668: [FAILED] DietPi-Software | Installing Python with Home Assistant module into pyenv

Martin-Fox, 2023-10-06:
### Creating a bug report/issue

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Oct  6 12:32:19 BST 2023
- Bug report     | 88babe09-f4ac-4fff-b802-72ec51cb8205
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `sudo -u homeassistant dash -c 
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.11.5
pyenv local 3.11.5
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install 
exec pip3 install homeassistant`
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Downloading Python-3.11.5.tar.gz...
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm

- Kernel version | `uname -a`

Linux DietPi 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

Native PC (x86_64)

#### Additional Information (if applicable)
- Software title | Home Assistant

- Was the software title installed freshly or updated/migrated?

I tried to install it

- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

88babe09-f4ac-4fff-b802-72ec51cb8205

#### Steps to reproduce

sudo dietpi-software install 157

#### Expected behaviour

HA should be installed

#### Actual behaviour

Error

Martin-Fox, 2023-10-06:
After reboot it was fixed by itself, this time installation was succesfull.

closed 2023-10-06 by Martin-Fox

MichaIng, 2023-10-06:
Looks like the download failed, though strange that there is no other error message. Did you "Retry" once from the error handler and ran into the same? And did it take long until the error handler showed up or was it immediately?

Martin-Fox, 2023-10-11:
@MichaIng Yes, I tried few times. Every time it was the same result. And it took some time, it wasn't instant.

MichaIng, 2023-10-11:
Probably the server, or a particular mirror was faulty, or an unexpected high RAM usage during the first install attempt, so that RAM filled up completely while the file was downloaded to `/tmp` tmpfs. However, good that a reboot solved it.