MichaIng/DietPi #6669: Uptime-Kuma

msmhome, 2023-10-06:
### Creating a software request

#### Formal software information
- Software title | Uptime Kuma
- Short description | Self-hosted uptime monitoring service
- Official URL (if available) | https://uptime.kuma.pet/
- GitHub URL (if available) | https://github.com/louislam/uptime-kuma
- (Official) install documentation | https://github.com/louislam/uptime-kuma/wiki

#### Are there similar/alternative software titles available with DietPi-Software?
Does not appear to be any uptime monitoring services like this, but there are titles like Prometheus available.

#### What makes your requested software better than the above solutions, if available?
Fully open source, easy-to-use webui, and intuitive easy config. 

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
Besides the docker options, there are [instructions ](https://github.com/louislam/uptime-kuma/wiki/%F0%9F%94%A7-How-to-Install)for a metal install requiring just npm to spin up the webserver. There are no databases to be set up.  

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
See [instructions ](https://github.com/louislam/uptime-kuma/wiki/%F0%9F%94%A7-How-to-Install).

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

Yes, I am happy to help maintain it and even can help put it into dietpi if needed.

Joulinar, 2023-10-06:
You are welcome to start implementation. We will support to review the PR.

aeonoea, 2024-07-17:
Any updates on this? Such a powerful and useful little tool, and it is offered on two other projects (CasaOS and UmbrelOS), so would be great to have it also available on DietPi systems.

Joulinar, 2024-07-17:
> and it is offered on two other projects (CasaOS and UmbrelOS)

If I'm not mistaken, both use Docker and don't install the native application. Usually, we don't install Docker app directly as it is quite easy for end user to do it themselves.

Following will simulate a DietPi script 😄 

```
#!/bin/bash
. /boot/dietpi/func/dietpi-globals

# Check for existing Kuma container
eval container=$(docker container ls -a | mawk '/louislam\/uptime-kuma/{print $1;exit}')
[[ $container ]] && G_EXEC docker container rm -f "$container"
eval image=$(docker image ls -a | mawk '/louislam\/uptime-kuma/{print $3;exit}')
[[ $image ]] && G_EXEC docker image rm "$image"
docker volume ls -q | grep -xq 'kuma_data' || G_EXEC docker volume create kuma_data

# Deploy the Kuma container
G_DIETPI-NOTIFY 2 'Kuma Uptime will be deployed now. This could take a while...'
G_EXEC_OUTPUT=1 G_EXEC docker run -d -p '3001:3001' --name uptime-kuma --restart=always  -v 'kuma_data:/app/data' 'louislam/uptime-kuma:1'
```

aeonoea, 2024-07-17:
@Joulinar Thanks for the quick response! Yes, you're completely right with Docker vs. native install. Unfortunately the script is beyond my expertise, so will have to hope and wait for official support via dietpi-software sometime 🙏

Joulinar, 2024-07-17:
> Unfortunately the script is beyond my expertise

Simply copy the script if you already running Docker. It will do all the magic. Even container updates can be carried out afterwads.

aeonoea, 2024-07-17:
Oh, I'll try it but first will have to install Docker 😃