MichaIng/DietPi #6670: The AutoStart script custom.sh only execute the first command instead of all commands

tindecken, 2023-10-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 0

- Kernel version | `uname -a`

Linux Tindecken 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

 5V 1A RAVpower

- SD card used | (EG: SanDisk ultra)

SanDisk

#### Additional Information (if applicable)
AutoStart Options --> Custom Script (background, no autologin)

#### Steps to reproduce
1. Go to AutoStart Option --> Custom Script (background, no autologin)
2. add 2 new lines in custom.sh as below

root@Tindecken:~# cat /var/lib/dietpi/dietpi-autostart/custom.sh
#!/bin/bash
# DietPi-AutoStart custom script
# Location: /var/lib/dietpi/dietpi-autostart/custom.sh
sqlite_web -H 192.168.1.17 -p 9000 tindcv.db
smee -u https://smee.io/dT9mD6hh6Vj1wX0 --path /github-webhook/ --port 8080
exit 0
root@Tindecken:~#

3. reboot
4. htop to check

[image]

#### Expected behaviour
It will execute 2 commands: sqlite_web and smee

#### Actual behaviour
It only execute the first command: sqlite_web
If I swap the 2 commands: it only execute the first command: smee

MichaIng, 2023-10-07:
Please try to run the script from console and see whether it behaves different. The autostart option just does this from within a systemd service:
```sh
/var/lib/dietpi/dietpi-autostart/custom.sh
```
So there is not really something which can go wrong, unless it the script itself does something wrong.

tindecken, 2023-10-08:
Thanks @MichaIng  for the support, you're correct, the process locked each other, so the 2nt never run

I'm running in background as mentioned on step 1

[image]

Adding trailing `&"  --> works well

[image]

MichaIng, 2023-10-08:
Great! I'll mark this as closed then, feel free to reopen if required.

closed 2023-10-08 by MichaIng