MichaIng/DietPi #6671: Raspberry pi zero 2 w - tailscale

Rhiz3K, 2023-10-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 0
- Kernel version | `uname -a`

Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero 2 W (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
tailscale from dietpi-software
- Was the software title installed freshly or updated/migrated?
fresh during fresh install of os
- Can this issue be replicated on a fresh installation of DietPi?
yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. during 1st boot installation or after from dietpi-software
2. tailscale up

#### Expected behaviour

- generate link to verify account and connect device to your tailscale acc

#### Actual behaviour

- tailscale up
failed to connect to local tailscaled (which appears to be running as tailscaled, pid 749). Got error: Failed to connect to local Tailscale daemon for /localapi/v0/status; systemd tailscaled.service not running. Error: dial unix /var/run/tailscale/tailscaled.sock: connect: no such file or directory

#### Extra details

- user: root
- wlan connection

MichaIng, 2023-10-07:
I guess a duplicate of: #6649
Please try:
```sh
apt install iptables
systemctl restart tailscaled
```

Rhiz3K, 2023-10-08:
> I guess a duplicate of: #6649
> 
> Please try:
> 
> ```sh
> 
> apt install iptables
> 
> systemctl restart tailscaled
> 
> ```

It worked, thank you.

closed 2023-10-08 by MichaIng