MichaIng/DietPi #6672: Error while installing node-red

jawa634, 2023-10-08:
#### Details:
- Date           | Sun Oct  8 18:33:40 BST 2023
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux DietPi 4.9.312-arm64 #1 SMP PREEMPT Wed Nov 23 00:32:45 CET 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `sudo -u nodered npm i --no-audit node-red`
- Exit code      | 7
- Software title | DietPi-Software
#### Steps to reproduce:
try to install node-red
dietpi-software install 122
#### Expected behaviour:
should install node red
#### Actual behaviour:
[  OK  ] DietPi-Software | chown -R nodered:nodered /mnt/dietpi_userdata/node-red
[  OK  ] DietPi-Software | cd /mnt/dietpi_userdata/node-red
[ INFO ] DietPi-Software | sudo -u nodered npm i --no-audit node-red, please wait...
/usr/local/lib/node_modules/npm/lib/es6/validate-engines.js:31
    throw err
    ^

TypeError: Class extends value undefined is not a constructor or null
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/fs-minipass/lib/index.js:136:4)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
    at Module.load (node:internal/modules/cjs/loader:1091:32)
    at Module._load (node:internal/modules/cjs/loader:938:12)
    at Module.require (node:internal/modules/cjs/loader:1115:19)
    at require (node:internal/modules/helpers:130:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/cacache/lib/content/read.js:4:13)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)

Node.js v20.8.0
[FAILED] DietPi-Software | sudo -u nodered npm i --no-audit node-red

#### Extra details:

- ...
#### Additional logs:
```
/usr/local/lib/node_modules/npm/lib/es6/validate-engines.js:31
```

Thanks a lot for your response!

MichaIng, 2023-10-08:
Based on your pre-image, I guess you had Node-RED installed before, or another Node.js application?

It looks like the (local) Node-RED module, respectively one of its dependencies, conflicts with one of the globally installed ones. Some years ago, we installed Node-RED as global module, hence outdated dependencies might still be installed there.

If you have no other Node.js apps, you could remove all global modules. But its probably safer to try upgrading the global `fs-minipass`:
```sh
sudo npm up -g fs-minipass
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm i --no-audit node-red
```

jawa634, 2023-10-12:
hello, yes you were right. There was an second node-red installed. After reinstalling node.js everthing works.
Thanks a lot!

closed 2023-10-12 by jawa634