MichaIng/DietPi #6673: Tailscale doesn't start on new installation

Kreeblah, 2023-10-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.22.3`
- Distro version | `bookworm 0`
- Kernel version | `Linux ntp 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Official Raspberry Pi power supply
- SD card used | SanDisk Extreme

#### Additional Information (if applicable)
- Software title | Tailscale
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? I haven't tried, but this is very nearly a fresh installation, so I'd suspect so.

- Bug report ID | `5ac5ec45-2d97-4b86-9c04-7f9f21d12ffe`

#### Steps to reproduce

1. Install Tailscale via `dietpi-software`
2. Run `tailscale up`

#### Expected behaviour

- I'd expect Tailscale to come up.

#### Actual behaviour

- I'm receiving the following error:
```
failed to connect to local tailscaled; it doesn't appear to be running (sudo systemctl start tailscaled ?)
```

#### Extra details

- Trying to run `systemctl start tailscaled` fails, and running `systemctl status tailscaled` has this:

```
× tailscaled.service - Tailscale node agent
     Loaded: loaded (/lib/systemd/system/tailscaled.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-10-08 20:44:27 UTC; 1min 23s ago
   Duration: 432ms
       Docs: https://tailscale.com/kb/
    Process: 3366 ExecStartPre=/usr/sbin/tailscaled --cleanup (code=exited, status=0/SUCCESS)
    Process: 3378 ExecStart=/usr/sbin/tailscaled --state=/var/lib/tailscale/tailscaled.state --socket=/run/tailscale/tailscaled.sock --port=${PORT} $FLAGS (code=exited, status=1/FAILURE)
    Process: 3397 ExecStopPost=/usr/sbin/tailscaled --cleanup (code=exited, status=0/SUCCESS)
   Main PID: 3378 (code=exited, status=1/FAILURE)
        CPU: 603ms

Oct 08 20:44:27 ntp systemd[1]: tailscaled.service: Scheduled restart job, restart counter is at 6.
Oct 08 20:44:27 ntp systemd[1]: Stopped tailscaled.service - Tailscale node agent.
Oct 08 20:44:27 ntp systemd[1]: tailscaled.service: Start request repeated too quickly.
Oct 08 20:44:27 ntp systemd[1]: tailscaled.service: Failed with result 'exit-code'.
Oct 08 20:44:27 ntp systemd[1]: Failed to start tailscaled.service - Tailscale node agent.
```

MichaIng, 2023-10-08:
See her: #6649 
```sh
apt install iptables
systemctl restart tailscaled
```

Kreeblah, 2023-10-08:
Ah, yep, that was it.  I didn't think to check whether it was the `iptables` issue since they'd marked theirs as completed.

Sorry about that.

closed 2023-10-08 by Kreeblah

MichaIng, 2023-10-08:
No problem. Yes it is resolved in their code, but a new version + package has no yet been released.