MichaIng/DietPi #6675: Orange PI 5 plus fails to get an IP address

xaja34, 2023-10-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=22
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
root@DietPi:~#`
- Distro version | bookworn
- Kernel version | Linux DietPi 5.10.160-legacy-rk35xx #1 SMP Mon Aug 28 01:21:24 UTC 2023 aarch64 GNU/Linux
- SBC model | Orange PI 5 Plus
- Power supply used | 5V 5 amp
- SD card used | Sandisk

#### Additional Information (if applicable)
- Software title | N/A
- Can this issue be replicated on a fresh installation of DietPi? | YES

#### Steps to reproduce
Every 3rd or 4th boot, system fails to get an IP address
dietpi-config, Network options:adapters, Ethernet shows disconnected
IP address 0.0.0.0

ip a shows 192.168.0.238 which is correct
This happens with a static IP or DHCP

journalctl -u ifup@eth0.service
Oct 09 07:33:15 DietPi systemd[1]: Starting ifup@eth0.service - ifup for eth0...
Oct 09 07:33:15 DietPi systemd[1]: Finished ifup@eth0.service - ifup for eth0.

By removing the network cable and pluggng it into the 2nd network port the eth0 gets assigned an IP address

#### Extra details
This looks like the same as issue #6592

dirkhh, 2023-10-10:
since you already mention that this is a duplicate of #6592, why not just add details there instead of creating a new issue?

MichaIng, 2023-10-10:
Yes, seems to be the same indeed. I'll mark this as closed, let's discuss in the other topic.

closed 2023-10-10 by MichaIng