MichaIng/DietPi #6676: Raspberry Pi 5

TDuffinNTU, 2023-10-09:
- Device name | Raspberry Pi 5
- Official product URL | https://www.raspberrypi.com/products/raspberry-pi-5/
- GitHub resource URL (if available) | https://github.com/raspberrypi
- Image download URL |https://www.raspberrypi.com/software/

#### Is the SBC officially supported by the Debian installer?
- Yes(?) Looks like the board is inter-compatible with the same distro images as the 4B. That's pure speculation, mind you, but I did see a video of a reviewer transferring a Raspbian installation from a Pi4B to a Pi5 with no configuration necessary.

#### Notes
- Not sure if this needs any specific developer legwork as the previous Pis seem to be inter-compatible with OS images for the most part (though I'm sure testing and drivers will be necessary!). Thought I'd make this issue now since it might be good to mark the Pi5 as _provisionally_ supported on the website if it at least works out of the box.

MichaIng, 2023-10-19:
Okay this requires more work, sadly. With Bookworm and hence for RPi 5, the packages as well as the filesystem layout have changed dramatically:
- For RPi 4 and earlier, the package is called `linux-image-rpi-v8`, which is a meta package and current pulls `linux-image-6.1.0-rpi4-rpi-v8`.
- For RPi 5, the package is called `linux-image-rpi-2712`, which is a meta package and current pulls `linux-image-6.1.0-rpi4-rpi-2712`.
- These packages are more streamlined with mainline Linux packages, installing kernel image, config, map, device trees and overlays like common Debian Linux packages:
    ```
    /boot/System.map-6.1.0-rpi4-rpi-2712
    /boot/config-6.1.0-rpi4-rpi-2712
    /boot/vmlinuz-6.1.0-rpi4-rpi-2712
    ...
    /usr/lib/linux-image-6.1.0-rpi4-rpi-2712/broadcom/bcm2712-rpi-5-b.dtb
    ...
    /usr/lib/linux-image-6.1.0-rpi4-rpi-2712/overlays/README
    ```
- For the bootloader, there is now a `raspi-firmware` package, which installs the files as well more streamlined into `/usr/lib/raspi-firmware/`.
- `libraspberrypi0`, `libraspberrypi-bin` and `raspberrypi-sys-mods` remain the same.
- The RPi ROM bootloader stage still requires bootloader, config and kernel on a partition 1 with FAT filesystem, which was mounted to `/boot`. This cause the issue that, since `dpkg` does not support FAT, this `rpikernelhack` back and forth copying of all files in `/boot` was required, to bypass the need to `dpkg` to replace a file (which would fail). This procedure has now been replaced with another, IMO not less ugly workaround: The FAT filesystem of partition 1 is now mounted to `/boot/firmware`. The  `raspi-firmware` package strictly requires this directory to be a mount point (now allowing a dir within a mount point) and then copies all kernel, dtb and bootloader files from `/boot` and `/usr/lib/linux-image-` over to `/boot/firmware`, applying the known naming schemes. So a copy back and forth of all those files has been replaced with a single copy, but files remaining permanently on both partitions. And of course, for the 64-bit kernel, it is now actually two kernel images and all related modules, hence disk space requirements have increased.
- Furthermore, the `raspi-firmware` package does now strictly require an initramfs. I have to check whether this means that the kernel has essential modules no builtin anymore, or whether it is actually not required, but just installed and handled for in case. As far as I can see, there is a config flag to disable this copy to `/boot/firmware` stuff, which then would need to be done manually. So we could write our own script to do this, skipping the initramfs. But at least at the beginning, I am not keen to maintain such, as it will likely change a lot within the next months. Probably a `tiny-initramfs` is possible, like the moreless dummy one we do now use for Sparky SBC, one which does not include any kernel module, hence servers nearly no purpose (aside of supporting UUIDs on top of PARTUUIDs) aside of satisfying bootloader and/or package script expectations.
- Same for the 32-bit image, with the same kernel packages names and suffixes: `v6`, `v7`, `v7l` and `v8`.

Joulinar, 2023-10-20:
uhh does not sounds like a quick win. 🙄

axxis-creator, 2023-11-10:
Same Question. :-) I buy a RPI-5 and want to install DietPi for LoxBerry.
When can we expect a version?

gaurishhs, 2023-11-14:
any updates on this?

jboots07, 2023-11-16:
C04170

adrianog91, 2023-11-16:
Hi, i will receive the Raspberry next week. It is already shipped. I will update you when i receive it.

Very thanks for your amazing work!

gaurishhs, 2023-11-16:
I too would be receiving it in a week

MichaIng, 2023-11-16:
Thanks. Found it now here as well: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes

So "15" is the CM4S, which is a special industry customer variant, not publicly sold, and "16" for "internal use". Let's see whether "16" is a surprise, not announced yet 🙂.

MichaIng, 2023-11-16:
First step: https://github.com/MichaIng/DietPi/commit/6e848a1

jboots07, 2023-11-16:
Thanks for your hard work and update! Let me know if I can help in any way, I have a pi 5.

sorriso93, 2023-11-17:
I just received my PI5 and can't wait to install on it Dietpi to replace my PI4 :-)

m0jek, 2023-11-21:
@MichaIng when you are ready for testing in the wild on RaspberryPi 5, I am happy to test.

FabioEight, 2023-11-22:
Available for testing on Pi 5 as well! Thanks for your efforts!

elisenlebkuch, 2023-11-22:
Where can I get it?

gaurishhs, 2023-11-22:
I don't think so there has been any alpha release supporting Pi 5, You'll have to wait

elisenlebkuch, 2023-11-22:
Oh... This one won´t work? https://dietpi.com/downloads/images/testing/DietPi_Star64-RISC-V-Sid.img.xz[https://dietpi.com/downloads/images/testing/DietPi_Star64-RISC-V-Sid.img.xz](url)

gaurishhs, 2023-11-22:
I think that one is for Risc-V https://github.com/MichaIng/DietPi/issues/6212

LittleFreak, 2023-11-26:
Is it possible to add the dietpi later on a running raspbian?
I'll plan to docker all my programs anyway, so the host will stay pretty much raspbian lite until then.

MichaIng, 2023-11-26:
> Is it possible to add the dietpi later on a running raspbian?

You mean on a Raspberry Pi OS? Currently does not work as `dietpi-installer` installs the old kernel/bootloader/firmware packages and our scripts use `sed` in various places a way that the `/boot/config.txt => /boot/firmware/config.txt` (same with `cmdline.txt`) symlinks are replaced with the actual files, so that the intended changes will not actually apply. See my longer comment above about this issue. Once we added compatibility, `dietpi-installer` can run on a new RPi OS image, but it will remove all installed software, including Docker in your case. What you could do is backup your `/var/lib/docker`. Then once `dietpi-installer` ran through and you boot DietPi, do not install Docker at first, but restore `/var/lib/docker` to `/mnt/dietpi_userdata/docker-data`, then install Docker via `dietpi-software`.

cgmcfall, 2023-11-27:
Have a Pi5, happy to test :)

sergio-ingrao, 2023-12-01:
Here, I have another one ready to test.

Net0o, 2023-12-04:
I'm also available for testing.

bwims, 2023-12-09:
Just to clarify....

"[MichaIng](https://github.com/MichaIng) linked a pull request [2 days ago](https://github.com/MichaIng/DietPi/issues/6676#event-11187326565) that will close this issue"

Is that Github speak, or does it mean that the latest Dietpi image will work on a Pi 5?

(Sorry if that is a stupid question!)

B.

Joulinar, 2023-12-09:
> does it mean that the latest Dietpi image will work on a Pi 5

No

We are taking the first steps to adapt to the new firmware. For us, this means intensive testing, as we are making some significant code changes to our scripts. As a first step, we will provide a dedicated image for the RPi5 as soon as it is ready.

bwims, 2023-12-09:
Thanks for that.  I guess the standard Github dialogues are a bit
misleading in that respect! 😃

matttttttttttttt, 2023-12-09:
> No
>
> We are taking the first steps to adapt to the new firmware. For us, this means intensive testing, as we are making some significant code changes to our scripts. As a first step, we will provide [...]

Do you have any potential timeline that you could share with us? Your work is greatly appreciated!

LittleFreak, 2023-12-10:
> Do you have any potential timeline that you could share with us? Your work is greatly appreciated!

As stated above [https://wowwiki-archive.fandom.com/wiki/Soon](Soon™)

MichaIng, 2023-12-14:
I created a first alpha version of a firmware migration script. Would be great if you guys could test it on your RPi systems. With this done, it should also support the RPi 5, i.e. you should be able to swap the SD card (or whichever boot media you use) to the RPi 5 and boot it.

#### But you must apply the current DietPi v8.25 beta first: #6801

Joulinar, 2023-12-14:
Means you guys need another RPi where you can prepare the image to be able to run in RPi5. Correct @MichaIng ?

MichaIng, 2023-12-14:
For now yes. This was the first step, which I found more important than fresh RPi 5 compatible images: It allows testing our scripts with the new boot mountpoint and new kernel/firmware packages in general. This is also important for older RPi models, since their kernel did not receive an upgrade for too long, with at least one known bug open (Allo Boss2 device tree overlay) and a bunch of kernel vulnerabilities unfixed.

But I am working on fresh images as well.

Looking for your all opinion on something:
- The kernel packages are now split by SoC, i.e. one can install e.g. the RPi 1/Zero kernel only, skipping the others, which saves a lot of downloads, processing, disk writes/space etc, especially since now the kernel image is installed on the rootfs as well as additionally on the bootfs. And also an initramfs is now additionally generated for each kernel image (I want to check back whether this is really required, but currently it is a strick package dependencvy).
- I want to add a script which allows to select/deselect those kernel packages which are not required for the RPi model you are currently running.
- But the question is how we do it with new images:
  1. Install all kernel packages supported by the OS/userland architecture, so that the system card/drive can be swapped between RPi models like before.
  2. Install only 1 kernel package per image, i.e. providing 5 dedicated 32-bit images and 2 64-bit images, where especially the 32-bit images support only RPi models with the same SoC. When aiming to swap the SD card, it would then be required to first install the new kernel package, which would then be a `dietpi-config` option.

Isildur981, 2023-12-15:
> * But the question is how we do it with new images:
>
>       1. Install all kernel packages supported by the OS/userland architecture, so that the system card/drive can be swapped between RPi [...]

I vote for SoC and 32/64 bit specific images.  Most of the time people are either going to know ahead of time which RPi board they are going to be setting up, or have a computer/laptop with the different images on them and the correct one can then be written to a SD card.  

If you feel that it is going to be too much work to produce so many different images for the various RPi boards, I'm not opposed to a single image with all of the kernel packages.

v4run75, 2023-12-15:
```
* But the question is how we do it with new images:
  
  1. Install all kernel packages supported by the OS/userland architecture, so that the system card/drive can be swapped between RPi models like before.
  2. Install only 1 kernel package per image, i.e. providing 5 dedicated 32-bit images and 2 64-bit images, where especially the 32-bit images support only RPi models with the same SoC. When aiming to swap the SD card, it would then be required to first install the new kernel package, which would then be a `dietpi-config` option.
```

As a user of dietpit as long as I can restore the backups from my previous Pi (Pi 4) and carry it on to another one, any of the two options seems great. I don't generally swap out SD cards among different raspberry pis but have predetermined dedicated SD cards for each as @Isildur981 said

Trainax, 2023-12-15:
> But the question is how we do it with new images:
> 1. Install all kernel packages supported by the OS/userland architecture, so that the system card/drive can be swapped between RPi models like [...]

As stated by other users I think it would be better to have separate images because many users tend to have a pre-determined SD card for each Pi so swapping the SD card between Pis is not something they (me included) do.

However if creating multiple images is difficult I'm also OK with a single image with all the kernel packages even if I imagine it would be bigger in size

btb23, 2023-12-15:
I would vote for option 1, keep the images as they are now. Being able to swap from one pi model to another, the ability to develop and move around is worth keeping.

adrianog91, 2023-12-15:
Same for me option 1, keep the images as they are now. In this way we can swap from one model to another model.

btb23, 2023-12-15:
I had a go at migrating headless zeroW over ssh and did this..

`dietpi@DietPi70:~$ sudo bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')
bash: /dev/fd/63: No such file or directory
dietpi@DietPi70:~$ curl: (23) Failed writing body`

I wasn't sure if that was meant to run that way anyway via ssh command line, so then did this

` dietpi@DietPi70:~$ wget 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration'     --2023-12-15 08:42:17--  https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2441 (2.4K) [text/plain]
Saving to: ‘dietpi-rpi-firmware-migration’

dietpi-rpi-firmware-migration 100%[=================================================>]   2.38K  --.-KB/s    in 0.003s

2023-12-15 08:42:18 (753 KB/s) - ‘dietpi-rpi-firmware-migration’ saved [2441/2441]

dietpi@DietPi70:~$ sudo bash dietpi-rpi-firmware-migration
[  OK  ] dietpi-rpi-firmware-migration | Free space check: path=/ | available=17876 MiB | required=200 MiB
[ INFO ] dietpi-rpi-firmware-migration | Updating DietPi to latest version
[ SUB1 ] DietPi-Update > Checking for available DietPi update
[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.25.0
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v8.25.0
[ INFO ] DietPi-Update | Latest version  : v8.25.0
[ INFO ] DietPi-Update | Checking for new available live patches
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bookworm InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
[  OK  ] dietpi-rpi-firmware-migration | mkdir -p rootfs
[  OK  ] dietpi-rpi-firmware-migration | mount /dev/mmcblk0p2 rootfs
[  OK  ] dietpi-rpi-firmware-migration | cp -a /boot/dietpi /boot/dietpi-LICENSE.txt /boot/dietpi-README.md /boot/dietpi.txt rootfs/boot/
[  OK  ] dietpi-rpi-firmware-migration | mkdir -p rootfs/boot/firmware
[  OK  ] dietpi-rpi-firmware-migration | sed -i s|[[:blank:]]/boot[[:blank:]]| /boot/firmware | /etc/fstab
[  OK  ] dietpi-rpi-firmware-migration | systemctl daemon-reload
[  OK  ] dietpi-rpi-firmware-migration | umount /boot
[  OK  ] dietpi-rpi-firmware-migration | mount /boot/firmware
[  OK  ] dietpi-rpi-firmware-migration | umount rootfs
[  OK  ] dietpi-rpi-firmware-migration | ln -s firmware/cmdline.txt /boot/cmdline.txt
[  OK  ] dietpi-rpi-firmware-migration | ln -s firmware/config.txt /boot/config.txt
[  OK  ] dietpi-rpi-firmware-migration | dpkg --add-architecture arm64
[ INFO ] dietpi-rpi-firmware-migration | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bookworm InRelease
Get:3 https://archive.raspberrypi.org/debian bookworm/main arm64 Packages [352 kB]
Fetched 352 kB in 10s (34.2 kB/s)
Reading package lists...
[  OK  ] dietpi-rpi-firmware-migration | APT update
[ INFO ] dietpi-rpi-firmware-migration | APT install linux-image-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v6 linux-image-rpi-v7 linux-image-rpi-v7l raspi-firmware, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  cpio initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base
  linux-image-6.1.0-rpi7-rpi-2712:arm64 linux-image-6.1.0-rpi7-rpi-v6
  linux-image-6.1.0-rpi7-rpi-v7 linux-image-6.1.0-rpi7-rpi-v7l
  linux-image-6.1.0-rpi7-rpi-v8:arm64
Suggested packages:
  libarchive-dev linux-doc-6.1:arm64 debian-kernel-handbook:arm64
  linux-doc-6.1 debian-kernel-handbook bluez-firmware
Recommended packages:
  busybox | busybox-static zstd apparmor:arm64 apparmor
[... 159 lines cut ...]
(Reading database ... 61365 files and directories currently installed.)
Purging configuration files for raspberrypi-kernel (1:1.20230405-1) ...
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 6.1.21+ /boot/kernel.img
/etc/kernel/postrm.d/initramfs-tools:
update-initramfs: Deleting /boot/initrd.img-6.1.21+
run-parts: executing /etc/kernel/postrm.d/z50-raspi-firmware 6.1.21+ /boot/kernel.img
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 6.1.21-v7+ /boot/kernel7.img
/etc/kernel/postrm.d/initramfs-tools:
update-initramfs: Deleting /boot/initrd.img-6.1.21-v7+
run-parts: executing /etc/kernel/postrm.d/z50-raspi-firmware 6.1.21-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 6.1.21-v7l+ /boot/kernel7l.img
/etc/kernel/postrm.d/initramfs-tools:
update-initramfs: Deleting /boot/initrd.img-6.1.21-v7l+
run-parts: executing /etc/kernel/postrm.d/z50-raspi-firmware 6.1.21-v7l+ /boot/kernel7l.img
[... cut ...]
[  OK  ] dietpi-rpi-firmware-migration | rm -Rf /boot/firmware/dietpi /boot/firmware/dietpi-LICENSE.txt /boot/firmware/dietpi-README.md /boot/firmware/dietpi.txt /boot/firmware/COPYING.linux
dietpi@DietPi70:~$
 `
That seemed to work as expected. I rebooted the zeroW and that was OK. Powered off and put the sd card into the pi5 but it doesn't boot. I just hear the pi5 fan spinning, it isn't booting.

I'm probably doing something wrong, but haven't investigated further thinking I should just report back.

Joulinar, 2023-12-15:
Best is to use `root` user for such activities. Anyway. Did you moved to latest DietPi v8.25 Beta before? And the Zero W was running 32bit before? Right?

MichaIng, 2023-12-15:
I think I know the problem. I was told that the `2712` kernel supports 64-bit userland only, while you run 32-bit userland. I guess the RPi 5 picks this kernel with priority, but then fails to load the 32-bit init system. The `v8` kernel supports RPi 5 as well and 32-bit.

Hence, please attach the card to the RPi Zero again, then:
```sh
G_AGP linux-image-rpi-2712
```
Then try again to boot it on RPi 5.

matttttttttttttt, 2023-12-15:
> I created a first alpha version of a firmware migration script. Would be great if you guys could test it on your RPi systems. With this done, it should also support the RPi 5, i.e. you should be able to swap the SD card (or whichever boot media you use) to the RPi 5 and boot it.
> 
> #### But you must apply the current DietPi v8.25 beta first: #6801
> Then apply the script like that:
> 
> ```shell
> G_DEV_BRANCH beta # for a quick switch + update to beta branch :)
> bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')
> ```
> 
> It worked well on RPi Zero. Will migrate my RPi 2 now.

This worked perfectly going from [aspberry Pi 4 Model B to Raspberry 5! Thank you so much :)

MichaIng, 2023-12-15:
Do you use a 64-bit image? If so, can you check whether the 2712 kernel is used?
```sh
uname -a
```

matttttttttttttt, 2023-12-15:
> Do you use a 64-bit image? If so, can you check whether the 2712 kernel is used?
> 
> ```shell
> uname -a
> ```

Yes!

root@DietPi:~# uname -a
Linux DietPi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

matttttttttttttt, 2023-12-15:
Everything works out of the box. I think they are a few things that needs some adjustments, but I am so blown away that this already works flawless. 

Danke Micha :)

bwims, 2023-12-15:
Just a thought - will the power button be supported?
From [here](https://community.element14.com/products/raspberry-pi/f/raspberry-pi-5-faq/53770/does-the-raspberry-pi-5-have-a-power-button-and-how-does-it-function)

> 
> The Raspberry Pi 5 has a power button.
> 
> Once Pi OS has booted, in Linux it appears to the operating system as a normal key input providing KEY_POWER scan code.
> 
> /etc/systemctl/logind.conf can be used to define what happens when the button is pressed. On the Pi OS lite image it'll shut down by default. On the full desktop image the shutdown dialogue will be displayed, and if you press it again, it will shut down.
> 
> 

B.

jboots07, 2023-12-15:
> This worked perfectly going from [aspberry Pi 4 Model B to Raspberry 5! Thank you so much :)

Hmm I applied beta to an existing system, runs fine on 4 still, put into 5 and fan just stays on full speed and no boot.

LittleFreak, 2023-12-16:
> - But the question is how we do it with new Images
>   1. Install all kernel packages supported by the OS/userland architecture, so that the system card/drive can be swapped between RPi models like [...]

I assume there has to be a kernel present to boot, instead of waiting for a used input to download the selected kernel on the fly? 

If so, why don't choose option
3. All kernel packages supported will be installed with the first boot, but will be replaced with its specific kernel packages (unnecessary one deleted) after it's complete setup, if the user says so via input (or 'dietpi-config' for headless setups). Non-headless would require an input from the user. 

Why? It's the best of both worlds. 
- It would allow to install dietpi on a running raspbian os 
- some users switch cards between different raspi Gen (as backup f. I.) 
- others use docker for those backup purposes and only need 1 kernel
- unity would eliminate the need of multiple images and ease the download for consumer (but might also increase server traffic)

MichaIng, 2023-12-16:
> Hmm I applied beta to an existing system, runs fine on 4 still, put into 5 and fan just stays on full speed and no boot.

Did you run the second command/migration script after applying the beta? The beta (DietPi v8.25) is required for the migration script, but it does not imply the migration.

> 3. All kernel packages supported will be installed with the first boot, but will be replaced with its specific kernel packages (unnecessary one deleted) after it's complete setup, if the user says [...]

This is basically my option 1: The images would be shipped with all kernel packages pre-installed. The `dietpi-config` option to (de)select kernel packages (only those which are not required for the current model) will be added in both cases. The question is really only about whether we should provide less but larger images, compatible with more RPi models each, or more but smaller more targeted images, compatible with less RPi models each. "Diet" would be of course the 2nd option, but it would be also a change compared to our current RPi image lineup.

> It would allow to install dietpi on an existing/running raspbian os

This works regardless which of those options we choose. The `dietpi-installer` will allow both ways anyway, an "all models" option as well as separate options for individual RPi models, at least if we chose option 2. Because I want to offer the new images separately for testing first, and the old ones (with old kernel) as stable releases for now. And when we switch to the new images with new kernel, there is not reason to remove the "all models" option from the installer. And the userland architecture is anyway given by the base OS you run the installer on.

For us, our traffic, build systems (with is GitHub Actions runners anyway) etc it does not matter. We have plenty of free traffic and I do not mind if images builds take some hours longer over night. The question really is whether OOTB flexibility to flash the same image on multiple different RPi models and switching SD card between them without needing to select an additional kernel first, is worth the added disk usage, increased downloads/processing/disk writes on kernel upgrades this implies.

We could even host images for both variants 😄. But I would not want to overload the already overloaded download page with this, but keep then behind some "all downloads" button or so.

On the download page, option 2 would practically add two more RPi tiles: RPi 4 became a dedicated one, and RPi 5 an additional once. With option 1, RPi 5 would become part of the Raspberry Pi 2/3/4 tile: https://dietpi.com/#downloadinfo

LittleFreak, 2023-12-16:
Personally, I'm planning to use docker images for everything, so I don't need the flexibility to use the same SD card on different raspi gens.
So °2 would be sufficient for me.
I proposed °3 as compromise, because I'm not the center of the world (thanks god, we would be doomed XD)

MichaIng, 2023-12-16:
I think I did not understand the difference between my option 1 and your option 3 yet. Option 1 means all kernel packages are pre-installed, but users can de-select unneeded kernels via `dietpi-config`. Could you explain what would be different in your option 3?

jboots07, 2023-12-16:
> > Hmm I applied beta to an existing system, runs fine on 4 still, put into 5 and fan just stays on full speed and no boot.
> 
> Did you run the second command/migration script after applying the beta? The beta (DietPi v8.25) is required for the migration script, but it does not imply the migration.
> 

I'll admit I am a noob, but I did my best.

The results were:

```
root@DietPi:~# bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')
[FAILED] dietpi-rpi-firmware-migration | You can run this script on Debian Bookworm or Trixie systems only!
```

LittleFreak, 2023-12-16:
> I think I did not understand the difference between my option 1 and your option 3 yet. Option 1 means all kernel packages are pre-installed, but users can de-select unneeded kernels via `dietpi-config`. Could you explain what would be different in your option 3?

~~Basically there is no difference, unless the required (specific) kernel could be downloaded on the fly.~~
Now when I think it through, It would basically assume the opposite of °1=>
Installing all other kernel packages _only_ on the user's behalf

jboots07, 2023-12-16:
> > > Hmm I applied beta to an existing system, runs fine on 4 still, put into 5 and fan just stays on full speed and no boot.
> > 
> > 
> > Did you run the second command/migration script after applying the beta? The beta (DietPi v8.25) is required for the migration script, but it does not imply the migration.
> 
> EDIT: I didn't realize the
> 
> ```
> G_DEV_BRANCH beta
> ```
> 
> was needed before
> 
> I'll admit I am a noob, but I did my best.
> 
> The results were:
> 
> ```
> root@DietPi:~# bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')
> [FAILED] dietpi-rpi-firmware-migration | You can run this script on Debian Bookworm or Trixie systems only!
> ```

Error persists

MichaIng, 2023-12-16:
@jboots07 
You are running a Bullseye system, I guess?
```sh
echo $G_DISTRO_NAME
```
You would need to upgrade to Bookworm first: https://dietpi.com/blog/?p=3128#howto
The new firmware packages, including the ones required for RPi 5, are available on Bookworm only.

@LittleFreak 
Jep, all kernel packages can be installed on the fly. It is really not much more than an `apt install` call or some clicks through `dietpi-config` and waiting some minute(s) for download and install.

btb23, 2023-12-16:
> Best is to use `root` user for such activities. Anyway. Did you moved to latest DietPi v8.25 Beta before? And the Zero W was running 32bit before? Right?

> I think I know the problem. I was told that the `2712` kernel supports 64-bit userland only, while you run 32-bit userland. I guess the RPi 5 picks this kernel with priority, but then fails to load the 32-bit init system. The `v8` kernel supports RPi 5 as well and 32-bit.
> 
> Hence, please attach the card to the RPi Zero again, then:
> 
> ```shell
> G_AGP linux-image-rpi-2712
> ```
> 
> Then try again to boot it on RPi 5.

Thank you both. I abandoned the first zero W migration as I'd messed up, went again as root this time with a rpi2 i also had running and this time the migration went ahead.

Many thanks.

jboots07, 2023-12-16:
All is well and working when I followed proper procedure, thank you!

alfredoanton82, 2023-12-16:
Installed from an old Rpi3B, everything well and running! Thanks a lot @MichaIng, much appreciated all your work guys.
Regarding the options, I would prefer 2nd, as I love dietpi for being light, so I would prefer install only the required kernel... but if it is less work for you or simplifies for rookies/starters... go ahead with 1st... 
Best regards!

thejimp, 2023-12-16:
> I created a first alpha version of a firmware migration script. Would be great if you guys could test it on your RPi systems. With this done, it should also support the RPi 5, i.e. you should be [...]

Can confirm this works fine going from RPi4 to RPi5 for me ,thanks!   

I'd say option 1 was probably the most beginner-friendly option (and allows unexpected swapping between different models (e.g. if your main RPi dies, straightforward to swap to another one, no need to downlaod the relevant kernels).  If someone wanted a super-minimal  install, they can just uninstall them.  I think that's a fair trade-off as a user.

Cheers for all your work maintaining DietPi, I've had my RPi for a couple of weeks, but I have been saving it until I can run DietPi on it!

tonystower, 2023-12-17:
> I created a first alpha version of a firmware migration script. 

Thank you so much for this @MichaIng 
Once I had figure out that I had to do this using a root shell rather than just sudo, it worked perfectly. 

I was moving from a Pi 400 to a pi5 and all seems fine so far. No hiccups at all. 
thanks again

sorriso93, 2023-12-18:
Hello I assume 8.25.1 is ok to try the migration script for RPI5, correct?

Joulinar, 2023-12-18:
Yes

LittleFreak, 2023-12-18:
I'd like to test the install script for a setup on a running/existing raspbian os, to give it a diet 🫠😉
Pls hit me up, once it's ready for testing in that flavour =)

[341 more replies]