MichaIng/DietPi #6677: WiFi not working on Raspberry Pi 4 with WPA3 5 GHz/6GHz only-Network

thomaswitt, 2023-10-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.22.3
- Distro version | bookworm 0
- Kernel version | Linux MyPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Unifi PPPOE Converter
- SD card used | SanDisk Ultra

#### Steps to reproduce
I run the latest dietpi on a RPi4. Network Connectivity via Ethernet works fine.

I want to activate the WiFi interface as well, so I did via dietpi-config. Activated, SSID and Password is correctly set (and double checked via wpa_passphrase command). The network I am connecting to is a WPA3 5/6 GHz Network only. When restarting the interface, I constantly get timeouts on the DHCP Discovery:

```
Listening on LPF/wlan0/e4:5f:…
Sending on   LPF/wlan0/e4:5f:…
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 10
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 19
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 18
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

I tried a different network which also offer 2.4 GHZ with no probems.

#### Expected behaviour
WiFi should connect to WPA3 Networks.

#### Actual behaviour
WiFi doesn't connect to WPA3 Networks.

#### Extra details
- WiFi itself works, when I do a `iwlist wlan0 scan | grep ESSID`, I see the whole Wifi Network list including the ESSID I want to connect to.
- The AP is a Unifi Hotspot with hundreds of devices connected without any problems.
- Apparently there are [ways to patch/fix this](https://github.com/raspberrypi/linux/issues/4718#issuecomment-1155509563)

MichaIng, 2023-10-12:
Looks like WPA3 does not use "WPA-PSK", but "SAE". And it looks like WPA3 is even mandatory for 6 GHz, probably Wi-Fi 6 in general. Although Wi-Fi 6 with 2.4 GHz seems to work with WPA2 as well. But probably with 5 GHz and 6 GHz is not not work without WPA3: #6636

Can you try to add/replace the related settings from the post you linked to `/etc/wpa_supplicant/wpa_supplicant.conf` and then retry connecting to the network, e.g. via `ifup wlan0`?

MichaIng, 2023-10-12:
Ah right, they are talking about possibly needed kernel patches.

So generally to test, after setting up Wi-Fi via `dietpi-config` with Wi-Fi 6 and 5 or 6 GHz, in `/etc/wpa_supplicant/wpa_supplicant.conf`:
1. Replace `key_mgmt=WPA-PSK` with `key_mgmt=SAE`
2. Replace `psk=<64-digit-hex-password>` with `sae_password=<password>`
3. Add `ieee80211w=2`
4. Bring up again the network: `ifdown wlan0; ifup wlan0`
5. If it still does not work, add `proto=RSN` as well.

thomaswitt, 2023-10-13:
FYI, I entered the configuration without any luck, still DHCP discover timeout. Adding `proto=RSN` doesnt work at all:
```
Pi:/etc/wpa_supplicant # ifup wlan0
wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
ifup: failed to bring up wlan0
```

MichaIng, 2023-10-18:
Milestone applied just to keep easier track of the RPi kernel upstream issue.

kwinz, 2024-01-24:
So if you're like me and you found this thread to be one of the first results for a Google search for raspberry WPA3, and you think you can just use the `wpa_supplicant.config` above: no, this doesn't work. Currenly (January 2024) you need to manually change the Rasperry cypress WiFi chip firmware, and even then it doesn't work with `wpa_supplicant`. See linked upstream issue.

MichaIng, 2024-01-24:
Jep, as thomaswitt stated. Actually just a few days ago, a new firmware package has been released which others stated adds WPA3 support for RPi 3B+, RPi 4 and 5:
- https://github.com/raspberrypi/linux/issues/4718#issuecomment-1900683905
- https://github.com/RPi-Distro/firmware-nonfree/commit/3db4164

It is on APT servers as well since two days. What I am not sure about is whether this requires an up-to-date driver, hence the new set of (Bookworm) firmware/kernel packages. For DietPi users this would mean to do the migration: https://github.com/RPi-Distro/firmware-nonfree/commit/3db4164
For RPi OS users it means to use a new RPi OS Bookworm image.

TomBayne, 2024-05-06:
Any updates on this? Unable to connect to a WiFi 6 access point with WPA2/WPA3 mixed mode :(

MichaIng, 2024-05-06:
You could test whether it works with the new kernel/firmware package stack: #6676
Note that this migration cannot be (easily) reverted.

Marzal, 2024-05-12:
Same problem here:
- DietPi : v9.3.0
- Device model : RPi 3 Model A+ (aarch64)
- Kernel:  6.1.21-v8+ #1642 SMP PREEMPT
- Debian : 12 (bookworm)

Tested config:
`network={
        ssid="SSID"
        scan_ssid=1
        key_mgmt=SAE
        sae_password="REAL PASSWORD"
        ieee80211w=2
        proto=RSN
}`

Also tested without proto=RSN

MichaIng, 2024-05-12:
Please read through the thread. It is a know limitation with the RPi kernel. See my above post with a link to migrate to a new RPi kernel/firmware stack, which should fix it, according to this PR: https://github.com/raspberrypi/linux/pull/5945

Marzal, 2024-05-13:
> Please read through the thread. It is a know limitation with the RPi kernel. See my above post with a link to migrate to a new RPi kernel/firmware stack, which should fix it, according to this PR: [raspberrypi/linux#5945](https://github.com/raspberrypi/linux/pull/5945)
> 
> If you have a spare SD card, you can also test it first with a fresh image. The `RPi234` named ones (`RPi1`/`RPi2`/`RPi5` for other RPi models respectively) here have the new kernel/firmware packages pre-installed: https://dietpi.com/downloads/images/testing/

Thanks, I did read the thread, but reading https://github.com/MichaIng/DietPi/issues/6677#issuecomment-1908962210 I mistakenly thought that being on the lastest Dietpi version of Bookworm would imply using the lastest stack (probably because English is not my native language, sorry).

I will try [[DietPi_RPi234-ARMv8-Bookworm.img.xz](https://dietpi.com/downloads/images/testing/DietPi_RPi234-ARMv8-Bookworm.img.xz)](https://dietpi.com/downloads/images/testing/DietPi_RPi234-ARMv8-Bookworm.img.xz) on a new SD and report back :+1:

MichaIng, 2024-05-13:
The 2nd part of the post is relevant, where I state that a new driver (hence a new kernel) may be required. Obviously, the new firmware alone (meaning the `firmware-brcm80211` WiFi firmware package, not `raspi-firmware`, which is a meta package for RPi kernel and bootlaoder + configs) did not fix it.

MichaIng, 2024-05-14:
According to this comment, it still does not work: https://github.com/raspberrypi/linux/issues/4718#issuecomment-2109098969
If I re-read the PR, it seems that, while that one added support with a certain extension feature to the kernel driver, it still requires a different firmware blob to work: #5945

Marzal, 2024-05-14:
I have updated my https://github.com/MichaIng/DietPi/issues/6677#issuecomment-2106355642 with all the info I have found relevant.

After a few hours reading github issues I've found enough people with working solutions, I thinks this comment summarize the situation:
* https://github.com/raspberrypi/linux/issues/4718#issuecomment-1925580044
> So for now, the options are:
>     Use IWD with v.234 of the firmware (what is in the latest repo: https://github.com/RPi-Distro/firmware-nonfree/commit/ad23f33a29fb7f8bc344d80d0eb40abe1953d145)
>     Use wpa-supplicant with newer firmware with the SAE_OFFLOAD feature disabled in the brcm config file (to tell the userspace app to not attempt offloading onto the chip)

* https://github.com/raspberrypi/linux/issues/4718#issuecomment-1918621939
> The upcoming wpa_supplicant version 2.11 will also include support for "old firmware" (SAE_OFFLOAD extended feature).

* https://github.com/RPi-Distro/firmware-nonfree/issues/41#issuecomment-1900453324
wpa_supplicant 2.10 :-1:  / iwd :+1: 

* [This blog states the same](https://rachelbythebay.com/w/2024/01/24/wpa3/) iwd :+1:  vs  wpa_supplicant :-1: 
** 7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15
** But with [stability issues](https://rachelbythebay.com/w/2024/01/24/fail/)

I will try iwd to see if https://github.com/raspberrypi/linux/issues/4718#issuecomment-2109098969 is having another issue. If this doesn't work I will try to use the new firmware with wpa_supplicant without SAE_OFFLOAD. But all this is bit out of my expertise.

TomBayne, 2024-05-15:
> I have updated my [#6677 (comment)](https://github.com/MichaIng/DietPi/issues/6677#issuecomment-2106355642) with all the info I have found relevant.
> 
> After a few hours reading github issues I've found enough people with working solutions, I thinks this comment summarize the situation:
> 
>     * [Wi-Fi Protected Access 3 (WPA3) support raspberrypi/linux#4718 (comment)](https://github.com/raspberrypi/linux/issues/4718#issuecomment-1925580044)
> 
> 
> > So for now, the options are:
> > Use IWD with v.234 of the firmware (what is in the latest repo: [RPi-Distro/firmware-nonfree@ad23f33](https://github.com/RPi-Distro/firmware-nonfree/commit/ad23f33a29fb7f8bc344d80d0eb40abe1953d145))
> > Use wpa-supplicant with newer firmware with the SAE_OFFLOAD feature disabled in the brcm config file (to tell the userspace app to not attempt offloading onto the chip)
> 
>     * [Wi-Fi Protected Access 3 (WPA3) support raspberrypi/linux#4718 (comment)](https://github.com/raspberrypi/linux/issues/4718#issuecomment-1918593641)
> 
> 
> > Use iwd with old firmware
> > Use patched wpa-supplicant with new firmware
> 
>     * [Wi-Fi Protected Access 3 (WPA3) support raspberrypi/linux#4718 (comment)](https://github.com/raspberrypi/linux/issues/4718#issuecomment-1918621939)
> 
> 
> > The upcoming wpa_supplicant version 2.11 will also include support for "old firmware" (SAE_OFFLOAD extended feature).
> 
>     * [BRCM/Cypress 43455 firmware doesn't support WPA3-SAE RPi-Distro/firmware-nonfree#41 (comment)](https://github.com/RPi-Distro/firmware-nonfree/issues/41#issuecomment-1900453324)
>       wpa_supplicant 2.10 👎  / iwd 👍
> 
>     * [WiFi not working on Raspberry Pi 4 with WPA3 5 GHz/6GHz only-Network #6677 (comment)](https://github.com/MichaIng/DietPi/issues/6677#issuecomment-1908947912)
>       wpa_supplicant 👎  / iwd 👍
> 
>     * [Linux Kernel discussion](https://lore.kernel.org/all/c392f901-789a-42e2-8cf7-5e246365a1ca@broadcom.com/t/)  : wpa_supplicant 2.10 👎  / iwd 👍
> 
>     * [This blog states the same](https://rachelbythebay.com/w/2024/01/24/wpa3/) wpa_supplicant 👎  / iwd 👍
>       ** 7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15
>       ** But with [stability issues](https://rachelbythebay.com/w/2024/01/24/fail/)
> 
> 
> I will try iwd to see if [raspberrypi/linux#4718 (comment)](https://github.com/raspberrypi/linux/issues/4718#issuecomment-2109098969) is having another issue. If this doesn't work I will try to use the new firmware with wpa_supplicant without SAE_OFFLOAD. But all this is bit out of my expertise.

Sorry, I'm possibly being dumb here, but did you actually find a solution? Or is the conclusion just 'it doesn't work yet'?

TomBayne, 2024-05-15:
https://github.com/raspberrypi/linux/pull/5945

This one fixed it for me. Super quick fix.

MichaIng, 2024-05-15:
> This one fixed it for me. Super quick fix.

Which one did you mean exactly? Installing the new kernel only, without updating any firmware?

Interesting, I didn't know about `iwd`. Seems a nice alternative to `wpasupplicant`. But it does not have any native `ifupdown` integration. However, running the daemon alone seems to work, so we could add an own `if-pre-up.d` script, which starts `iwd` in case a WiFi interface is brought up.

TomBayne, 2024-05-17:
> Which one did you mean exactly? Installing the new kernel only, without updating any firmware?
>
> Interesting, I didn't know about `iwd`. Seems a nice alternative to `wpasupplicant`. But it does not [...]

I just installed that WiFi driver in the first message. Although, it didn't actually fix it for me, but it got me closer. My Pi5 now connects to the WPA3 network, and shows up as connected on the router, but will get disconnected by the router for inactivity after a couple of minutes.

MichaIng, 2024-05-19:
So you updated the firmware but not the kernel (driver). This is then probably the missing part: #6676

In case this works, we could create a small package which replaces this particular firmware binary on RPi systems with the new kernel stack.

dwery, 2025-02-25:
FWIW , WPA3 works on my RPi5 with 7.45.265, 6.12.15-v8-16k+ and wpa_supplicant 2.10

MichaIng, 2025-02-25:
Are you aware of a particular commit in this kernel compared to the current stable v6.6.74 packages?