MichaIng/DietPi #6681: DietPi-Software | Pi-hole: Add initial v6 support

MichaIng, 2023-10-12:
https://discourse.pi-hole.net/t/pi-hole-v6-beta-testing/65413

Plan:
- Add a check whether Pi-hole v6 has been released or not, and define the deps + do the install steps accordingly.
- Add an env var or CLI flag for a Pi-hole v6 install. This would currently be a regular v5 install followed by a v6 upgrade/migration.

GvY85, 2023-10-22:
Will V6 support automatically any installed webservers (if only used by Pihole)?

Joulinar, 2023-10-22:
PiHole v6 has one integrated web server and don't require any in addition. I guess we will keep Pihole v6 web interface running on a different port anyway to avoid conflict with other web server apps like Nextcloud.

MichaIng, 2024-12-05:
In case the webserver is needed for another application, when upgrading Pi-hole to v6, we can make Pi-hole listen on port 8089, and then implement a reverse proxy, like this for Apache2:
```
ProxyPreserveHost On
ProxyPass /admin http://127.0.0.1:8089/admin
ProxyPassReverse /admin http://127.0.0.1:8089/admin
ProxyPass /api http://127.0.0.1:8089/api
ProxyPassReverse /api http://127.0.0.1:8089/api
```

closed 2025-02-23 by MichaIng