MichaIng/DietPi #6682: public key can't be found anywhere

j-lakeman, 2023-10-14:
#### Steps to reproduce

1. download an image including checksum and signature
2. `gpg --verify dummy.img.xz.asc dummy.img.xz.sha256`
3. output:

```
gpg: Signature made So 24 Sep 2023 04:57:51 CEST
gpg:                using RSA key C2C4D1DEF7C96C6EDF3937B2536B2A4A2E72D870
gpg: Can't check signature: No public key
```

4. try to obtain key:
`gpg --keyserver keyring.debian.org --receive-keys C2C4D1DEF7C96C6EDF3937B2536B2A4A2E72D870`
or
`gpg --keyserver hkp://keyserver.ubuntu.com --recv-keys C2C4D1DEF7C96C6EDF3937B2536B2A4A2E72D870`
5. output:

```
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
``` 

#### Expected behaviour

- public key should be present on some key server, e.g. `keyring.debian.org` or `hkp://keyserver.ubuntu.com`

#### Actual behaviour

- key cant be found

Joulinar, 2023-10-14:
Did you check our website > downloads? There all signatures should be available.

MichaIng, 2023-10-14:
The public key the one I use here on GitHub: https://github.com/MichaIng.gpg
You are right, I should upload it to one of the key servers. I was just not reading into it yet.

werdahias, 2023-10-15:
@MichaIng keyserver.debian.org is a debian-internal keyserver for developers / maintainers. I'd suggest publishing to either one of those: https://pgp.mit.edu/ , https://keyserver.ubuntu.com/ or https://keys.openpgp.org/

MichaIng, 2023-10-15:
Key uploaded to [keys.openpgp.org](https://keys.openpgp.org/search?q=974105F494304547F1A9E5E00442B9ADE65643FE), which seems to be the most modern and open public key server. I see `keyserver.ubuntu.com` so regularly used by other projects, any argument against uploading it there as well? If I understood correctly, they do not sync keys between each other, do they?

werdahias, 2023-10-15:
iirc they do not sync their servers. uploading the keys to more than one server is a good idea imho as that simplifies key discovery

MichaIng, 2023-10-15:
On [keyserver.ubuntu.com](https://keyserver.ubuntu.com/pks/lookup?search=C2C4D1DEF7C96C6EDF3937B2536B2A4A2E72D870&fingerprint=on&op=index) now as well.

closed 2023-10-15 by j-lakeman