MichaIng/DietPi #6683: Home Assistant 2023.10 failed to start on armv6 (Illegal instruction)

steals, 2023-10-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.22.3`
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi B (armv6l)`

#### Additional Information (if applicable)
- Software title | Home Assistant 2023.10.3

#### Steps to reproduce

1. Upgrade Home Assistant to the latest version (2023.10.3)
2. Restart Home Assistant

After upgrading Home Assistant to the latest version it fails to start due to the "Illegal Instruction" error. I believe this is because some of the libs were upgraded, and looks like one of them is not compiled correctly for the armv6 architecture. Could you please help to identify what lib is throwing that error?

Here is the fault.log

```
homeassistant@DietPi:~$ cat .homeassistant/home-assistant.log.fault
Fatal Python error: Illegal instruction

Thread 0xae97b420 (most recent call first):
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.11/site-packages/pyudev/_util.py", line 164 in eintr_retry_call
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.11/site-packages/pyudev/_os/poll.py", line 94 in poll
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.11/site-packages/pyudev/_util.py", line 164 in eintr_retry_call
  File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.11/site-packages/pyudev/monitor.py", line 537 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xaf266420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 320 in wait
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/_base.py", line 451 in result
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/components/recorder/core.py", line 680 in _wait_startup_or_shutdown
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/components/recorder/core.py", line 739 in _run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/components/recorder/core.py", line 691 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xb10fb420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/thread.py", line 81 in _worker
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 975 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xb18fc420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/thread.py", line 81 in _worker
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 975 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xb20fd420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/thread.py", line 81 in _worker
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 975 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xb30ff420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/thread.py", line 81 in _worker
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 975 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap

Thread 0xb3aff420 (most recent call first):
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/concurrent/futures/thread.py", line 81 in _worker
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 975 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 1038 in _bootstrap_inner
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/threading.py", line 995 in _bootstrap
[... 38 lines cut ...]
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1232 in _handle_fromlist
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/components/cloud/__init__.py", line 35 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1204 in _gcd_import
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/importlib/__init__.py", line 126 in import_module
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/loader.py", line 818 in get_component
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/setup.py", line 215 in _async_setup_component
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/asyncio/events.py", line 80 in _run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/asyncio/base_events.py", line 1922 in _run_once
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/asyncio/base_events.py", line 607 in run_forever
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/asyncio/base_events.py", line 640 in run_until_complete
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/runner.py", line 186 in run
  File "/home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/site-packages/homeassistant/__main__.py", line 204 in main
  File "/home/homeassistant/.pyenv/versions/3.11.4/bin/hass", line 8 in <module>

Extension modules: _cffi_backend, multidict._multidict, yarl._quoting_c, ciso8601, aiohttp._helpers, aiohttp._http_writer, aiohttp._http_parser, aiohttp._websocket, frozenlist._frozenlist, ulid_transform._ulid_impl, markupsafe._speedups, PIL._imaging, sqlalchemy.cyextension.collections, sqlalchemy.cyextension.immutabledict, sqlalchemy.cyextension.processors, sqlalchemy.cyextension.resultproxy, sqlalchemy.cyextension.util, fnv_hash_fast._fnv_impl, dbus_fast.signature, dbus_fast._private.marshaller, dbus_fast.message, dbus_fast.service, dbus_fast._private.address, dbus_fast.unpack, dbus_fast.message_bus, dbus_fast._private.unmarshaller, dbus_fast.aio.message_reader, home_assistant_bluetooth.models, zeroconf._utils.time, zeroconf._dns, zeroconf._protocol.outgoing, zeroconf._protocol.incoming, zeroconf._cache, zeroconf._updates, zeroconf._handlers.record_manager, zeroconf._listener, zeroconf._handlers.answers, zeroconf._handlers.multicast_outgoing_queue, zeroconf._history, zeroconf._services.info, zeroconf._services.registry, zeroconf._handlers.query_handler, zeroconf._services.browser, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator (total: 56)
```

Joulinar, 2023-10-16:
can you share the system log

```
journalctl -u home-assistant
```

steals, 2023-10-17:
```
dietpi@DietPi:~$ sudo journalctl -u home-assistant
Oct 17 07:51:24 DietPi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
dietpi@DietPi:~$ sudo journalctl -u home-assistant
Oct 17 07:51:24 DietPi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
dietpi@DietPi:~$ sudo journalctl -u home-assistant
Oct 17 07:51:24 DietPi systemd[1]: Started home-assistant.service - Home Assistant (DietPi).
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.753 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.768 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.778 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.788 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration ble_monitor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.798 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration gismeteo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:51:54 DietPi Home Assistant[12734]: 2023-10-17 07:51:54.808 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration yandex_weather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
Oct 17 07:52:25 DietPi Home Assistant[12734]: 2023-10-17 07:52:25.507 WARNING (MainThread) [homeassistant.setup] Setup of person is taking over 10 seconds.
Oct 17 07:52:25 DietPi Home Assistant[12734]: 2023-10-17 07:52:25.844 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //mnt/dietpi_userdata/homeassistant/home-assistant_v2.db was shutdown cleanly
Oct 17 07:52:26 DietPi Home Assistant[12734]: 2023-10-17 07:52:26.214 WARNING (Recorder) [homeassistant.components.recorder.util] Ended unfinished session (id=362 from 2023-10-16 12:39:14.857642)
Oct 17 07:52:46 DietPi systemd[1]: home-assistant.service: Main process exited, code=killed, status=4/ILL
Oct 17 07:52:46 DietPi systemd[1]: home-assistant.service: Failed with result 'signal'.
Oct 17 07:52:46 DietPi systemd[1]: home-assistant.service: Consumed 1min 18.488s CPU time.
```

MichaIng, 2023-10-18:
> The system could not validate that the sqlite3 database at //mnt/dietpi_userdata/homeassistant/home-assistant_v2.db was shutdown cleanly

I think this is a symptom, not the issue. The SIGILL should be the problem. I hope that not some one of the Python modules dropped support for ARMv6. How did you update HA, just via `pip` or via `dietpi-software reinstall`?

steals, 2023-10-18:
I've updated with `. /home/homeassistant/homeassistant-update.sh`, so basically via `pip`
I guess the root cause of the issue in one of the modules, as I remember that previously some modules were built locally during the installation of homeassitant but with the latest update all modules were installed with a prebuilt binaries.

MichaIng, 2023-10-18:
Do you have a system Python 3 instance and does it use piwheels?
```sh
cat /etc/pip.conf
```

MichaIng, 2023-10-18:
For the record, the modules which should be compiled from source:
```
[ INFO ] DietPi-Software | Installing Python with Home Assistant module into pyenv, please wait...
Downloading Python-3.11.5.tar.gz...
-> https://www.python.org/ftp/python/3.11.5/Python-3.11.5.tgz
Installing Python-3.11.5...
Installed Python-3.11.5 to /home/homeassistant/.pyenv/versions/3.11.5
Requirement already satisfied: pip in ./.pyenv/versions/3.11.5/lib/python3.11/site-packages (23.2.1)
Collecting pip
  Obtaining dependency information for pip from https://files.pythonhosted.org/packages/e0/63/b428aaca15fcd98c39b07ca7149e24bc14205ad0f1c80ba2b01835aedde1/pip-23.3-py3-none-any.whl.metadata
  Downloading pip-23.3-py3-none-any.whl.metadata (3.5 kB)
Requirement already satisfied: setuptools in ./.pyenv/versions/3.11.5/lib/python3.11/site-packages (65.5.0)
Collecting setuptools
  Obtaining dependency information for setuptools from https://files.pythonhosted.org/packages/bb/26/7945080113158354380a12ce26873dd6c1ebd88d47f5bc24e2c5bb38c16a/setuptools-68.2.2-py3-none-any.whl.metadata
  Downloading setuptools-68.2.2-py3-none-any.whl.metadata (6.3 kB)
Collecting wheel
  Obtaining dependency information for wheel from https://files.pythonhosted.org/packages/b8/8b/31273bf66016be6ad22bb7345c37ff350276cfd46e389a0c2ac5da9d9073/wheel-0.41.2-py3-none-any.whl.metadata
  Downloading wheel-0.41.2-py3-none-any.whl.metadata (2.2 kB)
Downloading pip-23.3-py3-none-any.whl (2.1 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 2.5 MB/s eta 0:00:00
Downloading setuptools-68.2.2-py3-none-any.whl (807 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 807.9/807.9 kB 3.5 MB/s eta 0:00:00
Downloading wheel-0.41.2-py3-none-any.whl (64 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.8/64.8 kB 4.6 MB/s eta 0:00:00
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.5.0
    Uninstalling setuptools-65.5.0:
      Successfully uninstalled setuptools-65.5.0
  Attempting uninstall: pip
    Found existing installation: pip 23.2.1
    Uninstalling pip-23.2.1:
      Successfully uninstalled pip-23.2.1
Successfully installed pip-23.3 setuptools-68.2.2 wheel-0.41.2
Collecting homeassistant
  Downloading homeassistant-2023.10.3-py3-none-any.whl.metadata (3.8 kB)
Collecting aiohttp==3.8.5 (from homeassistant)
  Downloading aiohttp-3.8.5.tar.gz (7.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.4/7.4 MB 2.1 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Installing backend dependencies: started
  Installing backend dependencies: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting astral==2.2 (from homeassistant)
  Downloading astral-2.2-py2.py3-none-any.whl (30 kB)
Collecting attrs==23.1.0 (from homeassistant)
  Downloading attrs-23.1.0-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 kB 4.6 MB/s eta 0:00:00
Collecting atomicwrites-homeassistant==1.4.1 (from homeassistant)
  Downloading atomicwrites_homeassistant-1.4.1-py2.py3-none-any.whl (7.1 kB)
Collecting awesomeversion==23.8.0 (from homeassistant)
  Downloading awesomeversion-23.8.0-py3-none-any.whl.metadata (8.9 kB)
Collecting bcrypt==4.0.1 (from homeassistant)
  Downloading bcrypt-4.0.1.tar.gz (25 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting certifi>=2021.5.30 (from homeassistant)
  Downloading certifi-2023.7.22-py3-none-any.whl.metadata (2.2 kB)
Collecting ciso8601==2.3.0 (from homeassistant)
  Downloading ciso8601-2.3.0.tar.gz (26 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
[... 178 lines cut ...]
Downloading anyio-4.0.0-py3-none-any.whl (83 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 83.1/83.1 kB 4.4 MB/s eta 0:00:00
Building wheels for collected packages: aiohttp, bcrypt, ciso8601, cryptography, home-assistant-bluetooth, lru-dict, orjson, python-slugify, PyYAML, ulid-transform, yarl, cffi, frozenlist, MarkupSafe, multidict
  Building wheel for aiohttp (pyproject.toml): started
  Building wheel for aiohttp (pyproject.toml): still running...
  Building wheel for aiohttp (pyproject.toml): still running...
  Building wheel for aiohttp (pyproject.toml): still running...
  Building wheel for aiohttp (pyproject.toml): still running...
  Building wheel for aiohttp (pyproject.toml): finished with status 'done'
  Created wheel for aiohttp: filename=aiohttp-3.8.5-cp311-cp311-linux_armv6l.whl size=1036444 sha256=72af57192a991c75d07acccfef39b952ebfc847b7061b6aaebe7417c317cb3c3
  Stored in directory: /tmp/pip-ephem-wheel-cache-jmek75mc/wheels/86/de/56/a2c0346214531448dc562ea44a458e01e3616f1d221f6b5b9e
  Building wheel for bcrypt (pyproject.toml): started
  Building wheel for bcrypt (pyproject.toml): still running...
  Building wheel for bcrypt (pyproject.toml): still running...
  Building wheel for bcrypt (pyproject.toml): still running...
[... cut ...]
  Building wheel for bcrypt (pyproject.toml): still running...
```
This will take a few more hours on my RPi Zero W. Will report back if this works and try to replace the compiled modules one by one with pre-compiled wheels from piwheels, if available. But will start with `pyudev`, which is not installed here, but probably on first service start. If not, then it might be one of the integrations you use, which requires this module. Based on the error message, this one is what seems to cause the issue.

steals, 2023-10-19:
Interesting... I've rolled back to 2023.9.3 and it starts just fine without any Illegal Instrucructions. During the downgrade I've notied that only a few libs have been downgraded together with HA. They are `awesomeversion-22.9.0 cryptography-41.0.3 orjson-3.9.2`. I've tried to install homeassistant==2023.10.1 with the versions of these libs as for 2023.9.3 and again got Illegal Instruction error. So it looks like something is added in the homeassistant core itself that causing that error

MichaIng, 2023-10-19:
It works fine here, with the compiled modules. It might be however possible that the new HA version uses a new module, like `pyudev`, while the older does not. So the problem would still be that said module is pulled form piwheels and is compiled with faulty flags there. Testing is so down slow on my RPi Zero W. There are other missing new build dependencies for HA now, so figuring out those first, then check beck regarding the piwheels `pyudev` wheel.

steals, 2023-10-19:
Ok,  the root cause of the issue is the `webrtc-noise-gain` module that was introduced recently in HA 2023.10.1. After I uninstalled it and built locally HA back to normal.

MichaIng, 2023-10-19:
Ah great. I just wanted to write that installing `pyudev` and its dependency `six` from piwheels does not cause issues. Is it that the `webrtc-noise-gain` module from piwheels which causes the issue, or a source build as well? You could enforce it via:
```sh
pip3 install --no-binary ':all:' webrtc-noise-gain==1.2.3
```
... ah just testing to install the pre-compiled wheel here. The declaration looks fine: `webrtc_noise_gain-1.2.3-cp311-cp311-linux_armv6l.whl` and ... yes this is the problem:
```
home-assistant.service: Main process exited, code=killed, status=4/ILL
```

closed 2026-03-22 by MichaIng