MichaIng/DietPi #6687: Open Beta v8.23 | Please help testing and hardening our upcoming release

MichaIng, 2023-10-17:
RC version | v8.23.0
----------- | -------
v8.23.0 PR | #6686
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-10-21**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.23
____
### Beta v8.23.0
_(2023-10-17)_

#### Enhancements
- RPi | On Debian Bookworm and above, the RPi APT repository will be migrated to its new Bookworm suite. This solves issues and should enhance performance with FFmpeg, Kodi and some other A/V software, where we used the Debian packages with less hardware support before. The packages should be upgraded automatically on DietPi update. Please report back if you experience any issues during this process.
- ROCK 5B | Added an option to dietpi-config "Advanced Option" to flash the SPI bootloader, which enables NVMe boot for DietPi images.
- DietPi-Software | Firefox: Enabled the software option for RISC-V, since Debian provides packages now. But do not expect good performance, as GPU-acceleration is missing.
- DietPi-Software | ADS-B Feeder: The uninstall has been hardened to rule out the removal of unused Docker images which were not created by ADS-B Feeder. Furthermore, beta tags are now correctly shown in the version string. Many thanks to @andreagdipaolo for reporting a related issue and @dirkhh for implementing the enhancement: https://github.com/MichaIng/DietPi/pull/6587#issuecomment-1743744008
- DietPi-Software | DietPi-Dashboard: Experimental support for RISC-V has been added. Note that this will be an alpha build with alpha versions of certain dependencies, required to compile on RISC-V. Hence this is not suitable for production usage, which basically applies for RISC-V and the VisionFive 2 in general.

#### Bug fixes
- Raspberry Pi | Resolved an issue where the /dev/serial* symlinks were missing if binutils was not installed, which broke Bluetooth support among other things. Many thanks to @Rhiz3K for reporting this issue: https://github.com/MichaIng/DietPi/issues/6666
- Sparky SBC | Resolved an issue where our current image was not booting because the old initramfs did not support new ext4 filesystem capabilities. Many thanks to `@tenoritiger` for reporting this issue: https://dietpi.com/forum/t/sparky-sbc-allo-not-booting-os/18011
- DietPi daily cron | Resolved an issue where daily APT update checks failed if daily DietPi update checks were disabled. Many thanks to @lz1aam for reporting this issue: https://github.com/MichaIng/DietPi/issues/6651
- DietPi-DDNS | YDNS has been added to the list of natively supported DDNS providers. Many thanks to @edmundlaugasson for requesting and @TDuffinNTU for implementing it: https://github.com/MichaIng/DietPi/issues/5128, https://github.com/MichaIng/DietPi/pull/6674
- DietPi-Config | Resolved an issue where the WiFi channel selection was not possible with 5 GHz mode enabled. Many thanks to @lukaszsobala for reporting this issue: https://github.com/MichaIng/DietPi/issues/6636#issuecomment-1734427451
- DietPi-Software | Resolved an issue where motionEye failed to build on Bullseye systems since piwheels currently have no wheel for the latest Pillow version.
- DietPi-Software | frp: Resolved an issue where the frp client could not connect to the frp server, because of missing spaces around the equal sign for the token setting in /etc/frp/frpc.ini. Many thanks to @josemahj for reporting this issue: https://github.com/MichaIng/DietPi/issues/6647

lukaszsobala, 2023-10-18:
I switched from the master branch to the beta branch but the repository in `/etc/apt/sources.list.d/raspi.list` did not get changed from bullseye to bookworm. Also, the file `/etc/apt/preferences.d/dietpi-ffmpeg` was not removed. Why? 

On another system I switched from the master to the dev branch and these things did happen.

lukaszsobala, 2023-10-18:
> it would be appreciate if you can share some log from running `dietpi-update`

I am sorry but I don't have it any more, unless they are saved in some place. But there were no errors.

Joulinar, 2023-10-18:
it's not about errors but to check which steps are executed. For me this seems to be working fine. Just tested on a RPi3B+

- on DietPi v8.22.3

```
root@DietPi3:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: Skipping acquire of configured file 'non-free-firmware/binary-armhf/Packages' as repository 'http://raspbian.raspberrypi.org/raspbian bookworm InRelease' doesn't have the component 'non-free-firmware' (component misspelt in sources.list?)
root@DietPi3:~#
```

- switching die Beta will apply the update

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Migrating RPi APT repository to Bookworm and removing obsolete FFmpeg workaround
[  OK  ] DietPi-Pre-patches | eval echo 'deb https://archive.raspberrypi.org/debian/ bookworm main' > /etc/apt/sources.list.d/raspi.list
[  OK  ] DietPi-Pre-patches | rm /etc/apt/preferences.d/dietpi-ffmpeg
[ INFO ] DietPi-Pre-patches | Removing faulty non-free-firmware component if present
[  OK  ] DietPi-Pre-patches | sed -i s/ non-free-firmware$// /etc/apt/sources.list
[  OK  ] DietPi-Update | Successfully applied pre-patches
```

- new packages become available during update

```
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  console-setup console-setup-linux firmware-atheros firmware-brcm80211
  firmware-iwlwifi firmware-misc-nonfree firmware-realtek
  keyboard-configuration libc-bin libc-l10n libc6 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin
  libraspberrypi0 libssl3 locales openssl
20 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 59.8 MB of archives.
```

- all fine on DietPi v8.23.0 (beta)

```
root@DietPi3:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian bookworm InRelease
Hit:2 https://archive.raspberrypi.org/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@DietPi3:~#
```

MichaIng, 2023-10-18:
There was a bug in pre-patches, not being applied on DietPi v8.22 systems: https://github.com/MichaIng/DietPi/commit/04c0daa

MichaIng, 2023-10-18:
If it was v8.22 and you did `dietpi-update -1`, then it is internally reverted to v8.21. Probably this was the case?

lukaszsobala, 2023-10-18:
What is your suggestion now to apply all the pre-patches? Should I go back to the master branch and then again to the beta branch or are there no other pre-patches than the two things relating to the repositories?

MichaIng, 2023-10-18:
I just merged Beta v8.23.1 with the fix applied: #6691

anarkia47, 2023-10-19:
When applying the update from 8.21.1 to 8.23.1, the update  doesn't go through and i get the following error message:
#### Details:
- Date           | Thu Oct 19 14:02:39 BST 2023
- DietPi version | v8.21.1 (MichaIng/beta)
- Image creator  | 
- Pre-image      | 
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux DietPi 6.1.50-current-meson64 #2 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `gpg --keyserver hkp://keyserver.ubuntu.com:80 --no-default-keyring --keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg --recv-keys 2009837CBFFD68F45BC180471F4F90DE2A9B4BF8`
- Exit code      | 2
- Software title | DietPi-Pre-patches
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
[image]

#### Additional logs:
```
gpg: failed to create temporary file '/root/.gnupg/.#lk0x0000aaab1d43c610.DietPi.2173': No such file or directory
gpg: connecting dirmngr at '/root/.gnupg/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr

anarkia47, 2023-10-19:
I then ran this command: **curl -sSf 'https://keyserver.ubuntu.com/pks/lookup?search=0x2009837CBFFD68F45BC180471F4F90DE2A9B4BF8&op=get' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg --yes**

And re-tried to update ( dietpi-update) And got the following error: 
#### Details:
- Date           | Thu Oct 19 16:26:38 BST 2023
- DietPi version | v8.21.1 (MichaIng/beta)
- Image creator  | 
- Pre-image      | 
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux DietPi 6.1.50-current-meson64 #2 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

[image]

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://apt.sonarr.tv/debian buster InRelease
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:5 https://deb.debian.org/debian bookworm-backports InRelease
Hit:6 https://download.mono-project.com/repo/debian buster InRelease
Get:7 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Hit:8 https://armbian.site-meganet.com/apt bullseye InRelease
Err:3 https://apt.sonarr.tv/debian buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493 NO_PUBKEY 1F4F90DE2A9B4BF8
Fetched 6560 B in 2s (2713 B/s)
Reading package lists...
W: https://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://apt.sonarr.tv/debian/dists/buster/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.sonarr.tv/debian buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493 NO_PUBKEY 1F4F90DE2A9B4BF8
W: https://deb.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://deb.debian.org/debian/dists/bookworm-backports/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://download.mono-project.com/repo/debian/dists/buster/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://pkgs.tailscale.com/stable/debian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://apt.armbian.com/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
E: Failed to fetch https://apt.sonarr.tv/debian/dists/buster/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493 NO_PUBKEY 1F4F90DE2A9B4BF8
E: Some index files failed to download. They have been ignored, or old ones used instead.

MichaIng, 2023-10-19:
Did the `curl` command succeed or was there any error? And can you show:
```sh
ls -l /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg
```

MichaIng, 2023-10-19:
Ah, the permissions are the issue, and was also before, since `gpg` applies root-only read permissions, not sure why for a public key. However, this should fix it:
```sh
sudo chmod 0644 /etc/apt/trusted.gpg.d/dietpi-sonarr.gpg
```

MichaIng, 2023-10-19:
Can you show the error message of Tailscale?
```sh
journalctl -u tailscaled
```
And does it fails as well on RPi 8.21 when you upgrade only the Tailscale package?
```sh
apt update
apt install tailscale
```

Joulinar, 2023-10-19:
I guess the `iptables` package missing?

MichaIng, 2023-10-19:
Ah right, makes sense: #6649
```sh
apt install iptables
```

anarkia47, 2023-10-19:
Here's the outcome from **journalctl -u tailscaled**

https://pastebin.com/GHiKWd6D

"And does it fails as well on RPi 8.21 when you upgrade only the Tailscale package? "

I can revert and try if you wish 

Will try this first and report back:
**apt install iptables**

anarkia47, 2023-10-19:
It's working :)

Many thanks, you guys are awesome !!

MichaIng, 2023-10-19:
Beta v8.23.2 has been merged with the Sonarr fix applied, among others: #6696

MichaIng, 2023-10-21:
### DietPi v8.23 has been released: #6693
Many thanks to all testers ❤️ !

closed 2023-10-21 by MichaIng

lutfor-diu, 2023-10-27:
any update about tailscale? cant connect to server when new installation.

Joulinar, 2023-10-27:
This is something Tailscale themselves will fix, not us. After update, iptables package would need to be installed. In general the issue is unrelated to DietPi. 

Or what do you mean?

anarkia47, 2023-10-27:
> any update about tailscale? cant connect to server when new installation.

Run this command , it should fix it.

apt install iptables